lvgl/lvgl #2336: [alpha mix] lv_color_mix alpha should also mix.

junhuanchen, 2021-06-29:
https://github.com/lvgl/lvgl/blob/1cb57494df5b9477e08703f6e0e22d6175f9d24b/src/misc/lv_color.h#L470

https://github.com/lvgl/lvgl/blob/release/v8.0/src/misc/lv_color.h

```c
LV_ATTRIBUTE_FAST_MEM static inline lv_color_t lv_color_mix(lv_color_t c1, lv_color_t c2, uint8_t mix)
{
    lv_color_t ret;
#if LV_COLOR_DEPTH != 1
    /*LV_COLOR_DEPTH == 8, 16 or 32*/
    LV_COLOR_SET_R(ret, LV_MATH_UDIV255((uint16_t) LV_COLOR_GET_R(c1) * mix + LV_COLOR_GET_R(c2) *
                                        (255 - mix) + LV_COLOR_MIX_ROUND_OFS));
    LV_COLOR_SET_G(ret, LV_MATH_UDIV255((uint16_t) LV_COLOR_GET_G(c1) * mix + LV_COLOR_GET_G(c2) *
                                        (255 - mix) + LV_COLOR_MIX_ROUND_OFS));
    LV_COLOR_SET_B(ret, LV_MATH_UDIV255((uint16_t) LV_COLOR_GET_B(c1) * mix + LV_COLOR_GET_B(c2) *
                                        (255 - mix) + LV_COLOR_MIX_ROUND_OFS));
    LV_COLOR_SET_A(ret, LV_MATH_UDIV255((uint16_t) LV_COLOR_GET_A(c1) * mix + LV_COLOR_GET_A(c2) *
                                        (255 - mix) + LV_COLOR_MIX_ROUND_OFS));
#else
    /*LV_COLOR_DEPTH == 1*/
    ret.full = mix > LV_OPA_50 ? c1.full : c2.full;
#endif

    return ret;
}

```

I wanted to confirm that in Release 7.

kisvegabor, 2021-07-01:
For compatibility with color formats where there is no `alpha` channel (e.g. RGB565) LVGL handles the alpha channel separately. 

Normally, when LVGL renders the content the rendered pixels always have `alpha=255` else the screen would be transparent. 
So the alpha channel of the background pixels is lost when it was merged to its "parent" layer. 

If you want the screen to have `alpha` channel (e.g. render to create an OSD menu) you need to enable `LV_COLOR_SCREEN_TRANSP` in `lv_conf.h`. It enables to use of more complex mix functions (namely `lv_color_mix_with_alpha`) that consider the alpha channel too.

junhuanchen, 2021-07-12:
In the case of image mixing with multiple layers, such as camera + LVGL, there will be single pixel hardware mixing. In this case, LV_color_set_a (ret, 0xFF) should not be used; Fixed LVGL Canvas for all Pixel Blends.

It can be configured with style as follows:

` ` `
lv_style_init(&xiezhu_style);
lv_style_set_bg_color(&xiezhu_style, LV_STATE_DEFAULT, (lv_color_t){0xFF000000});
` ` `

But because the final LV_color_mix function LV_color_set_a (ret, 0xFF); Causing alpha to lose its function.

can override the lv_color_mix function ?

kisvegabor, 2021-07-13:
To set the alpha of the bg_color you should use `lv_style_set_bg_opa(style, LV_STATE_DEFAULT, LV_OPA_...)`.

junhuanchen, 2021-07-20:
The Pixel Blend option should be able to support the design of a single pixel transparency

closed 2021-07-20 by junhuanchen