lvgl/lvgl #2337: Writing unit tests for core widgets

kisvegabor, 2021-06-29:
Hi,

We have added a new test engine (called [Unity](https://github.com/ThrowTheSwitch/Unity/)) to LVGL and planning to improve the coverage.  See this [README](https://github.com/lvgl/lvgl/blob/master/tests/README.md) about how to write and run tests.

I've already created a test for the [drop-down list](https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/test_dropdown.c) as an example and proof of concept.

Writing unit tests for the whole library at once would be a huge work, so let's do it step by step.

As the first step, I suggest adding tests for every [core widget](https://github.com/lvgl/lvgl/tree/master/src/widgets).

If you are interested in contributing with writing tests please let us know here and pick a widget. :slightly_smiling_face: 

- [ ] lv_arc.c
- [ ] lv_bar.c
- [ ] lv_btn.c
- [ ] lv_btnmatrix.c
- [ ] lv_canvas.c
- [ ] lv_checkbox.c
- [x] lv_dropdown.c
- [ ] lv_img.c
- [ ] lv_label.c
- [ ] lv_line.c
- [ ] lv_roller.c
- [ ] lv_slider.c
- [ ] lv_switch.c
- [ ] lv_table.c
- [ ] lv_textarea.c

C47D, 2021-08-24:
Are you going to update this test?

https://github.com/lvgl/lvgl/blob/088b2bd49080cdd19e45d79bfd7e72e59f9bd6ec/tests/src/test_cases/test_dropdown.c#L50

kisvegabor, 2021-08-25:
@C47D 
Why is it need to be updated?

C47D, 2021-09-02:
It should be nice to add tests for fixed bugs, first we need to add a test that replicates the bug (this test should fail), then we make the bug fix, then the test should pass.

kisvegabor, 2021-09-03:
I agree, it'd be the ideal case. However, IMO before adding very special test cases it'd be better to add some simple tests for the simple features first. 

To get started with we can even set up a a very simple goal: have at least a basic "render" test for every widget. 
What do you think?

C47D, 2021-09-05:
I agree, but what do you mean by render test?

I've been seeing some talks about how to test Qt projects, and I was thinking on using the same approach here. They don't tend to use screen captures.

I was thinking about #2522 when I made the previous comment. What arc properties controls the drawing direction of the arc, we could replicate the user issue with their code snippet and assert the expected propertie values. But I don't know how LVGL do that.

kisvegabor, 2021-09-06:
A lot of things can be tested by setting and getting properties. And IMO it should be the main approach when writing tests.

However comparing some screenshots will easily test a lot of things at once. E.g. widget creation, setting something, drawing,   styles, position, size, etc.

C47D, 2021-09-10:
Did a test for 2522, just to test the new test improvements, I got the value of `expected_start_angle` after running the test and letting it fail. But I don't understand why the value is 36, am I missing something?

```c
#if LV_BUILD_TEST
#include "../lvgl.h"

#include "unity/unity.h"

void test_bugfix_for_2522(void)
{
    uint16_t expected_start_angle = 36;
    uint16_t expected_end_angle = 90;

    lv_obj_t *arcBlack;
    arcBlack = lv_arc_create(lv_scr_act());

    lv_arc_set_mode(arcBlack, LV_ARC_MODE_REVERSE);
    lv_arc_set_bg_angles(arcBlack, 0, 90);

    /* lv_arc_set_value calls value_update, where the fix is done */
    lv_arc_set_value(arcBlack, 40);

    TEST_ASSERT_EQUAL_UINT16(expected_start_angle, lv_arc_get_angle_start(arcBlack));
    TEST_ASSERT_EQUAL_UINT16(expected_end_angle, lv_arc_get_angle_end(arcBlack));
}

#endif
```

C47D, 2021-09-10:
I find this article very useful to assign names to test, maybe you find it interesting https://www.codurance.com/publications/2014/12/13/naming-test-classes-and-methods

C47D, 2021-09-17:
@kisvegabor have you felt the need to add mocking capability to the test suite?

I've worked a bit with a framework named fff and CMock. We could add asserts to be sure there are some function calls, with specific parameters and mocking return values.

kisvegabor, 2021-09-17:
For first I think we can test many things without mocking, but later it'll be surely required to make more isolated and specific tests. 

If CMock has all the required features probably it should be added as it works well with Unity.

C47D, 2021-09-17:
Do you have any priority list for adding tests? I've seen some interesting bugs in the calendar widget.

Also, one question, when the checkbox state is disabled and the user clicks on it, should it's event handler be called? I guess it shouldn't, but I'm not sure about that.

vahidajalluian, 2021-09-17:
@C47D 
Have you encountered the same bug as the one I have reported recently for the calendar widget?

C47D, 2021-09-17:
Hi, no, but I was wondering if we can replicate it using unit tests.

kisvegabor, 2021-09-17:
> Do you have any priority list for adding tests?

It's not a strong opinion but I think adding tests to the widgets is a good start because users have direct connection with them. So we test what is directly used, and we can find contributors easier to such tangible topic. 

If the question is which widgets should be first, it doesn't really matter. If we get started with it all widgets should have test soon.

C47D, 2021-11-03:
I think we should update the list of widgets on the first commit. Do you want me to add test for a particular widget next?

kisvegabor, 2021-11-08:
> I think we should update the list of widgets on the first commit. 

At first I added only the core widgets and not the ones from the extra folder. I suggest finishing these first and opening a new issue for the rest.

> Do you want me to add test for a particular widget next?

Switch, bar and slider are simple, I think they can be easily tested. Are interested in them?

Probably Label and Textarea are the most complex ones and there are some related misc modules (lv_bidi, lv_txt, lv_txt_ap) that should be also tested. I can work on these.

C47D, 2021-11-08:
> At first I added only the core widgets and not the ones from the extra folder. I suggest finishing these first and opening a new issue for the rest.

I meant to update the checkboxes, as some of them have test already.

kisvegabor, 2021-11-09:
Update. Let me know if I missed something.

C47D, 2021-11-11:
> 
> 
> Update. Let me know if I missed something.

It's all good, I will add the tests for the widgets you requested on the coming weeks.

kisvegabor, 2021-11-11:
> I've just noticed that this commit is not applied? 7278089

I've added it again here b279f63d6 
Maybe it was overwritten by a force push, sorry for that. :slightly_smiling_face:

C47D, 2021-11-11:
No problem, thanks for fixing it.

embeddedt, 2021-11-11:
@kisvegabor I think force-pushing on branches we expect users/downstream projects to actively be using is dangerous. This would break any scripts/CI logic that expected there to be a single, sequential stream of commits.

kisvegabor, 2021-11-12:
I force pushed a few times only to fix trivial problems in my very last commit. But, yes, it seems it should be avoided.

puzrin, 2021-12-17:
Unity is certainly much better than nothing/handmade, but IMO still far from perfect. Are you sure lvgl should avoid  cpp frameworks like "google test" and so on?

kisvegabor, 2021-12-20:
I don't have a strong opinion about the framework. Unity was easy to get started with and we can port the tests (mainly refactor the asserts I suppose) to an other framework if required.

C47D, 2021-12-20:
I've used CPPUTest, that as far as I know has a migration script from unity to CPPUTest.

puzrin, 2021-12-20:
> Also AFAIK the python script gathers the files and dump the result into a cmake file.
> 
> With this note you mean creating the cmake manually?

With `GTest` you don't need to collect test functions, those are auto-detected. But you may wish to have several configs (for example, with different display sizes), and that's convenient to manage via cmake. In trivial case, you can just define list of files to build (`/test/*.c`). So, yes, cmake config should be created manually, but it's very simple (can use wildcards).

Advantage is - using standard tools.

CPUTest is certainly better than Unity. When i asked at local forum "please recommend modern testing for C/CPP", it was second popular recommendation after GTest.

C47D, 2022-01-07:
@puzrin I have a question, do you test assertions? At my previous work I used [CException](https://github.com/ThrowTheSwitch/CException) to be able to test assertions being fired when the condition they're verifying is not met.

puzrin, 2022-01-08:
@C47D i'm certaintly not the person who can give qualified answers about C/CPP :)

But you can check gtest docs about assertions https://google.github.io/googletest/reference/assertions.html

C47D, 2022-01-08:
Thanks for the info @puzrin, I'm not an expert neither.

puzrin, 2022-01-08:
@C47D i can explain generic approaches about testing. Problem with Unity is that it has nothing but assertions. And tend users to create hand-made scripts. That's certainty possible, but very ineffective. And experience from such work it not portable to other projects.

When you work with tests, at some moments you will start to use all this features:

- assertions
- mocks
- grouping tests (convenient)
- selective run (while you work with specific feature)
- coverage reports (very important)
- auto-detect new added tests
- multiple builds (for example, with different config)
- hw-independent benchmarks (via simulator like Gen5 with cycles count). ideally - with logs on server to track perf regressions

It would be very boring/expensive to rework tests every time you wish to add new feature. Or spent time to reinvent the wheel for custom solution.

IMO gtest + cmake could be nice choice. The only disadvantage - it requires CPP. But AFAIK, this is not principal restriction for tests.

embeddedt, 2022-01-08:
> The only disadvantage - it requires CPP. But AFAIK, this is not principal restriction for tests.

I don't think it matters for tests either, as we don't expect users to run them on their target platform. Assuming a developer has access to C++ on their host machine is completely reasonable.

xiaoxiang781216, 2022-01-08:
> IMO gtest + cmake could be nice choice. The only disadvantage - it requires CPP. But AFAIK, this is not principal restriction for tests.

But it is nice feature to run test on the real device, I have saw that libc++ plus gtest need at least 300KB on Cortex-M, which is even large than lvgl self, so it's better to avoid to bring in C++ in this level.

puzrin, 2022-01-08:
> But it is a nice feature to run test on the real device,

Do you know real (not imagined or nice) need to run lvgl tests on real device? IMO you suggest to reject really useful things in favor of something rare.

HBehrens, 2022-07-05:
Are there any plans to add test coverage for the low-level drawing code?

I am considering making contributions to the masking code (and later to the public drawing API itself) and am worried about regressions. Also, I see that the SDL renderer differs significantly from the alternatives that for the most part share a substantial code from the SW renderer – it would be great to see how similar their rendering output really is.

HBehrens, 2022-07-05:
> > But it is a nice feature to run test on the real device,
> 
> Do you know real (not imagined or nice) need to run lvgl tests on real device? IMO you suggest to reject really useful things in favor of something rare.

When trying to validate the different hardware-dependent renderers, it is very helpful to run (a subset of) the suite on real hardware. In the past, I was doing hardware-in-the-loop tests to
1. validate assumptions about the specific hardware in isolation and
2. to a limited integration test using snapshot testing (here: comparing pixel output) for a given hardware against a reference

kisvegabor, 2022-07-07:
> Are there any plans to add test coverage for the low-level drawing code?

Yes, screenshot comparison would work very well here. We already have [an assert](https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/test_dropdown.c#L364) for screenshot compare.

Contribution with tests for the rendering engine(s) would be very welcome. I think we should start with SW render and make screenshots containing a lot of rectangles/labels/images/etc with different settings. What do you think?

C47D, 2022-07-10:
Hi @HBehrens 

> I don't see why HITL prevents one from having a convenient, modern testing framework, though.

For me I don't have enough hardware, only an ESP32 and two or three screens.

> From what I can see Unity does not match my expectation of such a testing framework.

Which framework do you like using? I personally use CppUTest

kisvegabor, 2022-07-26:
Hi,

I have some great news! 

LVGL has an Open collective account. It wasn't advertised much be we already collected some money there. From now on I'll pay more attention to properly highlight the sponsorship opportunity and with @embeddedt we figured out a great way to spend the donations. 

We will mark some issues with "Sponsored" label and if someone implements or fixes those issues he or she gets payment from the accumulated donations. You can learn more about it [here](https://github.com/lvgl/lvgl/blob/master/README.md#heart-sponsor).

And this this issue would be great get it started. So I marked this issue with "Sponsored" label and added some details to the first post.

embeddedt, 2022-07-26:
Quick clarification note: the wording "50 USD per unit test" is a bit unclear as to what a "unit test" is. Does this mean 50 USD is paid out per widget (e.g. if tests are added for all the major features of a roller)? I just wanted to make sure this is clear before anyone signs up.

kisvegabor, 2022-07-26:
Good point. I updated it ti "50 USD / fully unit testing a widget"

C47D, 2022-07-28:
What fully tested implies? 100% coverage?

C47D, 2022-08-09:
Coverage 90% sounds good to me.

mmurty, 2022-08-12:
I usually use this 
```
#ifdef TEST
     #define STATIC 
#else
     #define STATIC static
#endif
```

use the `STATIC` keyword while writing your code.

mmurty, 2022-08-12:
Hi @kisvegabor,
I would like to take up labels for unit testing. Let me know how to proceed.

kisvegabor, 2022-08-15:
Hi @mmurty,

`lv_label` is almost finished by @C47D [here](https://github.com/lvgl/lvgl/pull/3529). Maybe something else? :slightly_smiling_face:

mmurty, 2022-08-15:
@kisvegabor Sure.
What about buttons? Or you can suggest something that's simpler.

C47D, 2022-08-15:
@mmurty Yep, I was trying to use that but there's a lot of widget static functions named the same and I think all the source files are being compiled at the same time, so I think we will get a compilation error. Will have to take a look at that, thanks for the suggestion.

I'll continue with the spinbox widget to cover this issue #3559

kisvegabor, 2022-08-18:
@mmurty Button is very simple. What about the button matrix instead?

mmurty, 2022-08-19:
@kisvegabor I can take up both then. I would initially like to start with something simple as it is my first contribution for LVGL. Once I am comfortable I can take up the complex stuff.

kisvegabor, 2022-08-19:
@mmurty Okay, let's start with lv_btn. However I think all you can whether these flags are set correctly
```c
    lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLLABLE);
    lv_obj_add_flag(obj, LV_OBJ_FLAG_SCROLL_ON_FOCUS);
```

Regarding the sponsorship, what about we give 60 USD for the button and the button matrix together?

mmurty, 2022-08-19:
@kisvegabor I'm okay with 60USD for both because as you said buttons doesn't have much to test. So let me first get started with things and deliver something and later if you feel I have put more effort may be you can increase the amount.
Please direct me where I can get instructions to get started with setting up test the environment.

mmurty, 2022-08-20:
@kisvegabor Can you please point out what dependencies needs to be installed and how to run existing tests?

embeddedt, 2022-08-23:
@mmurty You will need a Linux environment, preferably Ubuntu or Debian-based since that's what the CI runners use.

Ensure you have [these prerequisites](https://github.com/lvgl/lvgl/blob/master/scripts/install-prerequisites.sh) installed, then go into the `tests` folder and run `./main.py test`. If the tests complete successfully, your environment should be working correctly.

To add new test cases just create a new source file in `tests/src/test_cases`.

mmurty, 2022-08-23:
@embeddedt Thanks for the inputs. I was actually able to setup things. The readme helped.

mmurty, 2022-08-26:
Hi All,
I am trying to write unit tests for the btnmatrix widget. I am pretty much done testing the getter and setter functions.
Now I want to test the event code. I am trying to forcefully call the event handler but as some portion of the code makes use of functions like lv_indev_get_act(), I am unable to force the event.

Does anyone know a better way to fake an event?

embeddedt, 2022-08-26:
@mmurty We have some APIs for faking input, see `lv_test_indev.h`.

mmurty, 2022-08-29:
@embeddedt Thanks for the tip. I could simulate the mouse input using `lv_test_mouse_click_at(x, y);` API.
I am trying to fake the right key input using `lv_test_key_hit(LV_KEY_RIGHT);` but a key press event is not getting triggered.
Can anyone suggest something in this direction?

embeddedt, 2022-08-29:
Your button matrix probably needs to be in a group. You can use [one of the dropdown test cases](https://github.com/lvgl/lvgl/blob/2c7760c6875dce510133c7c6788dc447cb8c2d5b/tests/src/test_cases/test_dropdown.c#L161-L189) as a reference.

C47D, 2022-09-01:
@kisvegabor @embeddedt Do you need unit tests for another widget in specific? I think the list in the first comment needs to be updated.

[81 more replies]

closed 2024-06-20 by kisvegabor