lvgl/lvgl #2340: Roller with different font.

xaowang96, 2021-07-01:
### Introduce the problem
I use roller and set the normal font to lv_font_montserrat_12, and select the font to lv_font_montserrat_48. This causes the selected text to not be displayed completely, like this:
[image]

### Examples and cases
The phenomenon can be reproduced by the following code：
```
static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_VALUE_CHANGED) {
        char buf[32];
        lv_roller_get_selected_str(obj, buf, sizeof(buf));
        LV_LOG_USER("Selected month: %s\n", buf);
    }
}

/**
 * An infinite roller with the name of the months
 */
void lvx_example_roller_1(lv_obj_t* parent)
{
    lv_obj_t *roller1 = lv_roller_create(parent);
    lv_obj_set_width(roller1, lv_pct(100));
    lv_roller_set_options(roller1,
                        "January\n"
                        "February\n"
                        "March\n"
                        "April\n"
                        "May\n"
                        "June\n"
                        "July\n"
                        "August\n"
                        "September\n"
                        "October\n"
                        "November\n"
                        "December",
                        LV_ROLLER_MODE_NORMAL);

    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_12, 0);
    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_48, LV_PART_SELECTED);
    lv_roller_set_visible_row_count(roller1, 3);
    lv_obj_set_style_bg_opa(roller1, LV_OPA_0, 0);
    lv_obj_center(roller1);
    lv_obj_add_event_cb(roller1, event_handler, LV_EVENT_ALL, NULL);
}
```
So what should I do to solve this problem?

xaowang96, 2021-07-01:
[image]
I used to modify LV_PART_MAIN in line 514 to LV_PART_SELECTED, and the result is as follows, which is not what I want:
[image]

kisvegabor, 2021-07-01:
Hi,

Try increasing the `text_line_height` of the roller's style. E.g.

```c
    lv_obj_set_style_text_line_height(roller1, 50, 0);
```

xaowang96, 2021-07-02:
Hi,
There is no such api, i used this.
`    lv_obj_set_style_text_line_space(roller1, 50, 0);
`
The result is as follows
[image]

This result looks better, but I don’t want the line space to be so big.

kisvegabor, 2021-07-02:
> There is no such api, i used this.

Ah, sorry.

> This result looks better, but I don’t want the line space to be so big.

50 was just an example, you can find the number that looks good.

xaowang96, 2021-07-02:
> > There is no such api, i used this.
> 
> Ah, sorry.
> 
> > This result looks better, but I don’t want the line space to be so big.
> 
> 50 was just an example, you can find the number that looks good.

I hope that the selected part of the font becomes larger or smaller, but the line space does not change.The rectangular box is set according to the height of the default font, I want this part to be calculated according to the height of the selected font

kisvegabor, 2021-07-05:
I was thinking about it and you are right. I've just pushed a fix that takes the selected font size into account too.

commit 01d1c873 (2021-07-05): fix(roller) adjust the size of the selected area correctly

fixes #2340
closed 2021-07-05 by kisvegabor