lvgl/lvgl #2341: undefined reference to `_Assert'

lisemi, 2021-07-01:
### Describe the bug

When porting LVGL8.0.0 to M3, the compiler has the following problems:
```
undefined reference to `_Assert'	lv_tlsf.o
more undefined references to `_Assert' follow	lv_tlsf.c
undefined reference to `_Assert'	lv_tlsf.c
expected identifier or '(' before 'const'	assert.h
```
May I ask how to solve the problem.

kisvegabor, 2021-07-01:
Thanks, I've pushed a fix. 

I don't see this error,  so please let me know if it's really fixed :slightly_smiling_face:

closed 2021-07-01 by kisvegabor

PadenHuang, 2023-12-06:
f:/project-ql-xueban/lte01r02a01_beta0711_c_sdk_g-hwq/prebuilts/win32/gcc-arm-none-eabi/gcc-arm-none-eabi-9.2.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: lib/liblvgl.a(qrcodegen.c.obj): in function `calcSegmentBitLength':
F:\project-ql-xueban\LTE01R02A01_BETA0711_C_SDK_G-hwq\out\8850CM_cat1_open_release/../../components/ql-application/lvgl_lib/lvgl/src/extra/libs/qrcode/qrcodegen.c:852: undefined reference to `_assert'

kisvegabor, 2023-12-06:
In both `release/v8.3` and `master` I replaced `assert` with our internal `LV_ASSERT`

PadenHuang, 2023-12-07:
Problem solved successfully. Thank you！

commit c9745b9c (2021-07-01): fix(tlsf) do not use <assert.h>

fixes: #2341
commit 4ac4d87f (2021-07-16): fix(tlsf) do not use <assert.h>

fixes: #2341