lvgl/lvgl #2348: In release/v7.7.1 version：  when the content of my picture is empty or bad, it will cause lvgl to crash,  my picture format is png

jianjunxia, 2021-07-05:
I deliberately used a bad picture for testing this is my test code:
```c
#define TY_IMG_TEST "S:" TY_IMG_PATH "bad.png"
STATIC OPERATE_RET  gui_socket_set_img(lv_obj_t *obj)
{
    lv_obj_t *socket_img=lv_img_create(obj,NULL);
    lv_img_set_src(socket_img, TY_IMG_TEST);
    lv_obj_align(socket_img, NULL, LV_ALIGN_CENTER,0, 0);
    return OPRT_OK;
}
```
this will cause the system to crash，the reason is as follows：

==38575==ERROR: AddressSanitizer: attempting double-free on 0x6270003e2100 in thread T11 (lv_task):
    #0 0x7ffff768e7cf in __interceptor_free (/lib/x86_64-linux-gnu/libasan.so.5+0x10d7cf)
    #1 0x7ffff6e0e9dc in decoder_close /lv_lib_png/lv_png.c:234
    #2 0x7ffff6ff3559 in lv_img_decoder_close/lvgl/src/lv_draw/lv_img_decoder.c:183
    **#3 0x7ffff6feb478 in _lv_img_cache_open/lvgl/src/lv_draw/lv_img_cache.c:132**
    #4 0x7ffff6fcaca2 in lv_img_draw_core/lvgl/src/lv_draw/lv_draw_img.c:245
    #5 0x7ffff6fca92a in lv_draw_img/lvgl/src/lv_draw/lv_draw_img.c:91
    #6 0x7ffff6fb41f6 in lv_img_design/lvgl/src/lv_widgets/lv_img.c:712
    #7 0x7ffff6eee216 in lv_refr_obj/lvgl/src/lv_core/lv_refr.c:677
    #8 0x7ffff6eee448 in lv_refr_obj/lvgl/src/lv_core/lv_refr.c:717
    #9 0x7ffff6eee448 in lv_refr_obj/lvgl/src/lv_core/lv_refr.c:717
    #10 0x7ffff6eeddef in lv_refr_obj_and_children/lvgl/src/lv_core/lv_refr.c:619
    #11 0x7ffff6eedaf2 in lv_refr_area_part/lvgl/src/lv_core/lv_refr.c:550
    #12 0x7ffff6eed0d7 in lv_refr_area/lvgl/src/lv_core/lv_refr.c:460
    #13 0x7ffff6eec6b0 in lv_refr_areas/lvgl/src/lv_core/lv_refr.c:382
    #14 0x7ffff6eeb8d8 in _lv_disp_refr_task/lvgl/src/lv_core/lv_refr.c:199
    #15 0x7ffff6fff1f5 in lv_task_exec/lvgl/src/lv_misc/lv_task.c:409
    #16 0x7ffff6ffe5d0 in lv_task_handler/lvgl/src/lv_misc/lv_task.c:142

freed by thread T11 (lv_task) here:
    #0 0x7ffff768e7cf in __interceptor_free (/lib/x86_64-linux-gnu/libasan.so.5+0x10d7cf)
    #1 0x7ffff6e0e9dc in decoder_close /lv_lib_png/lv_png.c:234
    #2 0x7ffff6ff3559 in lv_img_decoder_close/lvgl/src/lv_draw/lv_img_decoder.c:183
    **#3 0x7ffff6feb478 in _lv_img_cache_open/lvgl/src/lv_draw/lv_img_cache.c:118**
    #4 0x7ffff6fcaca2 in lv_img_draw_core/lvgl/src/lv_draw/lv_draw_img.c:245
    #5 0x7ffff6fca92a in lv_draw_img/lvgl/src/lv_draw/lv_draw_img.c:91
    #6 0x7ffff6fb41f6 in lv_img_design/lvgl/src/lv_widgets/lv_img.c:712
    #7 0x7ffff6eee216 in lv_refr_obj/lvgl/src/lv_core/lv_refr.c:677
    #8 0x7ffff6eee448 in lv_refr_obj/lvgl/src/lv_core/lv_refr.c:717
    #9 0x7ffff6eee448 in lv_refr_obj/lvgl/src/lv_core/lv_refr.c:717
    #10 0x7ffff6eeddef in lv_refr_obj_and_children/lvgl/src/lv_core/lv_refr.c:619
    #11 0x7ffff6eedaf2 in lv_refr_area_part/lvgl/src/lv_core/lv_refr.c:550
    #12 0x7ffff6eed0d7 in lv_refr_area/lvgl/src/lv_core/lv_refr.c:460
    #13 0x7ffff6eec6b0 in lv_refr_areas/lvgl/src/lv_core/lv_refr.c:382
    #14 0x7ffff6eeb8d8 in _lv_disp_refr_task/lvgl/src/lv_core/lv_refr.c:199
    #15 0x7ffff6fff1f5 in lv_task_exec/lvgl/src/lv_misc/lv_task.c:409
    #16 0x7ffff6ffe5d0 in lv_task_handler/lvgl/src/lv_misc/lv_task.c:142

Solution：
  in lv_lib_png/lv_png.c

old code：
```c
static void decoder_close(lv_img_decoder_t * decoder, lv_img_decoder_dsc_t * dsc)
{
    (void) decoder; /*Unused*/
    if(dsc->img_data) free((uint8_t *)dsc->img_data);
}
```
fix code：
```c
static void decoder_close(lv_img_decoder_t * decoder, lv_img_decoder_dsc_t * dsc)
{
    (void) decoder; /*Unused*/
    if(dsc->img_data) {
        free((uint8_t *)dsc->img_data);
        dsc->img_data = NULL;
    }
}
```

I will PR later

**In addition, I have another question to ask. When the content of a picture is bad, do we have some kind of verification or clarification mechanism?，This is an important question for me**

kisvegabor, 2021-07-06:
Hi,

Similarly to [this PR](https://github.com/lvgl/lv_lib_png/pull/18#issuecomment-874116029), please send one for the [release/v7](https://github.com/lvgl/lv_lib_png/tree/release/v7) branch too.

kisvegabor, 2021-07-06:
Let's continue in https://github.com/lvgl/lv_lib_png/pull/18

closed 2021-07-06 by kisvegabor