lvgl/lvgl #2351: Optimizing  style properties size

guoweilkd, 2021-07-05:
Currently, each style properties id is stored use `uint16_t`, But the maximum value of build in properties  id  is only 106 . 5 bits are occupied by `LV_STYLE_PROP_INHERIT .... LV_STYLE_PROP_FILTER`. Why don't we separate out this properties ( `LV_STYLE_PROP_INHERIT .... LV_STYLE_PROP_FILTER`), and use get function instead. At the same time, Reduce the properties  id size to `uint8_t`.
like this:
```
bool lv_is_style_prop_inherit(uint8_t prop)
{
    bool res = false;
    switch(prop) {
    case LV_STYLE_TEXT_COLOR:
    case LV_STYLE_TEXT_FONT:
    case LV_STYLE_TEXT_ALIGN: res = true;break;
    default:break;
    }

    //Custom prop judgment
    return res;
}
```

kisvegabor, 2021-07-05:
Hi,

Good idea!  

I see only one problem. If a new style property is regsitered with `lv_style_register_prop()` the resulting value can be ORed with `LV_STYLE_PROP_<FLAG>`. E.g. to register an inheritable property. 

[Here](https://github.com/lvgl/lvgl/blob/6bc40f8c4417a94ab26b25220324e471e03ce443/src/extra/layouts/flex/lv_flex.c#L93-L96) flexbox properties are set to show they are layout related.  (So LVGL can perform the necessary calculations if the properties change)

guoweilkd, 2021-07-06:
Maybe we can do it this way:
```
static uint8_t* g_prop_array = NULL;
static uint16_t act_id = 0;
lv_style_prop_t lv_style_register_prop(uint8_t prop)
{
    act_id = (uint16_t)_LV_STYLE_LAST_BUILT_IN_PROP;
    act_id++;

    if (g_prop_array == NULL) {
        g_prop_array = lv_mem_alloc();
    }
    else {
        g_prop_array = lv_mem_realloc();
    }

    g_prop_array[act_id - _LV_STYLE_LAST_BUILT_IN_PROP - 1] = prop;

    return act_id;
}

bool lv_is_style_prop_inherit(uint8_t prop)
{
    bool res = false;
    switch(prop) {
    case LV_STYLE_TEXT_COLOR:
    case LV_STYLE_TEXT_FONT:
    case LV_STYLE_TEXT_ALIGN: res = true;break;
    default:
        if (prop > _LV_STYLE_LAST_BUILT_IN_PROP) {
            res = g_prop_array[act_id - _LV_STYLE_LAST_BUILT_IN_PROP - 1] & LV_STYLE_PROP_<FLAG> ? true : false;
        }
        break;
    }
    return res;
}
```
E.g.  `lv_style_register_prop(LV_STYLE_PROP_INHERIT | LV_STYLE_PROP_LAYOUT_REFR)`

kisvegabor, 2021-07-06:
Sounds good. 

I've tried to estimate how much memory we can save with this update:
1. In `lv_theme_default` there is 175 `lv_style_set_*` calls so it's 175 byte save if all widgets are enabled.
2. If local styles are it's 1 byte/property. If size, align and position is set manually (5 properties) on 50 object it's 250 bytes.

So in total a few hundred bytes on an average UI. Not a game-changer but why not. :slightly_smiling_face: 

We should check one thing tough: `if(prop & <FLAG>){ .. }` is very fast. A switch case is definitely slower and there are many style property querries in LVGL. 

Could you make a speed test by replacing e.g. [this line](https://github.com/lvgl/lvgl/blob/master/src/core/lv_obj_style.c#L209) with a switch case version?

guoweilkd, 2021-07-07:
I tested it on the computer（no development board in hand）， use `cat /proc/cpuinfo` get frequency   is 2001Mhz. gcc optimize is -o0
```
uint8_t propbuff[256];
bool is_filter(lv_style_prop_t prop);
inline bool is_filter(lv_style_prop_t prop)
{
    switch(prop) {
        case LV_STYLE_WIDTH:
        case LV_STYLE_MAX_WIDTH:
        case LV_STYLE_HEIGHT:
        case LV_STYLE_MIN_HEIGHT:
        case LV_STYLE_MAX_HEIGHT:
        case LV_STYLE_X:
        case LV_STYLE_Y:
        case LV_STYLE_ALIGN:
        case LV_STYLE_TEXT_ALIGN:
        case LV_STYLE_TEXT_LINE_SPACE:
        case LV_STYLE_TEXT_LETTER_SPACE:
        case LV_STYLE_BORDER_WIDTH:
        case LV_STYLE_TEXT_COLOR_FILTERED:
        case LV_STYLE_RADIUS:
        case LV_STYLE_CLIP_CORNER:
        case LV_STYLE_OPA:
        case LV_STYLE_COLOR_FILTER_DSC:
        case LV_STYLE_COLOR_FILTER_OPA:
        case LV_STYLE_ANIM_TIME:
        case LV_STYLE_ANIM_SPEED:
        case LV_STYLE_TRANSITION:
        case LV_STYLE_BG_COLOR_FILTERED:return true;
        default: 
            if (prop > _LV_STYLE_LAST_BUILT_IN_PROP) {
                return propbuff[prop - _LV_STYLE_LAST_BUILT_IN_PROP - 1] & LV_STYLE_PROP_FILTER ? true : false;
            } else {
                return false;
            }
    }
}

void main(void)
{
    lv_style_prop_t prop;
    struct timespec ts1;
    struct timespec ts2;

    #define CNT_NUM 500u
    time_t timebuff[CNT_NUM];
    time_t max = 0;
    time_t min = 0xffffffff;
    time_t average = 0;

    for (int cnt = 0; cnt < CNT_NUM; cnt++) {
        /* & test */
        clock_gettime(CLOCK_REALTIME, &ts1);
        for (uint32_t i = 0; i < 1000000; i ++) {
            prop = i % 256;
            bool filter = prop & LV_STYLE_PROP_FILTER ? true : false;
            filter = filter ? false : true;
        }
        clock_gettime(CLOCK_REALTIME, &ts2);
        time_t tmp1 = (ts2.tv_sec * 1000000 + ts2.tv_nsec / 1000) - (ts1.tv_sec * 1000000 + ts1.tv_nsec / 1000);

        /* switch test */
        clock_gettime(CLOCK_REALTIME, &ts1);
        for (uint32_t i = 0; i < 1000000; i ++) {
            prop = i % 256;
            bool filter = is_filter(prop);
            filter = filter ? false : true;
        }
        clock_gettime(CLOCK_REALTIME, &ts2);
        time_t tmp2 = (ts2.tv_sec * 1000000 + ts2.tv_nsec / 1000) - (ts1.tv_sec * 1000000 + ts1.tv_nsec / 1000);

        timebuff[cnt] = tmp2 - tmp1;
        printf("%ld ",timebuff[cnt]);

        max = max > timebuff[cnt] ? max : timebuff[cnt];
        min = min < timebuff[cnt] ? min : timebuff[cnt];
        average += timebuff[cnt];
    }

    average = average / CNT_NUM;

    printf("\n max:%ld, min:%ld, average:%ld (unit:us)\n", max, min, average);
}
```
test result:
Here is the maximum, minimum, and average of 500 tests(time is `switch` run 1000000 times time - `&` run 1000000 times time )
max:12245, min:737, average:1632 (unit:us)

 I don't think there's much of a difference in efficiency,but As you said, it doesn't save much memory.Maybe it's more useful when style have a lot of extended properties as `LV_STYLE_PROP_INHERIT`

kisvegabor, 2021-07-09:
Thanks!  I also made some tests on an STM32F746 board (200MHz with -O3) with the widgets demo. I modifed `lv_obj_get_style_prop` like this:
```c
extern bool filter_g;
extern bool inherit_g;

lv_style_value_t lv_obj_get_style_prop(const lv_obj_t * obj, lv_part_t part, lv_style_prop_t prop)
{

    lv_style_value_t value_act;
    bool inherit = prop & LV_STYLE_PROP_INHERIT ? true : false;
    bool filter =  prop & LV_STYLE_PROP_FILTER ? true : false;
    filter_g = is_filter(prop);      //Added
    inherit_g = is_filter(prop);   //Added
     
    ...

```

The render time of the first frame was:
- original version: 39 ms
- `is_filter` inlined: 40 ms
- `is_filter` not inlined: 42ms

Note that the `AND` code was still there when `is_filter` was added so the times are slightly better.

So I think we can say that the performance loss is negligable. 

Are you interested in implementing it?

egonbeermat, 2021-07-11:
> The render time of the first frame was:
> 
> * original version: 39 ms
> * `is_filter` inlined: 40 ms
> * `is_filter` not inlined: 42ms
> 
> Note that the `AND` code was still there when `is_filter` was added so the times are slightly better.
> 
> So I think we can say that the performance loss is negligable.

I'd be interested in seeing that timing in micros to see true accuracy, but I wouldn't say that a performance loss of 3%-9% is negligible, especially as the memory gain is so small?! I think its safe to say most LVGL users would like to see render performance increased, not decreased, and as LVGL code is so well refined, gains would come in small increments rather than big wins. This type of change, considered "negligible", done 3 or 4 times, would lead to quite a performance degradation; it's the type of change you'd want to undo if you're trying to improve performance, at the cost of a few bytes.

kisvegabor, 2021-07-15:
This agreement also sounds reasonable. 

Note that these tests don't show real-life use cases. We could make a decision if someone implemented this update and we saw if there are any regressions.

closed 2021-08-07 by lvgl-bot