lvgl/lvgl #2356: A object created without a style produces garbage.

LiveSparks, 2021-07-07:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I am running LVGL on the pi pico with the RP2040 SDK. Most of it is working fine except any default objects draw garbage on the screen(see screenshots).
adding the following line solves the issue.
`lv_obj_set_style_bg_color(rect, lv_color_hex(0xff0000), 0);`
Adding a style with `lv_obj_add_style();` does not help.

### To Reproduce

The issue doesn't come up on the simulator. My hardware is Raspberry Pi Pico + ILI9486 display in 8-bit parallel mode.
This same code worked fine when running Arduino. The issue comes up only with the SDK. I had to remove the examples and tests folder from the LVGL repo to allow the code to compile. 
I am running LVGL v8.0.1
The full project can be downloaded [here](https://drive.google.com/file/d/1ZXJHjd1NPTQvcwWqaRQhFISoqPMezKr-/view?usp=sharing).
You can follow the instructions [here](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) to build it.

```c
static lv_obj_t* bad_obj = lv_obj_create(lv_scr_act());
lv_obj_set_size(bad_obj, 21, 300);
lv_obj_align(bad_obj, LV_ALIGN_CENTER, 22, 0);

static lv_obj_t* ok_obj1 = lv_obj_create(lv_scr_act());
lv_obj_set_size(ok_obj1, 21, 300);
lv_obj_set_style_bg_color(ok_obj1, lv_color_hex(0xff0000), 0);
lv_obj_align(ok_obj1, LV_ALIGN_CENTER, 0, 0);

static lv_obj_t* ok_obj2 = lv_obj_create(lv_scr_act());
lv_obj_set_size(ok_obj2, 100, 100);
lv_obj_set_style_bg_color(ok_obj2, lv_color_hex(0xff00ff), 0);
lv_obj_align(ok_obj2, LV_ALIGN_CENTER, 100, 100);
```
If I remove the first object(first 3 lines of code), the image looks good. 
This issue doesn't exist with images or labels.

### Expected behavior

An object should have been created with the default theme.

### Screenshots or video

With the "bad_obj"
[image]

Without the "bad_obj"
[image]

LiveSparks, 2021-07-07:
I'll be happy to run any tests that might give further insight. This exact code running on this exact hardware worked fine when running in the Arduino environment instead of the SDK.

embeddedt, 2021-07-07:
> This exact code running on this exact hardware worked fine when running in the Arduino environment instead of the SDK.

This suggests that there is a platform-specific issue (possibly memory corruption). Are your stack sizes similar, etc.?

LiveSparks, 2021-07-07:
The lv_conf.h file is exactly the same. The buffer allocation to lvgl is also exactly the same. I don't know what else I can check. Any suggestions?

LiveSparks, 2021-07-07:
> I had to remove the examples and tests folder from the LVGL repo to allow the code to compile.

I have modified my CMakeLists.txt file by adding `add_subdirectory("src/lvgl")` and removing the following lines:
```cmake
file(GLOB_RECURSE INCLUDES "src/lvgl/*.h"  "src/*.h" )
file(GLOB_RECURSE SOURCES  "src/lvgl/*.c" )
list(FILTER INCLUDES EXCLUDE REGEX "src/lvgl/examples/*.h")
list(FILTER SOURCES EXCLUDE REGEX "src/lvgl/examples/*.c")
add_library(lvgl STATIC
  ${SOURCES}
  ${INCLUDES}
)
```
Now LVGL is compiling without issues and I don't need to remove the examples and tests folders. I swear I had tried this before and it wasn't working. 🤷‍♀️
Regardless, this hasn't solved the issue.
I also changed to the latest branch 'v8.1.0-dev' with no difference.

I tried digging into the platform files that Arduino is using to find some differences in the configurations. But as far as I can figure Arduino is not using the cmake build system. There are a lot of mbed files there which I have no experience with.

This problem is really stumping me, as far as I can see, everything else with LVGL is working fine. The only issue that I can figure is with styles that are somehow messing up the buffer. If I use an animation that brings in the "bad object" from off-screen then only the space that the object is traveling looks corrupted instead of the whole screen.

LiveSparks, 2021-07-07:
VERY WEIRD UPDATE: Using dark theme partially solves the issue.
Since the issue was clearly with styles, I decided to try out the dark theme not expecting anything different. Turns out it does solve the problem (partially).
After setting `# define LV_THEME_DEFAULT_DARK     1` in lv_conf.h, the code I provided above works without issue. After this I decided to try applying a style. This again corrupted the image. 
I have narrowed down the issue to `lv_style_set_outline_color(&stp_style, lv_color_white());`
Setting an outline color corrupts the image buffer somehow. I have not tried all the style settings but the issue might be present with other settings as well.
Here's the style I am using:
```c
static lv_style_t stp_style;
lv_style_init(&stp_style);
lv_style_set_bg_color(&stp_style, lv_color_hex(0xff0000));
lv_style_set_border_width(&stp_style, 0);
lv_style_set_radius(&stp_style, 0);
lv_style_set_outline_width(&stp_style, 5);
lv_style_set_outline_color(&stp_style, lv_color_hex(0xffffff));
```
Removing the last line gives a good image.
I haven't yet looked into the LVGL source code to see what the difference with light and dark theme is that might be causing the issue.

embeddedt, 2021-07-07:
Any particular reason why you are declaring the objects as `static`? In C this is illegal, and I'm not sure what the C++ semantics are regarding that.

LiveSparks, 2021-07-07:
I am using C++. No particular reason for the static, It was just one of the things that I tried. I have just now removed `static` from the objects. The style is still declared as static as per the LVGL documentation. No change in behavior.

LiveSparks, 2021-07-07:
### Solved

> In C this is illegal

I thought I might as well try switching completely to C. Thank you @embeddedt for the inspiration.
Switching to C from C++ has solved the problem. All I did to fix the issue is rename the files from `*.cpp` to `*.c`. This only works in the test program that I have setup for this particular issue.

I don't understand why this is the case since both Arduino and the simulator use C++.
Further confusing the issue is the matter that switching to dark theme also partially solves the issue.

This solution doesn't help me much because I am using C++ features in my main application. Any suggestions on what I might be doing wrong while using C++ would be helpful.

embeddedt, 2021-07-07:
Can you send the current version of the C++ file? There must be something which the C++ compiler treats differently compared to C.

LiveSparks, 2021-07-07:
Github won't let me attach .zip for some reason. [google drive](https://drive.google.com/file/d/1jLpjBw3QmbBKHYKxbyJR1M_FhB5eWdyS/view?usp=sharing)
The main file is `src/main.cpp`

embeddedt, 2021-07-07:
I've tried your code in a C++ file on my STM32 board and it worked fine.

I suspect that your C++ library might be corrupting something/changing some behavior when it runs its initialization process. The simplest method I can think of to test this theory is to move all LVGL code to a function in a separate C file, and call that function from the C++ file with `main` in it.

LiveSparks, 2021-07-07:
Good idea. I'll try that as soon as I wake up tomorrow.

LiveSparks, 2021-07-08:
> The simplest method I can think of to test this theory is to move all LVGL code to a function in a separate C file, and call that function from the C++ file with main in it.

This has solved the issue. I'll have to make a few changes to my main application to be compatible with this approach but it should work out in the end. 
I checked, and my system is currently using `GCC 9.2.1-arm-none-eabi`.
I'll try upgrading this kit to see if that solves the issue and update this thread accordingly. 
Thank You @embeddedt for all your help.

closed 2021-07-08 by LiveSparks