lvgl/lvgl #2357: LV_ALIGN_OUT_* aligns in wrong place

DolphinChips, 2021-07-07:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When calling `lv_obj_align()` with any of LV_ALIGN_OUT_*, object ends up being aligned somewhere around top left corner inside the parent object.

### To Reproduce
```c
lv_obj_t* obj1 = lv_obj_create(lv_scr_act());
lv_obj_set_size(obj1, 200, 200);
lv_obj_align(obj1, LV_ALIGN_CENTER, 0, 0);
lv_obj_t* obj2 = lv_obj_create(obj1);
lv_obj_set_size(obj2, 50, 50);
lv_obj_align(obj2, LV_ALIGN_OUT_RIGHT_MID, 0, 0);
```

### Expected behavior
Object gets aligned outside of parent object

### Screenshots or video
[image]

embeddedt, 2021-07-07:
Not 100% sure if this is necessary, but try running `lv_obj_update_layout(obj1)` and `lv_obj_update_layout(obj2)` first.

DolphinChips, 2021-07-07:
I've inserted those calls right after my original code and still get the same result

kisvegabor, 2021-07-07:
`lv_obj_align` can't be used with ` LV_ALIGN_OUT_*`.  See https://github.com/lvgl/lvgl/blob/c751c11a87019f933d8596b7342ba37ce94a977f/src/core/lv_obj_pos.h#L151-L156

Try `lv_obj_align_to(obj2, obj1, LV_ALIGN_OUT_RIGHT_MID, 0, 0);`

DolphinChips, 2021-07-07:
Turns out this was an issue :\

Above code works perfectly with `lv_obj_align_to()` when parent of obj2 was changed from obj1 to `lv_scr_act()` and changing parent is seems like was I forgot to do while debugging this behavior. Thanks for noticing this!

closed 2021-07-07 by DolphinChips