lvgl/lvgl #2359: Focused first button of list widget not painted as focused at initial state

ercanersoy, 2021-07-08:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I use list widget in my code. Initial state is focused to first element of the list element.
But, in the initial state, the focused first element is not painted as focused.

### To Reproduce

My code part is given below.
```c
lv_obj_t *menu;

lv_obj_t *button_1;
lv_obj_t *button_2;
lv_obj_t *button_3;
lv_obj_t *button_4;
lv_obj_t *button_5;

lv_group_t *group;

menu = lv_list_create(lv_scr_act());

lv_list_add_text(menu, "Menu");

button_1 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "1");
button_2 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "2");
button_3 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "3");
button_4 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "4");
button_5 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "5");

group = lv_group_create();

lv_group_add_obj(group, button_1);
lv_group_add_obj(group, button_2);
lv_group_add_obj(group, button_3);
lv_group_add_obj(group, button_4);
lv_group_add_obj(group, button_5);

lv_group_set_default(group);

lv_group_focus_obj(button_1);
```

### Expected behavior

In the initial state, the focused first element should painted as focused.

### Screenshots or video

[image]

kisvegabor, 2021-07-08:
Thanks, I fixed it.

However, you should assign the group  to an input device.  See [here](https://docs.lvgl.io/master/overview/indev.html#groups).

closed 2021-07-08 by kisvegabor

ercanersoy, 2021-07-08:
Thanks a lot.

embeddedt, 2021-07-15:
@kisvegabor One of the focus-related changes seems to be causing this on the [keyboard example](https://docs.lvgl.io/master/widgets/extra/keyboard.html):

[image]

kisvegabor, 2021-07-15:
The problem is the keyboard is focused or the focus looks ugly on this small size? :slightly_smiling_face:

embeddedt, 2021-07-15:
There are three issues I see (listed in order of importance):

1. The focus outline is visible even when an encoder/keypad has not been used. This is confusing when a touchscreen/mouse is the main form of input.
1. The focus outline for the button seems to be drawn underneath the button beside it. In CSS I usually get around this problem by using `z-index` to put the focused element above the other ones, but I don't think we have that feature here.
1. IMO the keyboard shouldn't need a focus outline drawn if one of the keys has an outline.

kisvegabor, 2021-07-15:
> 1. The focus outline is visible even when an encoder/keypad has not been used. This is confusing when a touchscreen/mouse is the main form of input.

The demos in the docs do have keypad and encoder support. That's why the focused state is visible by default. 

> 2. The focus outline for the button seems to be drawn underneath the button beside it. In CSS I usually get around this problem by using z-index to put the focused element above the other ones, but I don't think we have that feature here.

As z-index is not supported this way, I'm thinking about recoloring the button instead of adding an outline.

embeddedt, 2021-07-15:
> The demos in the docs do have keypad and encoder support. That's why the focused state is visible by default.

True. They do seem to disappear after I click the first time, so I think it's reasonable in the current state. The outline rendering should probably be adjusted, though.

kisvegabor, 2021-07-16:
Fixed, now looks like this:
[image]

embeddedt, 2021-07-16:
That looks much better!

commit f03d4b8c (2021-07-08): fix(obj) detecting which indev sent LV_EVENT_FOCUS

fixes #2359