lvgl/lvgl #2368: the childs obj is not expected if the padding  of parent obj changes

guoweilkd, 2021-07-13:
### Describe the bug

 the start coordinates of the childs obj do not corresponding change when the parent obj's pad changes dynamically.
the reason is  not invoke `lv_obj_mark_layout_as_dirty(child);` when set  the padding of parent obj .

### To Reproduce

the follow is the test code:

```c
static void pad_test_timer(struct _lv_timer_t *t)
{
    static uint8_t pad = 0;
    lv_obj_t* obj = (lv_obj_t*)t->user_data;
    lv_obj_set_style_pad_left(obj, pad, LV_PART_MAIN);
    pad += 5;

    /* only open the comments, child objs coordinates will be correct */
    // uint32_t child_cnt = lv_obj_get_child_cnt(obj);
    // for(uint32_t i = 0; i < child_cnt; i++) {
    //     lv_obj_t * child = lv_obj_get_child(obj, i);
    //     lv_obj_mark_layout_as_dirty(child);
    // }
}

void pad_test_main(void)
{
    lv_obj_t * par = lv_obj_create(lv_scr_act());
    lv_obj_set_size(par, 200, 200);
    lv_obj_center(par);
    lv_obj_set_style_pad_all(par, 0, LV_PART_MAIN);

    lv_obj_t * btn2 = lv_btn_create(par);
    lv_obj_align(btn2, LV_ALIGN_TOP_LEFT, 0, 0);
    lv_obj_set_size(btn2, 60, 40);

    lv_timer_create(pad_test_timer, 1000, par);
}
```

### Expected behavior

### Screenshots or video

the follow is my solution idea:
```c
// in lv_style.h
#define LV_STYLE_PROP_CHILD_LAYOUT_REFR              (1 << 15)

//in lv_obj_style.c
lv_obj_refresh_style(lv_obj_t * obj, lv_style_selector_t selector, lv_style_prop_t prop)
{
  ....
  if (prop & LV_STYLE_PROP_CHILD_LAYOUT_REFR) {
        uint32_t child_cnt = lv_obj_get_child_cnt(obj);
        for(uint32_t i = 0; i < child_cnt; i++) {
            lv_obj_t * child = lv_obj_get_child(obj, i);
            lv_obj_mark_layout_as_dirty(child);
        }
    }
}
```

kisvegabor, 2021-07-13:
Hi,

Thanks for the report!

As `LV_EVENT_STLYE_CHANGE` is sent on padding change from `lv_obj_refresh_style` I'd rather add your fix to a `LV_EVENT_STLYE_CHANGE` event in `lv_obj_event()`. What do yout think?

guoweilkd, 2021-07-14:
good idea

kisvegabor, 2021-07-15:
I've pushed a fix.

commit 57cf6610 (2021-07-15): fix(obj style) fix children reposition if the parent's padding changes.

fixes #2368
closed 2021-07-15 by kisvegabor