lvgl/lvgl #2369: Offload Rasterization to custom GPU

AfflatusX, 2021-07-13:
### Introduce the problem
Today we are using LVGL to drive our MCU graphic needs - really appreciate the amazing library as well as the supportive community! We foresee that going forward, we will have our own GPU unit that can rasterize/render things much more efficiently than CPU. We would like to continue to utilize LVGL but offload the draw/rasterization step to such hardware.

### Examples and cases
Say if we have a standard UI list. Developers would be still using standard LVGL API to define it - but instead of calling various draw_xxx calls, right before LVGL start to actually rasterize the content, we send the primitives through a different API to draw.

### Suggested solution
I'm more looking for guidance here - is this something reasonable to achieve through LVGL? We're not looking for LVGL community to implement/support it, but more looking for suggestions. If this is not feasible, we would like to understand why, and if it's feasible, some suggestion on a general direction would be great!

kisvegabor, 2021-07-13:
Hi,

We have an [open issue](https://github.com/lvgl/lvgl/issues/2058) about adding support for "higher level" GPU APIs. We have taken OpenGL (ES) or SDL as examples but your GPU seems to fall into this category as well. 

In short, the idea is to add support for common GPUs right into LVGL but once we are there we can add hooks for external GPUs too.

To make the implementation cleaner, some restructuring in the draw functions would be required. I can do that restructuring and take care of the API part but it'd be great if someone could add support for a GPU as a proof of concept. Not all LVGL draw features are required to be reimplemented. If something is not supported it can fall back to SW rendering. 

How does it sound to you?

AfflatusX, 2021-07-13:
> Hi,
>
> We have an [open issue](https://github.com/lvgl/lvgl/issues/2058) about adding support for "higher level" GPU APIs. We have taken OpenGL (ES) or SDL as examples but your GPU seems to fall into [...]

That sounds great to me. Let me circle back after chatting with the team see how can we best collaborate on this.

kisvegabor, 2021-07-13:
>  Let me circle back after chatting with the team see how can we best collaborate on this.

Great, looking forward to it :slightly_smiling_face:

AfflatusX, 2021-07-21:
@kisvegabor they'll allocate a dedicated engineer to work with you on this - I'll share more details once the logistics are ironed out.

kisvegabor, 2021-07-23:
Amazing! :slightly_smiling_face:

kisvegabor, 2021-08-09:
@AfflatusX Are there any news on this topic?

closed 2021-09-01 by lvgl-bot