lvgl/lvgl #2370: LVGL slows down with more widgets, even if hidden

fredbar, 2021-07-14:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version 
- [x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2021-07-14:
Some code to see the issue would be great even if they are only containers. 

What is sluggish? The rendering or the reaction to button presses?

fredbar, 2021-07-14:
Hi.

RIght, i'll send code that got tested on my stm32 in a moment. 

Both get slower, by the way.

fredbar, 2021-07-14:
ok, please check the code below. There is a define for the count of panes added. 
I also made a video with two pane count; 1 and 5. That should make it more clear.
Event code might not work, i had to embed the code in my program and it was not happy with the interface change, causing it to hang without me having a chance to test that. So this part is pretty untested.

video:
https://youtu.be/O-KyST3Ox4M

code:
```
#define LVGLISSUE_1_PANECOUNT 5

lv_obj_t *painPanes[LVGLISSUE_1_PANECOUNT];

void lvglIssue_1_btnCB(lv_obj_t *obj, lv_event_t event) {
// on a button press, go to next pane.
	static uint8_t pageIndex = 0;
	lv_obj_set_hidden(painPanes[pageIndex], true);
	pageIndex++;
	pageIndex %= LVGLISSUE_1_PANECOUNT;
	lv_obj_set_hidden(painPanes[pageIndex], false);
}

void lvglIssue_1_createStyles() {
	lv_style_init(&buttonStyle);
	lv_style_init(&bgStyle);

	lv_style_set_text_color(&buttonStyle, LV_STATE_DEFAULT,
			lv_color_make(0x00, 0x00, 0x00));
	lv_style_set_image_recolor(&buttonStyle, LV_STATE_DEFAULT,
	LV_COLOR_WHITE);
	lv_style_set_image_recolor_opa(&buttonStyle, LV_STATE_DEFAULT, 255);
	lv_style_set_image_opa(&buttonStyle, LV_STATE_DEFAULT, 255);
	lv_style_set_bg_opa(&buttonStyle, LV_STATE_DEFAULT, LV_OPA_COVER);
	lv_style_set_bg_color(&buttonStyle, LV_STATE_CHECKED, LV_COLOR_BLUE);
	lv_style_set_bg_color(&buttonStyle, LV_STATE_DEFAULT, LV_COLOR_SILVER);
	lv_style_set_radius(&buttonStyle, LV_STATE_DEFAULT, 5);
	lv_style_set_clip_corner(&buttonStyle, LV_STATE_DEFAULT, true);
	lv_style_set_border_width(&buttonStyle, LV_STATE_DEFAULT, 2);
	lv_style_set_border_color(&buttonStyle, LV_STATE_DEFAULT,
	LV_COLOR_WHITE);
	lv_style_set_bg_color(&bgStyle, LV_STATE_DEFAULT, bgBlue);
}

void lvglIssue_1_createwidgets() {
	lvglIssue_1_createStyles();
	for (int index = 0; index < LVGLISSUE_1_PANECOUNT; index++) {
		painPanes[index] = lv_obj_create(lv_scr_act(), NULL);

		lv_obj_t *pane = painPanes[index];
		lv_obj_set_pos(pane, 0, 0);
		lv_obj_set_size(pane, 800, 480);				//yeah...:)
		lv_obj_add_style(pane, LV_OBJ_PART_MAIN, &bgStyle);
		lv_obj_set_hidden(pane, true);
		for (int yPos = 10; yPos < 480 - 60; yPos += 60) {
			for (int xPos = 10; xPos < 800 - 60; xPos += 60) {
				lv_obj_t *btn = lv_btn_create(pane, NULL);
				lv_obj_set_pos(btn, xPos, yPos);
				lv_obj_set_size(btn, 50, 50);				//yeah...:)
				lv_obj_add_style(btn, LV_BTN_PART_MAIN, &buttonStyle);
				lv_obj_set_event_cb(btn, lvglIssue_1_btnCB);
			}
		}
	}
	lv_obj_set_hidden(painPanes[0], false);
  lv_scr_load(painPanes[0]);
}

```

fredbar, 2021-07-14:
I also tried moving the hidden panes far out of the display in case there was a bounding box elimination but that did not help.

kisvegabor, 2021-07-16:
I've checked the source code of v7 and haven't found anything obvious at first look.

But there is something strange in your code. You use `lv_scr_load(painPanes[0]);` but `painPanes[0]` is not screen as it was created with `painPanes[index] = lv_obj_create(lv_scr_act(), NULL);`. To create it as a screen use `painPanes[index] = lv_obj_create(NULL, NULL);`. But in this case you don't need to hide the panes as the screens are "separated" from eachother very well.

fredbar, 2021-07-16:
Yes, i somewhat fumbled here. In my application, i have a single pane that then references all other panes.
Sorry for the misleading representation.

However, in the end that makes no difference. There is still a slowdown depending on the count of created buttons only, not on those that are displayed.

fredbar, 2021-07-16:
I thought about only creating panes when i have to display them and dispose them when i swap, but then i have an other problem with the app freezing two seconds when i create the keyboard. Not really user friendly.
And also i worried about memory fragmentation if going that route. Not sure how LVGL reacts  on that point.

kisvegabor, 2021-07-16:
> However, in the end that makes no difference. There is still a slowdown depending on the count of created buttons only, not on those that are displayed.

You have fixed the screen creation issue, right? 

If you are using STM32 you can use SWV as a profiler to see which functions take so much time. Can you give it a try?

Memory fragmentation is usually not an issue in normal use cases.  So you can create the panes on demand.

fredbar, 2021-07-16:
Screen creation is okay. I just did not do exactly what i had in my original program.
Anyway, i did some tracing.
Here. are the CSV of the most important part of traces: (and yes, i made no mistake, they are not swapped)
5 panes
```
"lv_obj_get_hidden()";"23.29%";"5017";"0x8031e49";"0x78"
"HAL_GetTick()";"10.22%";"2202";"0x80054ed";"0x18"
"obj_valid_child()";"9.29%";"2000";"0x80347cd";"0x58"
"HAL_Delay()";"6.97%";"1502";"0x8005505";"0x48"
"_lv_ll_get_next()";"5.05%";"1087";"0x8047d11";"0x34"
"_lv_ll_get_head()";"4.59%";"989";"0x8047cc5";"0x26"
"_lv_memset_00()";"4.23%";"911";"0x8048bfd";"0x102"
"get_style_prop()";"4.04%";"870";"0x8037dd5";"0x3e"
"get_prop_size()";"3.92%";"845";"0x8037e53";"0x54"
"get_property_index()";"2.98%";"641";"0x8037c25";"0xe0"
"ent_get_next()";"2.88%";"620";"0x8048e15";"0x58"
"get_next_prop_index()";"2.58%";"555";"0x8037ea7";"0x22"
"lv_obj_get_type()";"2.01%";"432";"0x80323ad";"0xf0"
"get_style_prop_id()";"1.92%";"414";"0x8037e13";"0x1e"
"lv_debug_check_obj_valid()";"1.54%";"331";"0x80331f3";"0x6e"
"lv_obj_handle_get_type_signal()";"1.22%";"263";"0x8032659";"0x44"
"lv_debug_check_obj_type()";"1.16%";"250";"0x8033173";"0x80"
"ent_alloc()";"1.06%";"229";"0x8048e6d";"0x4e"
"I2C_WaitOnFlagUntilTimeout()";"1.02%";"220";"0x800acfb";"0x80"
"strcmp()";"0.78%";"167";"0x8000241";"0x14"
"lv_mem_alloc()";"0.78%";"167";"0x80480c5";"0x94"
"lv_mem_test()";"0.75%";"162";"0x804834d";"0x60"
"_lv_obj_get_style_int()";"0.60%";"129";"0x80314f1";"0x57a"
"lv_cont_signal()";"0.57%";"122";"0x8053995";"0xf8"
"lv_debug_check_null()";"0.56%";"121";"0x80477e3";"0x20"
"lv_obj_signal()";"0.52%";"112";"0x8033775";"0x174"
"_lv_style_list_get_int()";"0.47%";"102";"0x8037741";"0x124"
"lv_disp_get_next()";"0.35%";"75";"0x80460a5";"0x30"
"lv_btn_signal()";"0.33%";"71";"0x80533b5";"0x150"
```

1 pane
```
"lv_obj_get_hidden()";"68.74%";"10372";"0x8031e29";"0x78"
"HAL_GetTick()";"2.60%";"392";"0x80054d5";"0x18"
"_lv_memset_00()";"2.47%";"373";"0x8048bdd";"0x102"
"get_style_prop()";"2.39%";"361";"0x8037db5";"0x3e"
"obj_valid_child()";"1.96%";"295";"0x80347ad";"0x58"
"get_prop_size()";"1.88%";"283";"0x8037e33";"0x54"
"get_property_index()";"1.75%";"264";"0x8037c05";"0xe0"
"HAL_Delay()";"1.70%";"256";"0x80054ed";"0x48"
"get_next_prop_index()";"1.50%";"226";"0x8037e87";"0x22"
"_lv_ll_get_head()";"1.27%";"191";"0x8047ca5";"0x26"
"lv_obj_get_type()";"1.23%";"186";"0x803238d";"0xf0"
"_lv_ll_get_next()";"1.03%";"156";"0x8047cf1";"0x34"
"lv_debug_check_obj_valid()";"1.01%";"152";"0x80331d3";"0x6e"
"get_style_prop_id()";"0.99%";"149";"0x8037df3";"0x1e"
"memcpy()";"0.76%";"115";"0x8056d7d";"0x1c"
"lv_obj_handle_get_type_signal()";"0.71%";"107";"0x8032639";"0x44"
"ent_get_next()";"0.68%";"103";"0x8048df5";"0x58"
"lv_debug_check_obj_type()";"0.54%";"82";"0x8033153";"0x80"
"strcmp()";"0.46%";"70";"0x8000241";"0x14"
"_lv_obj_get_style_int()";"0.40%";"61";"0x80314d1";"0x57a"

```

kisvegabor, 2021-07-19:
Ah, I think I see the problem. Be sure the `LV_USE_DEBUG` is `0` is `lv_conf.h`.

fredbar, 2021-07-19:
Of course, that was it.. What an idiot i am.
Speed, not only the one of this use case but  the speed in general, increased tenfolds.
Thanks a huge bunch, all is nice and shiny. now.

kisvegabor, 2021-07-19:
I made this mistake several times as well. :slightly_smiling_face: 
In v8 I added warnings if slow asserts are enabled so I hope it will happen less often.

Anyway, glad to hear it's fast now :tada:

closed 2021-07-19 by kisvegabor