lvgl/lvgl #2371: Focused first button of list widget not painted as focused at initial state with mouse

ercanersoy, 2021-07-14:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I use list widget in my code. Initial state is focused to first element of the list element.
But, in the initial state with mouse, the focused first element is not painted as focused.

The bug is simlar as #2359. The previous bug is resolved in commit f03d4b8.
I use commit f03d4b8 of LVGL.

### To Reproduce

My code is given below.
```c
//Keyboard related codes

lv_indev_drv_t keyboard_driver;
lv_indev_t *keyboard_device;

lv_indev_drv_init(&keyboard_driver);

keyboard_driver.type = LV_INDEV_TYPE_KEYPAD;
keyboard_driver.read_cb = read_keyboard;

keyboard_device = lv_indev_drv_register(&keyboard_driver);

//Mouse related code

lv_indev_drv_t mouse_driver;
lv_indev_t *mouse_device;

lv_indev_drv_init(&mouse_driver);

mouse_driver.type = LV_INDEV_TYPE_POINTER;
mouse_driver.read_cb = read_mouse;

mouse_device = lv_indev_drv_register(&mouse_driver);

// Menu related code

lv_obj_t *menu;

lv_obj_t *button_1;
lv_obj_t *button_2;
lv_obj_t *button_3;
lv_obj_t *button_4;
lv_obj_t *button_5;

lv_group_t *group;

menu = lv_list_create(lv_scr_act());

lv_list_add_text(menu, "Menu");

button_1 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 1");
button_2 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 2");
button_3 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 3");
button_4 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 4");
button_5 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 5");

// Menu group related code

lv_group_t *menu_group;

group = lv_group_create();

lv_group_add_obj(group, button_1);
lv_group_add_obj(group, button_2);
lv_group_add_obj(group, button_3);
lv_group_add_obj(group, button_4);
lv_group_add_obj(group, button_5);

lv_group_set_default(group);

lv_group_focus_obj(button_1);

// Cursor related code

lv_obj_t *cursor;

cursor = lv_img_create(lv_scr_act());

lv_img_set_src(cursor, &mouse_cursor);

lv_indev_set_cursor(mouse_device, cursor);
```

### Expected behavior

In the initial state, the focused first element should painted as focused.

kisvegabor, 2021-07-15:
Try this when you create the group:
```c
  group = lv_group_create();
  lv_indev_set_group(keyboard_device, group);
```

When you manually focus a button LVGL checks if there is an encoder or keyboard indev for that group. If so it adds the `LV_STATE_FOCUS_KEY` state too. And in the default theme focused style are added only for `LV_STATE_FOCUS_KEY`.

ercanersoy, 2021-07-15:
My original code can not copy-paste here. Because, my original code is properitary.

I have added `lv_indev_set_group(keyboard_device, group);` line before`lv_group_set_default(group);` line as the order is in my original code.

```
//Keyboard related codes

lv_indev_drv_t keyboard_driver;
lv_indev_t *keyboard_device;

lv_indev_drv_init(&keyboard_driver);

keyboard_driver.type = LV_INDEV_TYPE_KEYPAD;
keyboard_driver.read_cb = read_keyboard;

keyboard_device = lv_indev_drv_register(&keyboard_driver);

//Mouse related code

lv_indev_drv_t mouse_driver;
lv_indev_t *mouse_device;

lv_indev_drv_init(&mouse_driver);

mouse_driver.type = LV_INDEV_TYPE_POINTER;
mouse_driver.read_cb = read_mouse;

mouse_device = lv_indev_drv_register(&mouse_driver);

// Menu related code

lv_obj_t *menu;

lv_obj_t *button_1;
lv_obj_t *button_2;
lv_obj_t *button_3;
lv_obj_t *button_4;
lv_obj_t *button_5;

lv_group_t *group;

menu = lv_list_create(lv_scr_act());

lv_list_add_text(menu, "Menu");

button_1 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 1");
button_2 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 2");
button_3 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 3");
button_4 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 4");
button_5 = lv_list_add_btn(menu, LV_SYMBOL_DIRECTORY, "Menu Item 5");

// Menu group related code

lv_group_t *menu_group;

group = lv_group_create();

lv_group_add_obj(group, button_1);
lv_group_add_obj(group, button_2);
lv_group_add_obj(group, button_3);
lv_group_add_obj(group, button_4);
lv_group_add_obj(group, button_5);

lv_group_focus_obj(button_1);

lv_group_set_default(group);

lv_indev_set_group(keyboard_device, group);

// Cursor related code

lv_obj_t *cursor;

cursor = lv_img_create(lv_scr_act());

lv_img_set_src(cursor, &mouse_cursor);

lv_indev_set_cursor(mouse_device, cursor);
```

kisvegabor, 2021-07-15:
`lv_indev_set_group` needs to be before `lv_group_focus_obj` else LVGL won't see that you have a keyboard indev and won't add `LV_STATE_FOCUS_KEY`.

ercanersoy, 2021-07-31:
> `lv_indev_set_group` needs to be before `lv_group_focus_obj` else LVGL won't see that you have a keyboard indev and won't add `LV_STATE_FOCUS_KEY`.

It works. Thanks a lot.

kisvegabor, 2021-08-02:
Great, thanks for the feedback.

closed 2021-08-02 by kisvegabor