lvgl/lvgl #2375: Add functions to get span children once added to spangroup

ahoenerBE, 2021-07-16:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Normally if you need an object's child, the method to use would be `lv_obj_get_child`, but it doesn't seem like (at least as far as I can tell)`lv_spangroup` has any sort of method to get an `lv_span_t` out of its child list.
This can be worked around by getting the actual spangroup on the screen, casting it to `lv_spangroup_t` and manually going through `child_ll` but this isn't a very future-proof method if the underlying code decides to change.

I've checked the latest docs and they don't seem to mention any way of getting a child back from the group.
And at least from what I can see of lv_span.c/h, no such `get_child_span` function exists.

The span structs are still there, but there's no API friendly way to actually get them back, unlike nested objects.
This makes editing spans later much harder.
[image]

### To Reproduce

Create a spangroup, add a child. If you access the raw lv_spangroup_t struct, you can see the child waiting inside the `child_ll` list, but there's no way to get it out without accessing raw fields.

```c
    lv_obj_t* spangroup = lv_spangroup_create(lv_scr_act());
    lv_span_t* child = lv_spangroup_new_span(spangroup);
    lv_span_set_text(child, "Hello World");

    auto* spangroup_raw = (lv_spangroup_t*) spangroup;
    lv_span_t* first_child = NULL;
    first_child = (lv_span_t*) spangroup_raw->child_ll.head;

    LV_ASSERT(first_child == child);

    lv_obj_del(spangroup);

```

The addresses are identical and the assert check passes

[image]

### Expected behavior

Spangroup should provide a function `lv_span_t* lv_spangroup_get_child(lv_obj_t*, uint)` (or similar).
I could submit a PR if that helps?

### Screenshots or video

(see screenshots above)

guoweilkd, 2021-07-16:
I agree. we also should provide `lv_spangroup_get_child_cnt()`
But I think it's more reasonable to use this name:`lv_spangroup_get_span`,lv_spangroup_get_span_cnt() . because spangroup  also is a obj, There is confusion between `lv_obj_get_child()` and `lv_spangroup_get_child_cnt()`
@kisvegabor , what do you think.

kisvegabor, 2021-07-16:
I agree with both the feature and the names recommended by @guoweilkd (`lv_spangroup_get_child` and `lv_spangroup_get_child_cnt`).

It might be useful to allow using negative index in `lv_spangroup_get_child_cnt` to index from the back. (Similarly to `lv_obj_get_child`)

ahoenerBE, 2021-07-16:
Would you be accepting PRs for it? I already made the methods for my own project with the raw pointers like I put in my issue.

PR #2379 (2021-07-19): feat(span) Add missing spangroup functions 
### Description of the feature or fix

Adds 2 functions: 
* `lv_spangroup_get_child` -- Gets a span stored in the spangroup linkedlist, returns span pointer
* `lv_spangroup_get_child_cnt` -- Gets the total number of spans in the spangroup, returns uint

Tested in LVGL sim using lv_example_span. I added variables for every span added (e.g. `span1`, `span2`, etc) and checked them with the following code after they were all added.

[Gist link for my test example (works in LVGL sim)](https://gist.github.com/ahoenerBE/056302e95e6f8daf3fc669cd7ed3b15b)

Up [... cut]
closed 2021-07-19 by embeddedt