lvgl/lvgl #2385: use lv_png display png file failed under uefi

hurryboylqs, 2021-07-20:
when I use lv_png decode png file and display under uefi,the picture is broken,I don't know why,
At first， I suspect the ploblem is in decoding,but when I add some code to display it after decode,there is no problem!
So I'm sure UEFI display driver is OK
I don't know what's going on inside lvgl rendering

lvgl version is 8.0

```c
static lv_res_t decoder_open(lv_img_decoder_t * decoder, lv_img_decoder_dsc_t * dsc)
{

    (void) decoder; /*Unused*/
    uint32_t error;                 /*For the return values of PNG decoder functions*/

    uint8_t * img_data = NULL;

    /*If it's a PNG file...*/
    if(dsc->src_type == LV_IMG_SRC_FILE) {
        const char * fn = dsc->src;

        if(!strcmp(&fn[strlen(fn) - 3], "png")) {              /*Check the extension*/

            /*Load the PNG file into buffer. It's still compressed (not decoded)*/
            unsigned char * png_data;      /*Pointer to the loaded data. Same as the original file just loaded into the RAM*/
            size_t png_data_size;          /*Size of `png_data` in bytes*/

            error = lodepng_load_file(&png_data, &png_data_size, fn);   /*Load the file*/
            if(error) {
                AsciiPrint("error %u: %s\n", error, lodepng_error_text(error));
                return LV_RES_INV;
            }

            /*Decode the PNG image*/
            uint32_t png_width;             /*Will be the width of the decoded image*/
            uint32_t png_height;            /*Will be the width of the decoded image*/

            /*Decode the loaded image in ARGB8888 */
            error = lodepng_decode32(&img_data, &png_width, &png_height, png_data, png_data_size);
            free(png_data); /*Free the loaded file*/
            if(error) {
				AsciiPrint("error %u: %s\n", error, lodepng_error_text(error));
                return LV_RES_INV;
            }

            /*Convert the image to the system's color depth*/
            convert_color_depth(img_data,  png_width * png_height);
            dsc->img_data = img_data;

	    gop->Blt(gop,
				(EFI_GRAPHICS_OUTPUT_BLT_PIXEL *)img_data,
				EfiBltBufferToVideo,
				0,
				0,
				0,
				0,
				png_width,
				png_height,
				0
			);// display good !!!
			Print(L"png width:%d,png height:%d\n", png_width, png_height);
			Print(L"Screen width:%d,Screen height:%d", gop->Mode->Info->HorizontalResolution, gop->Mode->Info->VerticalResolution);

            return LV_RES_OK;     /*The image is fully decoded. Return with its pointer*/
        }
    }
   ...
    return LV_RES_INV;    /*If not returned earlier then it failed*/
}
```
The relatively complete codecode like this:
[... 443 lines cut ...]

	*pos_p = ftell(file_p);

	return LV_FS_RES_OK;
}

static void fs_drv_init(void)
{

	static lv_fs_drv_t drv;
	lv_fs_drv_init(&drv);                     /*Basic initialization*/

	drv.letter = 'S';                         /*An uppercase letter to identify the drive*/
	drv.open_cb = open_cb;                 /*Callback to open a file*/
	drv.close_cb = close_cb;               /*Callback to close a file*/
	drv.read_cb = read_cb;                 /*Callback to read a file*/
	drv.seek_cb = seek_cb;                 /*Callback to seek in a file (Move cursor)*/
	drv.tell_cb = tell_cb;                 /*Callback to tell the cursor position*/

	lv_fs_drv_register(&drv);                 /*Finally register the drive*/
}

#endif /*USE_UEFIDRV*/

```

kisvegabor, 2021-07-20:
Hi,

Some notes and questions:
- In v8 you can use `disp_drv.full_refresh = 1` instead of a rounder to full screen
- `gGraphicsOutput->Blt` is blocking?
- Does it work well if you use an image converted to C array?
- How does the broken image look like? Could you take a photo?

hurryboylqs, 2021-07-21:
It work well if  converted to C array
gGraphicsOutput->Blt Non blocking

Png original：
[image]

Png broken:
[image]

kisvegabor, 2021-07-21:
I'm sure if it's a problem in this case (but certainly can later) that `lv_disp_flush_ready` needs to be called the flushing is ready (i.e. the buffer is free). So you should call `lv_disp_flush_ready` in a ready interrupt. 

On the photo, the horizontal stripes are artifacts from the camera or part of the rendering issue? 

The resolution of the image is the same as the screens resolution?

hurryboylqs, 2021-07-21:
The horizontal stripes are  part of the rendering issue，it shows a part of the picture with stripes on it
The resolution of the image is the same as the screen's resolution 
image size is:1440x900,the screen resolution also 1440x900

I need to load the image from the file, because the conversion to C array is too large. After all, there are other places that can also use the image. I need to solve this problem

kisvegabor, 2021-07-21:
I see. Please try with this image and take a photo. With the current homogenous image, it's difficult to see how the order of the pixels changed.

[image]

hurryboylqs, 2021-07-21:
It seems that the picture only shows 2/ 3 of the original size, and it's still broken

[image]

kisvegabor, 2021-07-22:
2/3 is an interesting number because it's the ratio of 16 bit/24 bit. Are you sure `LV_COLOR_DEPTH` is set correctly?

hurryboylqs, 2021-07-23:
#define LV_COLOR_DEPTH     32

```c
/**
 * @file lv_conf.h
 * Configuration file for v8.0.1
 */

/*
 * COPY THIS FILE AS `lv_conf.h` NEXT TO the `lvgl` FOLDER
 */

#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H
/*clang-format off*/

#include <stdint.h>

#define LV_HOR_RES 1440
#define LV_VER_RES 900

/*====================
   COLOR SETTINGS
 *====================*/

/*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
#define LV_COLOR_DEPTH     32

/*Swap the 2 bytes of RGB565 color. Useful if the display has a 8 bit interface (e.g. SPI)*/
#define LV_COLOR_16_SWAP   0

/*Enable more complex drawing routines to manage screens transparency.
 *Can be used if the UI is above another layer, e.g. an OSD menu or video player.
 *Requires `LV_COLOR_DEPTH = 32` colors and the screen's `bg_opa` should be set to non LV_OPA_COVER value*/
#define LV_COLOR_SCREEN_TRANSP    0

/*Images pixels with this color will not be drawn if they are  chroma keyed)*/
#define LV_COLOR_CHROMA_KEY    lv_color_hex(0x00ff00)         /*pure green*/

/*=========================
   MEMORY SETTINGS
 *=========================*/

/*1: use custom malloc/free, 0: use the built-in `lv_mem_alloc()` and `lv_mem_free()`*/
#define LV_MEM_CUSTOM      1
#if LV_MEM_CUSTOM == 0
/*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
#  define LV_MEM_SIZE    (32U * 1024U)          /*[bytes]*/

/*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
#  define LV_MEM_ADR          0     /*0: unused*/
#else       /*LV_MEM_CUSTOM*/
#  define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#  define LV_MEM_CUSTOM_ALLOC     malloc
#  define LV_MEM_CUSTOM_FREE      free
#  define LV_MEM_CUSTOM_REALLOC   realloc
#endif     /*LV_MEM_CUSTOM*/

/*Use the standard `memcpy` and `memset` instead of LVGL's own functions. (Might or might not be faster).*/
#define LV_MEMCPY_MEMSET_STD    0

/*====================
   HAL SETTINGS
 *====================*/

/*Default display refresh period. LVG will redraw changed ares with this period time*/
#define LV_DISP_DEF_REFR_PERIOD     30      /*[ms]*/

[... 425 lines cut ...]
 * Layouts
 *----------*/

/*A layout similar to Flexbox in CSS.*/
#define LV_USE_FLEX     1

/*A layout similar to Grid in CSS.*/
#define LV_USE_GRID     1

/*==================
* EXAMPLES
*==================*/

/*Enable the examples to be built with the library*/
#define LV_BUILD_EXAMPLES   1

#define LV_PNG_USE_LV_FILESYSTEM 1

/*--END OF LV_CONF_H--*/

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/

```

kisvegabor, 2021-07-23:
The frame buffer's color format is also ARGB8888? Not RGB888 or RGB565?

hurryboylqs, 2021-07-23:
Yes,uefi frame buffer format requirement is BGRA8888

struct like like this:
```c
typedef struct {
  UINT8 Blue;
  UINT8 Green;
  UINT8 Red;
  UINT8 Reserved;
} EFI_GRAPHICS_OUTPUT_BLT_PIXEL;
```
and use the graphics output interface to display the frame buffer,like this:
```c
Status = gGraphicsOutput->Blt(
		gGraphicsOutput,
		(EFI_GRAPHICS_OUTPUT_BLT_PIXEL *)PixelData,
		EfiBltBufferToVideo,
		0,
		0,
		area->x1,
		area->y1,
		Width,
		Height,
		0
	);

```
The function declaration:
```c
typedef
EFI_STATUS
(EFIAPI *EFI_GRAPHICS_OUTPUT_PROTOCOL_BLT) (
  IN  EFI_GRAPHICS_OUTPUT_PROTOCOL            *This,
  IN  EFI_GRAPHICS_OUTPUT_BLT_PIXEL           *BltBuffer,   OPTIONAL
  IN  EFI_GRAPHICS_OUTPUT_BLT_OPERATION       BltOperation,
  IN  UINTN                                   SourceX,
  IN  UINTN                                   SourceY,
  IN  UINTN                                   DestinationX,
  IN  UINTN                                   DestinationY,
  IN  UINTN                                   Width,
  IN  UINTN                                   Height,
  IN  UINTN                                   Delta         OPTIONAL
  );
```

kisvegabor, 2021-07-23:
Okay, that part looks good. The color channels can be mixed but the scrambled image would look different then.

Please replace `lv_png.c` with 
[this file](https://github.com/lvgl/lvgl/files/6868014/lv_png.c.zip). I added some `printf`s. I got this result:
```
w: 1440, h: 900, cf: 2
w: 1440, h: 900, cf: 2
colors: 0xffadb079, 0xffa3ab66, 0xff8c9f3c, 0xff728a1c
```

hurryboylqs, 2021-07-23:
I got this result:
```c
w: 1607, h: 522, cf: 2
w: 1607, h: 522, cf: 2
colors: 0xffadb079, 0xffa3ab66, 0xff8c9f3c, 0xff728a1c
```

kisvegabor, 2021-07-23:
Hm, so LVGL see the width and height incorrectly. Can you debug `decoder_info` in `lv_png,c` how it happens?

hurryboylqs, 2021-07-23:
There is a problem with my fseek implementation. I have changed it now and can display PNG normally!
Thank you very much. You are the most enthusiastic person I have ever met

kisvegabor, 2021-07-23:
Glad to hear it works!

> You are the most enthusiastic person I have ever met

Haha, thanks :smile:

closed 2021-07-23 by kisvegabor