lvgl/lvgl #2388: lv_msgbox_close() can segment fault

trudelpaul, 2021-07-20:
### Perform all steps below and tick them with [x]
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I've been getting very weird behavior when closing a message box.

It faults in the _lv_event_mark_deleted() function.
I do find this part of the code very suspicious:

```
lv_res_t lv_event_send(lv_obj_t * obj, lv_event_code_t event_code, void * param)
{
    if(obj == NULL) return LV_RES_OK;

    LV_ASSERT_OBJ(obj, MY_CLASS);

    lv_event_t e;
    e.target = obj;
    e.current_target = obj;
    e.code = event_code;
    e.user_data = NULL;
    e.param = param;
    e.deleted = 0;

    /*Build a simple linked list from the objects used in the events
     *It's important to know if an this object was deleted by a nested event
     *called from this `event_cb`.*/
    e.prev = event_head;
    event_head = &e;

    /*Send the event*/
    lv_res_t res = event_send_core(&e);

    /*Remove this element from the list*/
    event_head = e.prev;

    return res;
}

```
The event_head is the head pointer of a linked list of lv_event.
However, the e lv_event_t is allocated on the **stack**!

So, when we get to _lv_event_mark_deleted, there's no assurance that the pointers to all lv_event_t pointers in the event_head linked list are still valid.

```
void _lv_event_mark_deleted(lv_obj_t * obj)
{
    lv_event_t * e = event_head;

    while(e) {
        if(e->current_target == obj || e->target == obj) e->deleted = 1;
        e = e->prev;
    }
}

```

This is where I get the seg fault as it traverses the link list.

### To Reproduce
I can reproduce this in my code but I couldn't get it with the lv_example_msgbox_1.c example. I'm pretty much doing the exact same thing as in the example ... nothing fancy.

### Expected behavior

### Screenshots or video

kisvegabor, 2021-07-21:
It'd be great if you sent a code to reproduce it. I haven't seen related issues a long time ago so I assume this part is working.

trudelpaul, 2021-07-21:
My application has many threads but I do believe I've been careful in protecting it correctly with semaphores.
I did try to allocate the lv_event_t e with a malloc() (in lv_event.c) instead of taking the address of e allocated on the stack. **Doing so, I didn't get any faults**. (This was just a test ... without any memory clean up)

The creation of the message box is quite simple but it's a bit difficult to give the whole application context, but it looks something like:
```
    static const char *btns[] = {""};
    helpMessageBox = lv_msgbox_create(getLvScreen(), "Help", "Text", btns, true);
    lv_obj_center(helpMessageBox);

```
This is within a class where helpMessageBox is a lv_obj_t * and getLvScreen() returns the LVGL screen object.
Going forward, I'll probably be not using message boxes in my app since I'll be managing specific screens for help functionalities. Up to now, the lvgl code has been quite solid, this is the first object type I've been having issues with.

To close the message box, I simply press the X button (using SDL on Linux). On my target (DSPG board), it doesn't have a touch screen so I send a LV_EVENT_CLICKED to the close button. Either way, I had the same problem.

I'll continue monitoring this case. I don't want to send anyone on a wild goose chase. As I said, I'll be dropping the message box in my app but I'll check if anything related to this comes up.

trudelpaul, 2021-08-02:
In my initialization code, it was possible that the lv_event_send() be called in two separate threads simultaneously. The problem only occurred only much later on, which threw me off. I'll be closing this.

closed 2021-08-02 by trudelpaul