lvgl/lvgl #2396: Area y2 < y1 in disp_flush

zhwxChaos, 2021-07-23:
### Describe the bug
I have used many lv_obj_clear_state and lv_obj_add_state methods,  but when lv_obj_clear_state is called in a certain part of the project, the area parameter{ x1(352)    y1(237)    x2(411)   y2(236) } in disp_flush will be abnormal. 

Why is y2 smaller than y1? 

[image]

static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
};

STM32F429+SDRAM
lvgl version: V7.7.0

kisvegabor, 2021-07-23:
We had a similar very strange issue with STM32F746-Disco and @embeddedt fixed it here: https://github.com/lvgl/lv_port_stm32f746_disco/commit/4f7dda9792996caf965625de02b3f72155f3d461

Is it applicable in your case too?

embeddedt, 2021-07-23:
@kisvegabor I'd love to be proven wrong, but I suspect this issue is different. LVGL seems to actually be sending incorrect coordinates in this case, whereas the other issue seemed to be correlated with some type of hardware bus contention, independent of LVGL.

@zhwxChaos Could you try the `release/v7` branch and see if it's fixed there? 7.7 is a pretty old version.

zhwxChaos, 2021-07-26:
@embeddedt I don't use the latter functions, so I didn't upgrade. I just upgraded and found that the problem still exists.

> hardware bus contention
Can you elaborate on the situation of hardware bus contention?

embeddedt, 2021-07-26:
I don't completely understand it myself, but the basic idea is that the memory controller can sometimes stall while the LTDC controller is trying to read from SDRAM. When that happens, a flickering effect appears on the screen.

The weird thing is that the issue we experienced on F746 seemed to be directly tied to the code the processor was running at that moment, as when we reordered one set of function calls inside LVGL, the problem went away.

I still don't fully understand what caused that issue, but moving the RAM to the other address seemed to fix it. I think that changes some semantics of what the memory controller does and reduces the chance of stalling, but I am not an expert.

zhwxChaos, 2021-07-27:
Hi @embeddedt ，
When this happens, the wrong value will be passed in to DMA2D->NLR, and the DMA2D->ISR value will be DMA2D_FLAG_TE.

I also tried to swap y1 and y2, and the upper border of a button on the UI is blurred. And always execute lv_task_handler without exiting, always call disp_flush。

I adjusted the log level to LV_LOG_LEVEL_WARN, and no abnormal information appeared in the log.

kisvegabor, 2021-07-28:
What if you simply return if there is a wrong area?

With a debugger can you check `lv_refr_area()` if there is a wrong area? The size of the area is set there, so hopefully you will see what calculation caused the issue.

closed 2021-08-20 by lvgl-bot