lvgl/lvgl #2398: Switching between two canvases will affect each other

alec-2018, 2021-07-23:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I ran into a problem. After I finished using the canvas, I called lv_obj_del to delete it, then created a canvas on another page, and then deleted it after using it. In this cycle, I found that the two canvases would interfere with each other. Found this problem is in v7.11.0, did not find this problem on v8

### To Reproduce

Use lv_sim_visual_studio simulator x86 debug to run it will appear
```c
#define NO_CONT 1
lv_obj_t* cont = NULL;
void lv_ex_canvas_2(void)
{
    /*Create a button to better see the transparency*/
#if !NO_CONT
    cont = lv_cont_create(lv_scr_act(), NULL);
    lv_obj_set_pos(cont, 0, 0);
    lv_obj_set_size(cont, 800, 600);
    lv_cont_set_layout(cont, LV_LAYOUT_CENTER);
#else
    cont = lv_scr_act();
#endif
    /*Create a buffer for the canvas*/
    static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_INDEXED_1BIT(CANVAS_WIDTH, CANVAS_HEIGHT)];

    /*Create a canvas and initialize its the palette*/
    lv_obj_t * canvas = lv_canvas_create(cont, NULL);
    lv_obj_set_pos(canvas, 400, 320);
    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_INDEXED_1BIT);
    lv_canvas_set_palette(canvas, 0, LV_COLOR_TRANSP);
    lv_canvas_set_palette(canvas, 1, LV_COLOR_RED);

    /*Create colors with the indices of the palette*/
    lv_color_t c0;
    lv_color_t c1;

    c0.full = 0;
    c1.full = 1;

    /*Transparent background*/
    lv_canvas_fill_bg(canvas, c1, LV_OPA_TRANSP);

    /*Create hole on the canvas*/
    uint32_t x;
    uint32_t y;
    for( y = 10; y < 30; y++) {
        for( x = 5; x < 20; x++) {
            lv_canvas_set_px(canvas, x, y, c0);
        }
    }
    lv_task_handler();
    Sleep(250);     //250millisecond
#if !NO_CONT
    lv_obj_del(cont);
#else
    lv_obj_del(canvas);
#endif
    lv_task_handler();
    Sleep(250);
}

void lv_ex_canvas_1(void)
{
    lv_draw_rect_dsc_t rect_dsc;
    lv_draw_rect_dsc_init(&rect_dsc);
    rect_dsc.radius = 10;
    rect_dsc.bg_opa = LV_OPA_COVER;
    rect_dsc.bg_grad_dir = LV_GRAD_DIR_HOR;
    rect_dsc.bg_color = LV_COLOR_RED;
    rect_dsc.bg_grad_color = LV_COLOR_BLUE;
    rect_dsc.border_width = 2;
    rect_dsc.border_opa = LV_OPA_90;
    rect_dsc.border_color = LV_COLOR_WHITE;
    rect_dsc.shadow_width = 5;
    rect_dsc.shadow_ofs_x = 5;
    rect_dsc.shadow_ofs_y = 5;

    lv_draw_label_dsc_t label_dsc;
    lv_draw_label_dsc_init(&label_dsc);
    label_dsc.color = LV_COLOR_YELLOW;

    static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];

    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_TRUE_COLOR);
    lv_obj_align(canvas, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_canvas_fill_bg(canvas, LV_COLOR_SILVER, LV_OPA_COVER);

    lv_canvas_draw_rect(canvas, 70, 60, 100, 70, &rect_dsc);

    lv_canvas_draw_text(canvas, 40, 20, 100, &label_dsc, "Some text on text canvas", LV_LABEL_ALIGN_LEFT);

    /* Test the rotation. It requires an other buffer where the orignal image is stored.
     * So copy the current image to buffer and rotate it to the canvas */
    static lv_color_t cbuf_tmp[CANVAS_WIDTH * CANVAS_HEIGHT];
    memcpy(cbuf_tmp, cbuf, sizeof(cbuf_tmp));
    lv_img_dsc_t img;
    img.data = (void *)cbuf_tmp;
    img.header.cf = LV_IMG_CF_TRUE_COLOR;
    img.header.w = CANVAS_WIDTH;
    img.header.h = CANVAS_HEIGHT;

    lv_canvas_fill_bg(canvas, LV_COLOR_SILVER, LV_OPA_COVER);
    lv_canvas_transform(canvas, &img, 30, LV_IMG_ZOOM_NONE, 0, 0, CANVAS_WIDTH / 2, CANVAS_HEIGHT / 2, true);
    lv_task_handler();
    Sleep(250);
    lv_obj_del(canvas);
    lv_task_handler();
    Sleep(250);
}

while (1) {
  lv_ex_canvas_2();
  lv_ex_canvas_1();
}
```
### Expected behavior

[image]
[image]

### Screenshots or video

https://user-images.githubusercontent.com/40708750/126750666-8b774b8f-ef06-4477-8a7c-d3876ea413c4.mp4

kisvegabor, 2021-07-23:
Probably it's an image caching issue in LVGL. 
Try calling `lv_img_cache_invalidate_src(NULL);` before `lv_task_handler()`s.

alec-2018, 2021-07-24:
> 可能是 LVGL 中的图像缓存问题。
> 尝试`lv_img_cache_invalidate_src(NULL);`在`lv_task_handler()`s之前调用。

Using your suggestion, it is effective and can solve my problem. Thank you, but I think it would be better to add it to LV_SIGNAL_CLEANUP in lv_canvas_signal. What do you think?

like this：
```c
static lv_res_t lv_canvas_signal(lv_obj_t * canvas, lv_signal_t sign, void * param)
{
    lv_res_t res;

    /* Include the ancient signal function */
    res = ancestor_signal(canvas, sign, param);
    if(res != LV_RES_OK) return res;
    if(sign == LV_SIGNAL_GET_TYPE) return lv_obj_handle_get_type_signal(param, LV_OBJX_NAME);

    if(sign == LV_SIGNAL_CLEANUP) {
        /*Nothing to cleanup. (No dynamically allocated memory in 'ext')*/
        lv_img_cache_invalidate_src(lv_canvas_get_img(canvas));
    }

    return res;
}
```

kisvegabor, 2021-08-09:
Sorry, I missed your reply. I've pushed a fix to both v8 and v7.

closed 2021-08-09 by kisvegabor

alec-2018, 2021-08-09:
> Sorry, I missed your reply. I've pushed a fix to both [v8](a1b362c) and [v7](3fce530).

It doesn't matter, I hope lvgl gets better and better

commit 3fce530c (2021-08-09): fix(canvas) invalidate the image on delete

fixes #2398
commit a1b362c9 (2021-08-09): fix(canvas) invalidate the image on delete

fixes #2398