lvgl/lvgl #2403: LVGL doesn't link when added to non-main ESP-IDF component

sidwarkd, 2021-07-24:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Using lvgl as an ESP-IDF component from within another ESP-IDF component that is not the main component fails to link. Throws multiple undefined reference errors.

### To Reproduce
Instead of adding lvgl code to main.c or main.cpp try using the library in another ESP-IDF component and calling that component from main. It will fail to link.

### Expected behavior
Using lvgl from ESP-IDF components other than `main` should link properly.

### Suggested Fix
The [CMakeLists.txt file for lvgl `REQUIRES main`](https://github.com/lvgl/lvgl/blob/105bf24318a6d70d758223bb87ac44c5521cc341/CMakeLists.txt#L7) which it should not to be a true component. This seems to create a circular dependency issue. Removing the REQUIRES fixed the issue for me.

kisvegabor, 2021-07-27:
cc @tore-espressif, @C47D

tore-espressif, 2021-07-27:
Related:
https://github.com/lvgl/lv_port_esp32/issues/274
https://github.com/espressif/esp-idf/issues/6932

@sidwarkd thank you for the troubleshooting. Do you want to file a PR? Or should we take over?

sidwarkd, 2021-07-27:
@tore-espressif I'm happy to submit a PR but want to make sure I at least have some time to test it in both the out-of-box main component config as well as in a custom component to make sure it doesn't cause a major regression. Super busy right now but I'll try to get to it in the next couple of days. Otherwise I'll let you know.

C47D, 2021-07-29:
Hi @sidwarkd , sorry for the late reply, I've seen in the referenced issues by @tore-espressif that your patch does seems to fix the issue, are you planning to submit a PR?

sidwarkd, 2021-07-30:
@C47D Yes, I will get a PR submitted this weekend

kisvegabor, 2021-08-02:
Thanks, I merged the PR.

PR #2420 (2021-08-02): Remove main component dependency
### Description of the feature or fix
Resolves #2403

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [ ] Update [CHANGELOG.md](https://github.com/lvgl/lvgl/blob/master/docs/CHANGELOG.md)
- [ ] Update the documentation
commit f2c2393b (2021-08-02): fix(build) remove main component dependency (#2420)

Resolves #2403
closed 2021-08-02 by kisvegabor