lvgl/lvgl #2405: Events

Miguel0101, 2021-07-24:
### It's not a problem, just a question.

When using two events on an object for example LV_EVENT CLICKED and LV_EVENT_LONG_PRESSED and long press and release both events are executed. Is there a way to fix this?

embeddedt, 2021-07-24:
I think you are looking for `LV_EVENT_SHORT_CLICKED`.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2021-07-24 by embeddedt

Miguel0101, 2021-07-24:
> I think you are looking for `LV_EVENT_SHORT_CLICKED`.
> 
> We use GitHub issues for development related discussions.
> Please use the [forum](https://forum.littlevgl.com/) to ask questions.
> 
> It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

Ok, thanks!