lvgl/lvgl #2408: Compile warning that can be avoided

icodk, 2021-07-26:
### Introduce the problem
Compiling lvgl for esp32 (idf) generates the following warning:
`../components/lvgl/src/core/lv_indev.c:347:131: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]`

It generate 4 warning for the following code:
```

  /*Simple sanity check*/
    if(data->point.x < 0) LV_LOG_WARN("X is %d which is smaller than zero", data->point.x);
    if(data->point.x >= lv_disp_get_hor_res(i->driver->disp)) LV_LOG_WARN("X is %d which is greater than hor. res", data->point.x);
    if(data->point.y < 0) LV_LOG_WARN("Y is %d which is smaller than zero", data->point.y);
    if(data->point.y >= lv_disp_get_ver_res(i->driver->disp)) LV_LOG_WARN("Y is %d which is greater than hor. res", data->point.y);

```

### Examples and cases
IMHO any `if `statement should at least have  one braced body in an embedded software, even if the language allows for omitting such.

### Suggested solution
 ```
/*Simple sanity check*/
    if(data->point.x < 0){ LV_LOG_WARN("X is %d which is smaller than zero", data->point.x);}
    if(data->point.x >= lv_disp_get_hor_res(i->driver->disp)){ LV_LOG_WARN("X is %d which is greater than hor. res", data->point.x);}
    if(data->point.y < 0){ LV_LOG_WARN("Y is %d which is smaller than zero", data->point.y);}
    if(data->point.y >= lv_disp_get_ver_res(i->driver->disp)){ LV_LOG_WARN("Y is %d which is greater than hor. res", data->point.y);}

```

embeddedt, 2021-07-26:
Hmm... I would suggest adding `do {} while(0)` to the related macros in `lv_log.h` instead. That way they are still valid statements even when disabled. I'm pretty sure that's a standard way of solving this problem.

icodk, 2021-07-26:
@embeddedt  This will not mitigate the warning regarding the `if ` statement (?)

embeddedt, 2021-07-26:
It should. An empty do-while loop is considered a statement, so the if statement's body should not be "empty" anymore. I think the issue right now is that the macros are compiling to a completely blank statement.

kisvegabor, 2021-07-27:
I've just fixed it as @embeddedt suggested.

closed 2021-07-27 by kisvegabor

SinglWolf, 2021-09-15:
> I've just fixed it as suggested.

The problem remained, despite the fixes...
[image]
I understand that it should be so, but any compiler warnings are just wildly annoying.

kisvegabor, 2021-09-15:
If warnings are disabled `#define LV_LOG_WARN(...) do {}while(0)` is used. So the body of the `if` statement shouldn't be empty. 

Do you have any idea why is these warnings?

SinglWolf, 2021-09-15:
> Do you have any idea why is these warnings?

You were told in the first message how to solve the problem. Just add curly braces.

```
/*Simple sanity check*/
   if(data->point.x < 0) {LV_LOG_WARN("X is %d which is smaller than zero", data->point.x);}
   if(data->point.x >= lv_disp_get_hor_res(i->driver->disp)) {LV_LOG_WARN("X is %d which is greater than hor. res", data->point.x);}
   if(data->point.y < 0) {LV_LOG_WARN("Y is %d which is smaller than zero", data->point.y);}
   if(data->point.y >= lv_disp_get_ver_res(i->driver->disp)) {LV_LOG_WARN("Y is %d which is greater than hor. res", data->point.y);}
```

No compiler warnings.
Profit!

kisvegabor, 2021-09-18:
Yes, but with https://github.com/lvgl/lvgl/commit/4dba8df2a196fc7a2b7a8686efb6e47fc6cf0fc6 the body of the `if` is not empty. It should be resolved to 
```c
if(data->point.x < 0) do{}while(0);
```

That's why I don't understand the reason of the warning.

SinglWolf, 2021-09-19:
> Yes, but with [4dba8df](https://github.com/lvgl/lvgl/commit/4dba8df2a196fc7a2b7a8686efb6e47fc6cf0fc6) the body of the `if` is not empty. It should be resolved to
> 
> ```c
> if(data->point.x < 0) do{}while(0);
> ```
> 
> That's why I don't understand the reason of the warning.

By default, the use of logs is disabled.
You fixed the part of the code that works when using logs.
The part of the code that works by default remained unchanged. Here's the whole secret. ;-)
```c
#if LV_USE_LOG

//  Everything is fine here

#else /*LV_USE_LOG*/

/*Do nothing if `LV_USE_LOG 0`*/
#define _lv_log_add(level, file, line, ...)
#define LV_LOG_TRACE(...)
#define LV_LOG_INFO(...)
#define LV_LOG_WARN(...)
#define LV_LOG_ERROR(...)
#define LV_LOG_USER(...)
#endif /*LV_USE_LOG*/
```
If you do not want to put curly braces, as you were suggested, do so (for e.g.).
```c
#else /*LV_USE_LOG*/

/*Do nothing if `LV_USE_LOG 0`*/
#define _lv_log_add(level, file, line, ...)
#define LV_LOG_TRACE(...)
#define LV_LOG_INFO(...)
#define LV_LOG_WARN(...) do {}while(0)
#define LV_LOG_ERROR(...)
#define LV_LOG_USER(...)
#endif /*LV_USE_LOG*/
```
The compiler will be very happy. :-D

kisvegabor, 2021-09-20:
God, my bad. :smile: 
I was thinking only about logs disabled due to the log level. 

I've pushed a fix.

commit 4dba8df2 (2021-07-27): fix(log) fix warning for empty log macros

Fixes #2408