lvgl/lvgl #2416: lv_img

Miguel0101, 2021-07-29:
### It's not a problem just a suggestion.

Is there any way to resize images per pixel, for example a 320x240 image and resize to 300x200?

embeddedt, 2021-07-30:
You can calculate the ratio between the two sizes, multiply it by 256, and use `lv_img_set_zoom(img, <the result>)`. See [the docs](https://docs.lvgl.io/master/widgets/core/img.html#transformations) for more information.

Miguel0101, 2021-07-31:
> You can calculate the ratio between the two sizes, multiply it by 256, and use `lv_img_set_zoom(img, <the result>)`. See [the docs](https://docs.lvgl.io/master/widgets/core/img.html#transformations) for more information.

In width and height?

embeddedt, 2021-07-31:
No. I mean the ratio between the area of the images. `(w2/w1)*256`

As far as I know you cannot set the scaling amount for individual axes.

Miguel0101, 2021-07-31:
> No. I mean the ratio between the area of the images. `(w2/w1)*256`
> 
> As far as I know you cannot set the scaling amount for individual axes.

So i have calculate the image area?

embeddedt, 2021-07-31:
Yes, but that’s just width times height.

Miguel0101, 2021-07-31:
> Yes, but that’s just width times height.

Okay, thanks @embeddedt,

closed 2021-07-31 by Miguel0101

kisvegabor, 2021-08-02:
> As far as I know you cannot set the scaling amount for individual axes.

It really isn't but in some cases, I found that an `aspect ratio` property would be great. E.g. if a flexbox sets a width for a container it'd be great to automatically make it a square.