lvgl/lvgl #2418: Input device not work due to user_data lost in lv_timer_create

JeziL, 2021-07-29:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

My input device (touch screen) isn't responding. After some investigation, it turns out that `indev_drv.read_cb` isn't called, because `lv_indev_read_timer_cb` is returned after [this line](https://github.com/lvgl/lvgl/blob/637b706ddc7adc237e8d63292068d2b859843843/src/core/lv_indev.c#L73):
```c
if(indev_act->driver->disp == NULL) return; /*Not assigned to any displays*/
```
However, the indev passed by `lv_timer_create`'s userdata ([here](https://github.com/lvgl/lvgl/blob/039ed0baa6443e08da7e45f635e019e18af73bf8/src/hal/lv_hal_indev.c#L97)) does have a driver->display:

```c
indev->driver->read_timer = lv_timer_create(lv_indev_read_timer_cb, LV_INDEV_DEF_READ_PERIOD, indev);
```

### To Reproduce
Not sure what's the cause but here's what I've done:

```c
// input device init
lv_indev_drv_t indev_drv;
lv_indev_drv_init(&indev_drv);
indev_drv.type = LV_INDEV_TYPE_POINTER;
indev_drv.read_cb = my_touchpad_read;
indev_drv.disp = disp;
lv_indev_t *my_indev = lv_indev_drv_register(&indev_drv);
```

```c
// src/hal/lv_hal_indev.c    Line 97
if (indev->driver->disp != NULL) {
  printf("has display\r\n");
}
indev->driver->read_timer = lv_timer_create(lv_indev_read_timer_cb, LV_INDEV_DEF_READ_PERIOD, indev);
```

```c
// src/core/lv_indev.c    Line 73
void lv_indev_read_timer_cb(lv_timer_t *timer) {
...
/*Read and process all indevs*/
if (indev_act->driver->disp == NULL) {
  printf("no display\r\n");
  return; /*Not assigned to any displays*/
}
...
```

### Expected behavior

- Expected:

```
has display
```
then input device works normally.

- What happened actually:

```
has display
no display
no display
...
```

### Screenshots or video

embeddedt, 2021-07-29:
`indev_drv` needs to be `static`; it's probably going out of scope.

JeziL, 2021-07-30:
problem solved.

closed 2021-07-30 by JeziL