lvgl/lvgl #2421: BASE_DIR.AUTO is not working

amirgon, 2021-07-30:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

In v8 `LV_BASE_DIR_AUTO` does not work correctly on TextArea.

### To Reproduce

```python
scr = lv.obj()
ta = lv.textarea(scr)
ta.set_style_text_font(lv.font_dejavu_16_persian_hebrew, 0)
ta.align(lv.ALIGN.CENTER, 0, 0)
ta.set_style_base_dir(lv.BASE_DIR.AUTO, 0)
ta.set_text('שלום עולם!')
lv.scr_load(scr)
```

[Online example](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=66923d562993ada961536858f279ec19998e3baf)

### Expected behavior

Should align text to the right when RTL text is used.

### Screenshots or video

Current behavior:
[image]

Expected behavior:
[image]

kisvegabor, 2021-08-02:
I've looked into it and it seems `LV_BASE_DIR_AUTO` is not really used by any of the widges. `LV_TEXT_ALIGN_AUTO` is used which resolves to `LV_TEXT_ALIGN_RIGHT` in case of `LV_BASE_DIR_RTL` and else `LV_BASE_DIR_LTR`.

The problem why it isn't used is when LVGL needs to choose either RTL or LTR from AUTO in general low-level features (e.g. align) it can't see whether the widget has a text to figure out the base dir. 

Do you think it's an important feature? If so I can add an `auto_as_rtl` bit to `lv_obj_t` where widgets can store result of `LV_BASE_DIR_AUTO`.

amirgon, 2021-08-03:
> `LV_TEXT_ALIGN_AUTO` is used which resolves to `LV_TEXT_ALIGN_RIGHT` in case of `LV_BASE_DIR_RTL` and else `LV_BASE_DIR_LTR`

This doesn't seem to be working.  
I've tried replacing `ta.set_style_base_dir(lv.BASE_DIR.AUTO, 0)` with `ta.set_style_text_align(lv.TEXT_ALIGN.AUTO, 0)` but text is not automatically aligned to the right as I would expect when characters are RTL chars.

> it can't see whether the widget has a text to figure out the base dir

I'm not sure I understand.  
We are not discussing general RTL/LTR alignment of widgets. We are talking about text alignment. When text is aligned, the widget is aware of the characters that are being set so it should be able to figure out the base dir automatically when configured so.

> Do you think it's an important feature? If so I can add an `auto_as_rtl` bit to `lv_obj_t` where widgets can store result of `LV_BASE_DIR_AUTO`

I'm currently converting an application from LVGL v7 to v8.  I have a text area that displays messages from the server, sometimes these are English messages and on other times these are Hebrew messages.  
On v7 I didn't need to do any text processing. The text area, configured to `lv.BASE_DIR.AUTO`, would align the text according to its contents.  
If in v8 this is not supported, I would now have to process the text myself and set RTL/LTR manually for each message. This doesn't make a lot of sense becuase LVGL already has the logic of BIDI processing.  

So I think this is a useful feature, and I don't see why we would want to remove it on v8.

kisvegabor, 2021-08-03:
You are right, I was overcomplicating it. 

Just pushed a fix, now it should work here and for other widgets too.

closed 2021-08-03 by kisvegabor

amirgon, 2021-08-03:
Great! Thanks.

commit 7672847c (2021-08-03): fix(bidi) fix the handling of LV_BASE_DIR_AUTO in several widgets

fixes #2421