lvgl/lvgl #2422: The default font cannot be changed through Kconfig file

DaMiBear, 2021-07-31:
On the esp32 platform, I added lvgl to the component, but I couldn't use menuconfig to change the default font. Then I found that CONFIG_LV_FONT_DEFAULT is never defined.
````
/*Always set a default font*/
#ifndef LV_FONT_DEFAULT
#  ifdef CONFIG_LV_FONT_DEFAULT
#    define LV_FONT_DEFAULT CONFIG_LV_FONT_DEFAULT
#  else
#    define  LV_FONT_DEFAULT &lv_font_montserrat_8
#  endif
#endif
```

DaMiBear, 2021-08-02:
I changed 
```
#  ifdef CONFIG_LV_FONT_DEFAULT
#    define LV_FONT_DEFAULT CONFIG_LV_FONT_DEFAULT
```

 to 
```
#  ifdef CONFIG_LV_THEME_DEFAULT_FONT
#    define LV_FONT_DEFAULT CONFIG_LV_THEME_DEFAULT_FONT
```
near line 923 of lv_conf_internal.h file, so that I can use menuconfig to change the default font.

kisvegabor, 2021-08-02:
[CONFIG_LV_FONT_DEFAULT](https://github.com/lvgl/lvgl/blob/master/Kconfig#L409) should be declared here in Kconfig.

DaMiBear, 2021-08-04:
Thank you for your correction. I just started learning lvgl. I need to learn the syntax of Kconfig before making changes.

closed 2021-08-04 by DaMiBear