lvgl/lvgl #2424: RTL DropDown layout is wrong

amirgon, 2021-07-31:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

DropDown list part layout is wrong when configured to RTL 

### To Reproduce

```python
options = "\n".join([
    "Apple\n"
    "Banana\n"
    "Orange\n"
    "Cherry\n"
    "Grape\n"
    "Raspberry\n"
    "Melon\n"
    "Orange\n"
    "Lemon\n"
    "Nuts\n"])

scr = lv.scr_act()
scr.set_style_base_dir(lv.BASE_DIR.RTL, 0)
dd = lv.dropdown(scr)
dd.set_width(150)
dd.set_options(options)
dd.align(lv.ALIGN.TOP_MID, 0, 20)
dd.open()
```

[Link to online example](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/145440dbce57babe6ef65936853944947dab7fe6/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/145440dbce57babe6ef65936853944947dab7fe6/examples/widgets/dropdown/lv_example_dropdown_1.py&script_direct=ad2a6704b0f84e4818b5c8dbef3f3ba7fd04bd58)

### Expected behavior

Text should be aligned to to the right, but layout should be corrected.

### Screenshots or video

RTL:
[image]

LTR:
[image]

kisvegabor, 2021-08-02:
Thanks for the report. I've just fixed it. 

Now the simulator is not updated automatically, right?

embeddedt, 2021-08-02:
The v8 simulator automatically follows lv_micropython's `master` branch, but yes, that is not updated automatically.

kisvegabor, 2021-08-02:
Thanks, @embeddedt.

@amirgon can confirm if it's work well now.

amirgon, 2021-08-03:
> @amirgon can confirm if it's working well now?

@kisvegabor Almost.  
The list scrollbar should be on the left and not on the right in case of RTL.

[image]

> The v8 simulator automatically follows lv_micropython's `master` branch, but yes, that is not updated automatically.

@embeddedt If we want, we could add an action on LVGL repo that updates the online simulator every time LVGL is updated.  
Not sure it's so important, just an idea.

kisvegabor, 2021-08-03:
> The list scrollbar should be on the left and not on the right in case of RTL.

If it's expected to place the scrollbar on the left in case of RTL base direction then it's a general issue with scrollbars. I can fix it if you say it'd be more natural.

amirgon, 2021-08-03:
> I can fix it if you say it'd be more natural.

I'm not sure what is the "rule" or "right way" for scrollbar placement on RTL text.  
I can only say that it feels more natural to me find the scrollbar on the left on RTL lists.  
Otherwise, when clicking an element or dragging an element, I often click/drag the scrollbar instead of the element in the list, which is confusing.

embeddedt, 2021-08-03:
I did some research. There seems to be no clear standard when it comes to this, but several people have suggested following the most common design pattern, and leaving the scrollbar on the right.

Setting `direction: rtl` in Chrome does seem to leave the scrollbar on the right, although this may be a Linux issue and not a Chrome issue.

Related reading:

* https://ux.stackexchange.com/questions/24437/scrollbars-in-right-to-left-layout
* https://stackoverflow.com/questions/1261869/setting-the-correct-orientation-for-scrollbars-in-right-to-left-pages

amirgon, 2021-08-03:
Is there at least a way to control the scrollbar side by setting some style to the object?  
Looking at `lv_obj_get_scrollbar_area`, it doesn't seem that there is any way to do that.

kisvegabor, 2021-08-03:
Seemingly there is no consensus about whether the right or left is better. So having an option to change it might be reasonable.

As @embeddedt  mentioned the CSS methods are using `direction:rtl` to swap the scrollbar. But if you only want to place to scrollbar to left without changing the write direction `direction:ltr` needs to be applied on the children.

In LVGL the scrollbar can read the direction from `LV_PART_SCROLLBAR`. This way if the RTL is set on the `LV_PART_SCROLLBAR` only it will be affected, but if set on `LV_PART_MAIN` or any parent the whole layout and the scrollbar will be affected.

The fix itself should be simple as it doesn't change anything on how scrolling works, just changes the coordinates of the drawing. 

What do you think?

amirgon, 2021-08-03:
> What do you think?

Sounds good!

Is it inherited on sub elements?  
If I set "direction" on a dropdown, would it affect the scrollbar of the "list" part?

kisvegabor, 2021-08-03:
> If I set "direction" on a dropdown, would it affect the scrollbar of the "list" part?

Dropdown is a special case because the "button" and the "list" are two separate objects. 
So you need to make them RTL one by one. To make it even more complicated the list is always created on the screen of the button and not the parent of the button. So it's not enough to add RTL to the parent of the button.

amirgon, 2021-08-03:
> To make it even more complicated the list is always created on the screen of the button and not the parent of the button. So it's not enough to add RTL to the parent of the button.

Why is it like that? Wouldn't it be easier to add the list on the parent of the button so there is more control over inherited styles?

embeddedt, 2021-08-03:
As far as I know, we still don't have a `position: fixed`/`position: absolute` concept in LVGL, which means that the only way to fully control the position of the opened list relative to the closed dropdown is to put it on the screen instead of the parent.

amirgon, 2021-08-03:
> the only way to fully control the position of the opened list relative to the closed dropdown is to put it on the screen instead of the parent.

Why can't the dropdown simply calculate the coordinates of the list? Doesn't it know the coordinates the button on the parent?

kisvegabor, 2021-08-03:
> Why is it like that? Wouldn't it be easier to add the list on the parent of the button so there is more control over inherited styles?

For 2 reasons:
1. if the parent of the button is not that large the list would be cropped by the parent
2. a younger sibling of the parent could overlap the list (if the list is the last object created on the screen it will be on top of everything)
 
> As far as I know, we still don't have a position: fixed/position: absolute concept in LVGL, which means that the only way to fully control the position of the opened list relative to the closed dropdown is to put it on the screen instead of the parent.

Maybe I don't understand something about `position:fixed/absolute` but wouldn't they be required to position the list on the screen instead of the on parent where the button and the list have the same coordinate system?

embeddedt, 2021-08-03:
> Wouldn't they be required to position the list on the screen instead of the on parent where the button and the list have the same coordinate system?

Sort of. We have an incomplete version of `position: absolute`, since we can set the X/Y coordinates of child widgets manually relative to their immediate parent. However, the real `position: absolute` doesn't necessarily use the immediate parent as the reference point. It uses the nearest parent with a `position` value set (excluding `static`), or the screen if there is no matching parent.

For our purposes here, `position: fixed` should be enough, as that would allow positioning relative to the screen (what we are doing now), but leave the list's logical parent as the dropdown button.

kisvegabor, 2021-08-03:
@embeddedt 
I see but unfortunately the [two issue](https://github.com/lvgl/lvgl/issues/2424#issuecomment-892156490) are still there :slightly_frowning_face:

amirgon, 2021-08-03:
> 2\. a younger sibling of the parent could overlap the list (if the list is the last object created on the screen it will be on top of everything)

In case of a dropdown, can't the button simply move the list to front when it opens?

kisvegabor, 2021-08-03:
> In case of a dropdown, can't the button simply move the list to front when it opens?

See this [example](http://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=974e23d5f719f0ac4c0bf9c4d3afc3177faf0b5a). Now it works well but if the list wasn't on the screen `cont2` would overlap it.

> What about some style, applied to the list, that overrides cropping?

There is no concept in LVGL to keep overflowing conent. Nothing can be out of the parent. 
PRobably it can be added somehow but it's a global and large change.

amirgon, 2021-08-18:
> As @embeddedt mentioned the CSS methods are using `direction:rtl` to swap the scrollbar. But if you only want to place to scrollbar to left without changing the write direction `direction:ltr` needs [...]

@kisvegabor - Is this supported already? Or should I open a new issue to track this?

> There is no concept in LVGL to keep overflowing conent. Nothing can be out of the parent.
> PRobably it can be added somehow but it's a global and large change.

@kisvegabor - Is this something you see as a future improvement for LVGL? Possibly on v8.1 or v9?

kisvegabor, 2021-08-19:
>  Is this supported already? Or should I open a new issue to track this?

I've just added it. Even this special config works: 
```c
    lv_obj_set_style_base_dir(child, LV_BASE_DIR_RTL, LV_PART_SCROLLBAR | LV_STATE_PRESSED);
```
It places the scrollbar to the left if the object is pressed. (Doesn't make sens, just for testing)

>  Is this something you see as a future improvement for LVGL? Possibly on v8.1 or v9?

I haven't planned to add it, but if we find a reasonable amount of use case we can add it to the roadmap.

amirgon, 2021-08-19:
> I've just added it. Even this special config works:

Great! :+1:  
Thanks.

> I haven't planned to add it, but if we find a reasonable amount of use case we can add it to the roadmap.

Ok.

Closing this issue.

closed 2021-08-19 by amirgon