lvgl/lvgl #2428: Markup Language for LVGL

cjheath, 2021-08-03:
### Introduce the problem
Coding LVGL widget hierarchies long-hand is time-consuming and tricky. If all capabilities of LVGL widgets were normalised according to a DOM, a markup language could be defined to quickly and succinctly express a desired layout. Many standard widgets could be replaced by markup fragments. It would be preferable to avoid the complexity of the web however, which has three languages (HTML, CSS, JS) with overlapping capabilities.

### Examples and cases
Here is an example of a syntax that should be achievable using ADL, to define something like the standard message box class (excluding extra buttons, which can be added in other ways). The details will of course depend on the design of the ADL LVGL library definitions.

```
Message Box : LVObject {
    Flex Flow = Row Wrap;
    Flex Align.Main = Start;
    Flex Align.Cross = Center;
    Flex Align.Track = Start;
    Title: Label {
        Text = "Alert";
        Long Mode = Scroll Circular;
        Flex Grow = true;
    }
    Text : Label {
        Text = "Here's something you should be aware of";
        Long Mode = Wrap;
        Width = 100%;
    }
    Close : Button {
        Extend Click Area = 10px;
        Label {
            Text = Symbol.Close;
            Align = Center;
        }
    }
}
```

And to use this message box, most of the above definitions are inherited:
```
My Message : Message Box {
    Text.Label.Text = "Some message from my application";
    Close.Label.Text = "Cancel";
}
```

### Suggested solution
I'd like to suggest the use of [ADL](https://github.com/cjheath/adl).
Advantages:
* Minimalistic and flexible syntax, with regexp to define arbitrary value syntaxes
* Inheritance and refinement of object classes and value assignments
* Lightweight parser (only 1000LOC)
Possible issues:
* There is no C implementation yet (only Ruby and C#)
* It requires a reasonable regexp library
* There is no example code showing how to implement bindings yet (an event-based API is proposed for the parser)

amirgon, 2021-08-03:
It might be easy to implement with Micropython.  

Your syntax already looks much like a Python dict. If it was a Python dict, it would be very easy to process it and generate the UI from it with Micropython + LVGL Bindings.

cjheath, 2021-08-03:
Yes, you could use Python, and you could use a Dict, and that might even be a good idea. But the similarity is very superficial - please don't confuse your example with ADL! ADL is a language quite unlike anything you have seen before, I guarantee it. For that reason, I'm even uncertain I want to put it forward. But if you dig in and actually understand why I'm saying that, you'll understand why it has special advantages. You can use it to define all the possible object types, variables, and new syntax where needed for new variable types, inheritance, defaults, there is a *lot* hidden under an apparently simple surface.

cjheath, 2021-08-03:
I should say, I was prompted to open this issue following a discussion [here](https://forum.lvgl.io/t/touchscreen-calibration-and-porting-to-version-8/5692/21?u=cjheath)

kisvegabor, 2021-08-04:
@cjheath Thanks for opening the issue.

ADL looks very powerful, although I need some time the benefits better. 

I have a few questions and comments:
1. How to write custom code e.g for events? For example how to add an event to the close button of the message box?
2. I assume we should support updating the UI without firmware updates. So to handle events (custom code) we need to support a scrip language anyway, right? 
3. We need a very straightforward way to get the ADL names from names in C to avoid documenting 2 APIs.

> [...] there is a lot hidden under an apparently simple surface.

It'd be important to understand these advantages compared to the Python version. 

@amirgon 
It's great to see that it works in a so simple way in MicroPython.

Could you create a message box example (similar to @cjheath's one) to better see the differences?

amirgon, 2021-08-04:
> Could you create a message box example (similar to @cjheath's one) to better see the differences?

Do you mean something like this?

```python
Widget(lv.msgbox, 
        "Hello", "This is a message box", ["ClickMe"], True) : 
    {
        'align' : [lv.ALIGN.CENTER, 0, 0],
        'add_event_cb' : [
            lambda e: print('Button clicked!'), lv.EVENT.CLICKED, None],
        Widget(lv.label) : {
            'set_text': 'Message from application!'                
        }
    }
```

[Link to online example](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=2be3e7e9d1e5b7c8d4dd3ec7a0da1578eb403f74)

cjheath, 2021-08-04:
Bear in mind that a large part of the value of a system like ADL (which we also has OpenUI's OPL) was the ability to create a complex class to be used as an inheritable template for further refinement or direct instantiation. We did this instead of having applied styles, as in CSS or LVGL. I don't see that a Python Dict can yield anything like inheritance. And certainly has no type-checking to prevent you attempting to assign an attribute that doesn't exist, or assign a value of the wrong type, etc. ADL has extensive lexical typing, though I have shown here no examples of such types being declared. I assume you can read the documentation for that (though I acknowledge that it is a dry formal spec, not a set of tutorial examples).

cjheath, 2021-08-04:
As far as expressing actions, I don't yet have a proposal. In fact I'd be happy if I still had to attach actions written in C or Python. However, the "regular expression" used for ADL value types is just a tiny step away from being extended to be non-regular, i.e to support extended parsing, which would allow an assigned value to have a syntax like a Javascript function body, for example. PCRE expressions have the required features, which are named-group back-references and look-aheads, but I have not yet decided to make that small leap.

In OPL, all events were messages, so rather than callbacks, we had _translations_ of the form `when SomeEventName <do block>`, and a similar one to subscribe to an event being delivered to an ancestor object. And of course these translations were all inherited by sub-classes and instances. Many translations were very simple, such as `when ButtonPressed Window.open = false;` or `when ButtonPressed Queue InsertNewCustomer(NameField.Value, ...)` - not the exact syntax but you get the idea. Translations are like callbacks, except that they're delayed, and the content fields of the message become local variables.

We could explore ways to do equivalent things with ADL and LVGL anyhow.

cjheath, 2021-08-04:
> 3. We need a very straightforward way to get the ADL names from names in C to avoid documenting 2 APIs.

Yes, but already LVGL needs proper documentation of the names of every property that applies to each class of object. The function names are nor really adequate. The same documentation would work in Python, C and ADL.

amirgon, 2021-08-04:
> Not exactly. I meant reassembling a message box widget using buttons, labels, layouts etc.
> 
> In your example above how to access `Widget(lv.label)` later, outside of the widget creation? I mean something like:

Yes it's possible. We can access specific sub elements by name, we can also utilize Python's class system and inheritance to create new widgets.  

So here is an example (really rudimentary, can probably be improved in many ways) :

```python
class msgbox(lv.obj):
    def __init__(self, parent):
        super().__init__(parent)
        run_ADL(self,[
            {
                'set_flex_flow': lv.FLEX_FLOW.ROW_WRAP,
                'set_flex_align': [lv.FLEX_ALIGN.START, lv.FLEX_ALIGN.CENTER, lv.FLEX_ALIGN.START],
            },
            {Widget(label, name='title'): {
                'set_text': "Alert",
                'set_long_mode': lv.label.LONG.SCROLL_CIRCULAR,
                'set_flex_grow': 1,
            }},
            {Widget(label, name='text'): {
                'set_text': "Here's something you should be aware of",
                'set_long_mode': lv.label.LONG.WRAP,
                'set_width': lv.pct(100),
            }},
            {Widget(btn, name='close'): {                
                Widget(label, name='label'): {                
                    'set_text': lv.SYMBOL.CLOSE,
                    'align' : [lv.ALIGN.CENTER, 0, 0],
                }
            }}
        ])

run_ADL(lv.scr_act(), {    
    Widget(msgbox, name='mymsgbox') : {
        'align' : [lv.ALIGN.CENTER, 0, 0],
        'text' : {'set_text': 'Some message from my application'},
        'close': {'label': {'set_text': 'Cancel'}}

    },
})
```

[Online Example](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=c23665744c3bf3ee7e26548b4f93f6844570d8cf)

cjheath, 2021-08-05:
Please don't use the acronym ADL for something that is not ADL.

amirgon, 2021-08-05:
> Please don't use the acronym ADL for something that is not ADL.

Sure, sorry for that.  
Just taking an inspiration from ADL for my tiny Python example.  

How about calling it "LDL" - LVGL Definition Language?

Also, I'm interested in seeing more ADL examples that could be useful for LVGL, like your Message Box example.  

One of the beautiful things about LVGL is its simplicity for the user.  It's very easy to look at the C API or at some example code and understand what's going on without knowing anything about LVGL.  

ADL on the other hand feels a bit complex. A new language with deep concepts that can take time for users to get used to.  

I'm asking myself whether it might be possible to take the useful parts from ADL but still keep it simple and easy to understand for new users.  
This is, kind of, what I was trying to show in my stupid Python example. Everyone already knows Python so it should be pretty easy to understand some simple Definition Language built on top of it.

cjheath, 2021-08-05:
Well, ADL is really not that complicated to use. It's just that I haven't published any good examples with associated writeup yet.

The best approach would be to use ADL to document the LVGL built-in class hierarchy, the attributes new to each class, and the value types (including enumerations) of those attributes (and even the documentation text for those things).

I would be happy to do this, but despite having been programming with LVGL for a couple of weeks (only) now, I still have no knowledge of an existing document that condenses this information. I mean, it's all there, in the many and various places in the documentation and header files, but I need to be able to get an overview without reading what is effectively a thousand-page book.

If I have missed out on finding this "reference summary of lVGL classes, attributes and values", please direct me to it and I will definitely tackle it in ADL. This kind of summary needs to be separate from the documentation of the API used to manipulate it.

BTW, full credit for having constructed and maintained that book! So many open source software projects are very poorly documented, including many of my own.

cjheath, 2021-08-05:
Having said that, it wasn't hard to fill out some initial ADL. Note that this is documenting the basic capabilities of LVGL. It is not a demonstration of what a user layout would look like.
```
LVGL: {
	Help Text: String;				// Will be used by documentation generators

	Alignment:;					// Two dimensional alignment packed into one enum.
		Align Top Left: Alignment;
		Align Top Mid: Alignment;
		Align Top Right: Alignment;
		Align Bottom Left: Alignment;
		Align Bottom Mid: Alignment;
		Align Bottom Right: Alignment;
		Align Left Mid: Alignment;
		Align Right Mid: Alignment;
		Align Center: Alignment;		// Aka Align Mid Mid :)
	// REVISIT: What to do about the LV_ALIGN_OUT members?
	// REVISIT: Fix LVGL documentation so it says what ALIGN_OUT even means, as the picture doesn't really indicate

	Style: {
		// REVISIT: Fill in the details of what a Style may contain
	}

	Coordinate: Integer {
		// REVISIT: Add a value Syntax to extend Integer to support % and pixel coordinates, for example
		// Syntax = /...../
	}

	LVObj: {
		// Parent is implicit in ADL

		// Coordinate/sizing attributes:
		Width: Coordinate { Minimum = 1 }
		Height: Coordinate { Minimum = 1 }
		X: Coordinate;
		Y: Coordinate;
		Align -> Alignment;
		Z: Integer;			// Position of this object in its parent's child list, implicit in lv_obj_move_foreground(), etc.

		// Flags:
		Flag: Boolean {
			Help Text;
		}

		Flags {
			// REVISIT: What are the default values for these flags?
			Hidden:		Boolean { Help Text = "Make the object hidden. (Like it wasn't there at all)" }
			Clickable:	Boolean { Help Text = "Make the object clickable by the input devices" }
			Click Focusable: Boolean { Help Text = "Add focused state to the object when clicked" }
			Checkable:	Boolean { Help Text = "Toggle checked state when the object is clicked" }
			Scrollable:	Boolean { Help Text = "Make the object scrollable" }
			Scroll Elastic:	Boolean { Help Text = "Allow scrolling inside but with slower speed" }
			Scroll Momentum: Boolean { Help Text = "Make the object scroll further when "thrown"" }
			Scroll One:	Boolean { Help Text = "Allow scrolling only one snappable children" }
			Scroll Chain:	Boolean { Help Text = "Allow propagating the scroll to a parent" }
			Scroll On Focus: Boolean { Help Text = "Automatically scroll object to make it visible when focused" }
			Snappable:	Boolean { Help Text = "If scroll snap is enabled on the parent it can snap to this object" }
			Press Lock:	Boolean { Help Text = "Keep the object pressed even if the press slid from the object" }
			Event Bubble:	Boolean { Help Text = "Propagate the events to the parent too" }
			Gesture Bubble:	Boolean { Help Text = "Propagate the gestures to the parent" }
			Adv Hit Test:	Boolean { Help Text = "Allow performing more accurate hit (click) test. E.g. consider rounded corners." }
			Ignore Layout:	Boolean { Help Text = "Make the object position-able by the layouts" }
			Floating:	Boolean { Help Text = "Do not scroll the object when the parent scrolls and ignore layout" }
			// REVISIT: Do we even need these? They reflect spare bits in a binary word, but they can be defined when needed.
			Layout 1:	Boolean { Help Text = "Custom flag, free to use by layouts" }
			Layout 2:	Boolean { Help Text = "Custom flag, free to use by layouts" }
			Widget 1:	Boolean { Help Text = "Custom flag, free to use by widget" }
			Widget 2:	Boolean { Help Text = "Custom flag, free to use by widget" }
			User 1:		Boolean { Help Text = "Custom flag, free to use by user" }
			User 2:		Boolean { Help Text = "Custom flag, free to use by user" }
			User 3:		Boolean { Help Text = "Custom flag, free to use by user" }
			User 4:		Boolean { Help Text = "Custom flag, free to use by user" }
		}

		// Styles:
		Styles => Style;		// Array of references to Styles.

		// Events:
		// REVISIT: Decide how to model events, and to represent event values (callbacks)

		// Groups:
		// REVISIT: I don't yet know how Groups work

		// Keys:
		// REVISIT: I don't know why these aren't just predefined event handlers

		// Parts:
		// REVISIT: I don't think the ADL needs parts, except the parts that might be present in all LVObj's (scrollbar?)
		// Any subclasses of LVObj that defines a Part will just use a member object that can be referenced normally,
		// or a member class that has instances that are members (e.g. Items, Ticks, etc).
		// But for the sake of discussion let's define a part here:
		HScrollBar -> LVObj;	// Note that this means there is no predefined HScrollBar, and any LVObj could be assigned as the HScrollBar.
		VScrollBar -> LVObj;	// This is probably sub-optimal!
	}

	// Core Widgets:
	Arc: LVObj {
	}
	Bar: LVObj {
	}
	Button: LVObj {
	}
	Button Matrix: LVObj {
	}
	Canvas: LVObj {
	}
	Checkbox: LVObj {
	}
	DropDownList: LVObj {
	}
	Image: LVObj {
	}
	Label: LVObj {
	}
	Line: LVObj {
	}
	Roller: LVObj {
	}
	Slider: LVObj {
	}
	Switch: LVObj {
	}
	Table: LVObj {
	}
	Test Area: LVObj {
	}

	// Extra Widgets
	Calendar : LVObj {
	}
	Chart : LVObj {
	}
	Color Wheel : LVObj {
	}
	Image Button : LVObj {
	}
	Keyboard : LVObj {
	}
	LED : LVObj {
	}
	List : LVObj {
	}
	Meter : LVObj {
	}
	Message Box : LVObj {
	}
	Span : LVObj {
	}
	Spinbox : LVObj {
	}
	Spinner : LVObj {
	}
	TabView : LVObj {
	}
	TileView : LVObj {
	}
	Window : LVObj {
	}
}
```

cjheath, 2021-08-05:
Notes to help understand what is going on here:

* the multi-word name before a colon `:` is the name of a new object. The words in the name may be purely numeric, not just alpha
* The name after a colon is the superclass. If no superclass is present the superclass is the ADL `Object`, i.e. could be anything
* Enumerated values in ADL are expressed as a subclass of the superclass. This allows type checking to enforce correct usage
* If `Coordinate.Syntax` gets assigned to a regular expression, that will be used to parse all values assigned to a Coordinate (including `Minimum`, as seen here)
* The syntax `Align -> Alignment` says that the member `Align` may only be assigned using a reference to `Alignment` or one of its subclasses
* The syntax `Styles => Style` means that Styles is a reference to an array of Style objects

amirgon, 2021-08-05:
> Having said that, it wasn't hard to fill out some initial ADL. Note that this is documenting the basic capabilities of LVGL. It is not a demonstration of what a user layout would look like.

It's still not so clear to me what you actually *do* with the ADL that describes the LVGL class hierarchy.  

I suppose there is a way to automatically generate both LVGL documentation and the C header files from it?  
Could you show us how to do that?

cjheath, 2021-08-05:
Generating code and documentation would be possible, but that's an application that is not yet written.

However, what I envisaged is that the file above would be used to describe (and perhaps generate) a C or Python language binding. The binding would include a method such as `create_from_adl(...)` that parses an end-user ADL file, and using the binding (whether it was produced manually or by a generator) and executes its instructions, generating an object hierarchy corresponding to the ADL file. Thus isolating the definition of the structure of a user interface from the details of how to call the low-level APIs to instantiate the user interface objects.

So far, ADL is a processor for a specific kind of syntax. The semantic details (the binding) for this application would have to be defined. Much the same as if we used YAML, XML, HTML or JSON. Or a Python Dict, for that matter. But I maintain is has substantial advantages over all those. The customers who allowed our company to grow to 120 staff certainly thought that was true of OPL, and ADL is even more powerful.

cjheath, 2021-08-15:
The main reason I am not in love with the idea of just using Python is it really doesn't fit with my hardware. I'm writing real-time signal processing code on an ESP32, for software-defined radios. The UI can lag behind, but I can't have an interpreter blocking either CPU core while it does garbage collection - and I don't know about the impact on available memory requirements either.

amirgon, 2021-08-15:
> but I can't have an interpreter blocking either CPU core while it does garbage collection - and I don't know about the impact on available memory requirements either.

Well, you can control gc (disable/enable it, and trigger it explicitly when you like).  
On ESP32 you can also control on which core Micropython runs. By default it runs on core 0 where Wifi and BT are running, so you still have an entire core available for other purposes.
Regarding RAM, on ESP32 boards where PSRAM is available Micropython uses it exclusively, leaving the precious RAM (fast, DMAable etc.) available.

In my projects I'm implementing the critical parts in C (or [Viper](https://docs.micropython.org/en/v1.9.3/pyboard/reference/speed_python.html#the-viper-code-emitter) in some cases), exposing Micropyton API for these parts and controlling them from Micropython.

I found the biggest advantage of Micropython is agility during development time - no need to wait for the build-link-flash cycles on every change, just change the code and run it immediately.  
This allows trying out many different things quickly, saving a lot of time.

cmumford, 2021-08-24:
@cjheath are you envisioning this as a standalone library that LVGL users could use, or a core part of LVGL? Would this be purely a runtime thing, or could it include a code generator? One downside of a runtime-only implementation is that you can't rely on the linker to omit unused code as that won't be know until the definition is parsed.

cjheath, 2021-08-24:
Very good questions @cmumford. It needs to stand alone as an optional layer on LVGL. But the reason I raised it here is because for it to work properly, it needs reliable reflection in the core. What I mean by that is that given any LVGL object hierarchy, it should be possible (knowing the default values) to call methods that return sufficient information to reconstruct the sequence of creations and attribute assignments (even regenerate the code) that produced that hierarchy (obviously not the content of callback functions though, but all static layout). At the moment that's not possible, because some attributes have values that are settable but also internally computed (this is bad hygiene - values set should stay set, and different values computed).

Regarding runtime/compile-time, both options are valuable. Even where a runtime interpretation is used, the set of available widgets might be limited by a compile-time configuration. But a CLI that allows creation and configuration of new hierarchies would be a lovely thing to use.

My preference is to make the runtime parts as lightweight as possible (certainly much lighter than micropython). ADL is very lightweight - but to complete a C implementation I need a complying regex implementation. There are suitable options for that, but all need some adaptation. Then there needs to be a binding layer, which will rely on the LVGL core having a reflection interface. If we can solve the reflection interface, I can do an ADL implementation for everyone to experiment with.

cjheath, 2021-08-24:
The reflection core needs the following features:

- A registry of the object class names and associated constructor functions
- A registry of all "parts" of each class (built-in sub-objects distinct from user-instantiated ones)
- A registry of all attributes (core, styles and callbacks) relevant to each class and/or part (name, type, set/get functions)
- The last-`set` value of each attribute to be available from a `get` call

The class registry can be populated by a static initialiser, such that any class which is linked in is registered. And a configuration file/menu could include/exclude classes to be available from this registry.

The work to construct the attribute/style registry would be worthwhile even for documentation purposes - it's not easy to know which attributes are relevant to which object classes as the (otherwise excellent) documentation currently stands.

excitedbox, 2021-09-05:
I had suggested something like a python dict using C last year. Basically re-creating a CSS syntax, an interpreter to read styles, events, etc. into a dictionary and at startup generating a DOM that is then updated as buttons are clicked. The benefits were that it sped up the entire style system which at the time was causing latency problems and instead using more memory at startup time but offered a 100x speed up overall. It also made adding application features very easy because you could create an event function, assign it an id and name in the dictionaries and have access to it in your "CSS". This gave LVGL the power to not only control your gui but also the rest of your application through this simplified syntax. (ie. add an on click event to a button that executed some function)

My goal was that the user would be able to design and program their entire GUI using just CSS statements if they wanted to. It required a lot of work to implement though and I guess I failed to explain it clearly enough since others didn't really understand my suggestion so after a while I dropped the issue. 

A quick description:
The user creates a layout description file and a style definition file that is read at startup and creates hash tables of all the defined styles (5 altogether for styles, classes of styles, widgets, events, actions) these use master lists (also 5 so ids stay within 1 byte) of all available statements (the lvgl widgets, styles, and features) to build a DOM (The layout description), this is looped through by lvgl and drawn to the screen. Any time an event is triggered an update is marked to an array of changes to make to the DOM on the next loop. Since it doesn't matter if an event triggers a DOM update or a function an event could also execute application code.

The speed advantage comes from being able to use a 1 byte id to get a pointer for whatever function is needed to generate a style, group of styles in case of a class, and also for style inheritance. The current style system has to search through multiple linked lists instead of jumping directly to an item in an array. This makes a a worst case difference of searching 30k plus styles down to only retrieving the 100-200 that you actually need.

kisvegabor, 2021-09-16:
Sorry for being out of this discussion so far. I was thinking about this topic and just came to a conclusion about how to approach it.

For first we should clarify what the goal of such markup language is and what features we expect. Here is my vision:
- Use the markup to describe a UI in a platform independent way. 
- From the markup it should be possible the generate LVGL code in any language (C, C++, MicroPython)
- Parse the markup files/snippets runtime. It allows UI updates without reprogramming the device. 
- Create the UI using the markup language directly. We need a viewer to do so. It is similar to how you can write HTML code in a text editor.   

The features I'd expect:
- Have a format (language) that is familiar for the developers. Basically that is XML or JSON. (I'd prefer JSON). 
- Reuse existing parsers with minor customization if required. 
- Follow LVGL's API names, conventions and "way of thinking" as much as possible to shorten the learning time. 
- Events can be handled with function calls instead of inline code. E.g. `click_event: my_event_handler`. It's the user's responsibility to ensure that  `my_event_handler` exists. That is, do not allow any arbitrary executable code in the markup. 
- Allow creating custom components. E.g. `superButton` with custom properties. 
- Do not focus on reactive programming a references in values. E.g. it's not necessary to support `btn1.width: btn2.width` and update `btn1.width` automatically if `btn2.width` changes

With this logic I think something like QML would be the best direction. What do you think? Do you know other goals, mandatory features, etc?

---

@cjheath 
> Inheritance (so you can't define re-usable macro types like BigRedButton, with some attributes predefined)
Composite types (markup-defined versions of messageBox - parent and content - for example)

You can create custom components with QML. See [here](https://doc.qt.io/qt-5/qmlfirststeps.html#defining-custom-qml-types-for-re-use).

@excitedbox 
You have mentioned that you made some experiments with the proposed idea and we asked if you can share it to see how it works in the practice. Back then you haven't shared it, but I'm still curious about it. :slightly_smiling_face:  So can you show it now?

cjheath, 2021-09-17:
That's great, thanks for your feedback @kisvegabor. I agree with most of what you've written. I have a few questions and suggestions still (not in any specific order):

> Use the markup to describe a UI in a platform independent way.

What does "platform independent" mean here? I assume you don't mean "independent of the geometry of specific screens" because AFAICS that's not really something LVGL helps with very much.

> Have a format (language) that is familiar for the developers

Yes to that. But don't limit it to just "one true" format. Provide support for markup languages, and make each such language a separate module that makes use of the intrinsic support. I don't like YAML, but it would work very well for this, and someone might want to implement it. I definitely want to implement ADL.

> Follow LVGL's API names

Yes, subject to lexical style considerations. For example many of the names are designed for a language without namespace support. For example in C we use `lv_` prefix, whereas in C++ a `LV::` prefix might be preferable. The suffix should match, of course.

> What do you think? Do you know other goals?

I would aim to minimise the code size overhead. It should not be necessary to include an entire Python interpreter, for example. I am working on ESP32 and need to limit the total code size. Not every application is on a RPi or bigger.

You didn't mention any expectation that if markup sets an attribute, that attribute value should read back consistently. This expectation allows generating markup from an LVGL hierarchy that has been created using another method, for example. It also greatly simplifies building an interactive UI design tool. This means that if an object's size and position will be moved by a geometry manager, there needs to be two sets of attributes: the ones that are inputs to geometry management, and the ones that are outputs (the latter can be re-calculated without the former being changed).

> You can create custom components with QML

Excellent - that's exactly what OpenUI provided (34 years ago) :) It is more convenient than using LVGL's styles.

kisvegabor, 2021-09-17:
> What does "platform independent" mean here? I assume you don't mean "independent of the geometry of specific screens" because AFAICS that's not really something LVGL helps with very much.

By "platform" I meant not the target platform but the language itself. E.g. using MicroPython would require to add the interpreter which is a limitation if someone needs to get C code from the markup.

> Yes to that. But don't limit it to just "one true" format. Provide support for markup languages, and make each such language a separate module that makes use of the intrinsic support. I don't like [...]

YAML is very flexible but somehow I also don't like it. Do you mean there could be different markup languages for LVGL? I was thinking about only one to make maintenance simpler. Do you think we can change ADL to be more or less a superset of JSON? At lest syntactically (`key:value`, `{..}`, dynamically typed keys, etc)

> You didn't mention any expectation that if markup sets an attribute, that attribute value should read back consistently. 

Yes, I'll improve it in LVGL. 

> the ones that are inputs to geometry management, and the ones that are outputs (the latter can be re-calculated without the former being changed).

Speaking about a designer tool, it's an interesting question how to handle "getters". E.g. how can the tool as the bounding box an an object using a "markup-like" commands? A markup language doesn't give back values.

cjheath, 2021-09-17:
Thank you for the explanations.

> it's an interesting question how to handle "getters".

The markup language requires metadata that describes the types of objects available to make up a DOM, and the available attributes and allowable values of each of those types. Markup is then used to create the objects and set the attributes - but other application code has a right to expect that the attribute values will still contain the values that were set. Markup might not need to `get` values, but setting and getting must be coherent.

In my opinion there are two separate tasks here:
- Creating a runtime readable registry of available object types, and the attributes of each type
- Creating a markup language that makes use of this metadata.

I started this thread in the hope that we can agree to implement the first task. Without that, the second task is severely compromised. As I said in the first message:

> If all capabilities of LVGL widgets were normalised according to a DOM, a markup language could be defined to quickly and succinctly express a desired layout.

kisvegabor, 2021-09-17:
I also thank you :) 

> Markup might not need to get values, but setting and getting must be coherent.

Let's say in HTML you write:
```html
<button>Submit</button>
```
It's very difficult to figure out the actual size of the button without simply "getting" its width and height. It's because there are many default values in HTML coming from a wide variety of sources. But it's the same for LVGL too. `lv_btn_create` will create a button with default styles, sizes, attributes, etc.

We can say that it's out of scope for the markup language. If the designer tool needs getters, they can add a custom getter mechanism as they need.

> Creating a runtime readable registry of available object types, and the attributes of each type
Creating a markup language that makes use of this metadata.

I agree. The MicroPython binding generation emits such registry in JSON. Probably we need to filter it the markup doesn't need to create displays, handle drivers, call drawing functions directly, etc.

> capabilities of LVGL widgets were normalised according to a DOM

Exactly what do you mean by that?

cjheath, 2021-09-18:
> It's very difficult to figure out the actual size of the button without simply "getting" its width and height.

However it's not at all difficult to get the `width` or `height` attribute values - these are not the actual displayed width or height, but just the values that were requested. Internally the browser also has actual width and height, but you can't fetch those using these attributes. And that is my point; the requested geometry is one thing and the actual geometry is another. These should be separate things.

> > capabilities of LVGL widgets were normalised according to a DOM
> Exactly what do you mean by that?

I'm sorry I wasn't clear. In many old widget toolkits, there is no single parent-child hierarchy. For example in the old Windows API, a menu button is not an object, but is created using APIs that only work with menu buttons. QT used to be similar, and had to be fixed to allow QTML to be built. LVGL has this problem in some areas where objects are composite, and you use `parts` to refer to the different parts of the object, instead of treating it as an object that has sub-objects. So I suggest to normalise all object creation (including implicit sub-objects) under a single parent-child hierarchy. Every object (explicit and implicit) should have a type, and it should be possible to look up the metadata tables for every type. This table should contain a list of all attributes available for the object (I used the word *capability* for those). Each attribute should have a data type (defining the allowable values) and the default value.

In that way, generic reflection code can look at any object at runtime, discover all attributes it has, get the values, and decide if any value is different from the default, so it can generate minimal code that will recreate that object. Or generate markup, for that matter. And of course it needs to be able to do the same for all child objects (including knowing how to ignore implicit child sub-objects).

But the same feature allows a markup language implementation to be extremely simple, so there is not much code to maintain. When markup asks for a new object of class X, it can ask the metadata table whether that class exists, and can create one. When markup wants to assign an attribute, it can determine whether the attribute name exists, what its type should be, and can coerce the value to the correct type and assign it. All this is a tiny amount of code, _if the metadata tables/APIs exist_.

My post above goes even further, suggesting how a language like ADL can be used to define this metadata, in a way that allows generation of the run-time tables and also the documentation for the DOM objects. Sure, you could do this in XML or JSON too, but then you have two kinds of JSON - one that defines the metadata, and one that defines markup for a screen layout. There is also an in-between feature, which allows specification of composite objects, like the QML example you pointed to. Only ADL provides a way to unite all those things in one language, which is why I'd like to implement that.

But I'm not prepared to implement a markup language unless we have agreement to implement the required support for them. I don't want to make any systematic changes to LVGL itself unless we first agree to do that.

kisvegabor, 2021-09-20:
> However it's not at all difficult to get the width or height attribute values - these are not the actual displayed width or height, but just the values that were requested. Internally the browser [...]

Do you mean e.g `width:50%` is set by the user, and it's calculated e.g. to `100px` based on the context/parent/etc? Anyway, I think we should not care much with it as it's not the goal of the markup to provide a  "getter" interface, but rather only a "setter" interface.

> LVGL has this problem in some areas where objects are composite, and you use parts to refer to the different parts of the object, instead of treating it as an object that has sub-objects.

`part`s are usually only for styling. E.g. a slider has `LV_PART_KNOB` the target the knob with a color, radius, padding, border, etc. Actually, the knob is not a real object, just drawn on the fly based on the set parameters. 

IMO there is no point to expose the knob as a real object because it's way too simple. It's not required to set x, y, width, height, scroll related properties, add children, change parent, add events, etc with it. It's just a rectangle with some styles and its position its calculated by the slider. 

I understand the need for a metadata tables but IMO `part`s can be handled in it.  Probably you imagined it differently so just as a topic starter I write here a sketch to show what's in my mind now (just as a bullet list):

```
- obj 
  - x: int (default: 0)
  - y: int (default: 0)
  - width: int (default: 100)
  - height: int (default: 50)
  - ...and all lv_obj paramter...
- slider 
   - value: int (default: 0)
   - range: int, int (default: 0, 100)
   - mode: enum(NORMAL, SYMMETRICAL, RANGE) (default: NORMAL)
   - parts: MAIN, INDICATOR, KNOB
```

For styles:
```
- bg_color: color (default: white)
- border_color: color  (default: black)
- border_width: int  (default: 0)
- etc
```

All together to create styles and objects:
```
- style1:
  - bg_color: blue 

- slider1:
  - x: 100
  - y: 200
  - local_stye, indicator, pressed: 
  - style_main: style1, style2, bg_color: red         # Mix with local style?
  - style_main_pressed: style_3
  - style_knob: style4
  - event_pressed: "my_func" 
```
 
It's my first attempt to think about a markup language so probably you have a more advanced view on how things should be structured. :slightly_smiling_face:

cjheath, 2021-09-21:
> Do you mean e.g width:50% is set by the user, and it's calculated e.g. to 100px based on the context/parent/etc?

Yes. Those are two *different* attributes - what I asked for, and what I got. If the asked-for value gets lost, it's not possible to recompute an actual layout when a widget gets added/removed/hidden/shown.

> it's not the goal of the markup to provide a "getter" interface,

That's the established tradition, but I think it's wrong. The web is seriously compromised by using three different languages (HTML, JS and CSS) instead of one unified language. That's why there have been all the experiments with CSS-in-JS, HTML-in-JS, and the opposite (behaviour in HTML), and they're all a bit crap. It would be nice to avoid that, rather than copy it.

> no point to expose the knob as a real object because it's way too simple

But it *is* an object, it's just not an `lv_obj`. It doesn't matter that it has a different implementation, the same APIs to get and set attributes, control behaviour, etc, should all apply. I'm talking here about a get/set that looks like `OBJ.setAttr(AT_Background, "green")`, where AT_Background is a metadata item. Not a `OBJ.setBackgroundAttr("green")` style of thing - since these proliferate API calls and make the task of a scripting engine much larger.

The entire API of OpenUI was like 20 API calls. We implemented those in C, C++, Pascal, ADA and COBOL - and the same across a client-server network protocol - on 40 different machine architectures and operating systems - and a dozen different windowing toolkit versions.

The built-in and application-specific metadata required by the API was generated to those target languages from the high-level scripting language.

> It's my first attempt to think about a markup language

When you start to write up a *metadata* language (to be used by a *markup* language - which might be the same language) you will get new insights about how to optimise and simplify the implementation.

I think a first target towards resolving this issue would be to produce the required metadata for the existing implementation - in any machine-readable form. It can easily be transformed to a different language (and documentation) when needed - but it's a lot of work to look through all the code and figure out what attributes are used/useful and document the data types.

kisvegabor, 2021-09-21:
> If the asked-for value gets lost, it's not possible to recompute an actual layout when a widget gets added/removed/hidden/shown. That leads to unstable/unpredictable behaviour.

LVGL saves the `50%` so it can recalculate the size if the layout changes. But `lv_obj_get_width()` return the current physical width (always in pixels).

> That's the established tradition, but I think it's wrong. The web is seriously compromised by using three different languages (HTML, JS and CSS) instead of one unified language. That's why there have been all the experiments with CSS-in-JS, HTML-in-JS, and the opposite (behaviour in HTML, e.g. Ember, etc), and they're all a bit crap - and the proliferation of experiments means there's no one core skill-set for developers. It would be nice to avoid that, rather than copy it.

I also see many issues with the current web technologies so I'm definitely in favor of not copying it blindly :slightly_smiling_face: 

> But it is an object, it's just not an lv_obj. It doesn't matter that it has a different implementation, the same APIs should all apply to get and set attributes, control behaviour, etc.

I'd say a subset of the total API can be applied to the parts. E.g. you can not change the parent of  a slider's knob, it can not have children, it can not have its own events, etc. But the style related attributes works the same way on "normal" objects and its parts too. It's because the normal object is usually referred as `LV_PART_MAIN` so it's also a "part".

> When you start to write up metadata in a suitable language (to be used by a markup language - which might be the same language) you will get new insights about how to optimise and simplify the implementation.

True. :slightly_smiling_face: 

> When you start to write up metadata in a suitable language (to be used by a markup language - which might be the same language) you will get new insights about how to optimise and simplify the implementation.

Let me rearrange my previous example with simpler parts and states "support":
```
```
- style1:
  - bg_color: blue 

- slider1:
  - x: 100
  - y: 200
  - event
     - pressed: "my_func" 
  -styles
     - pressed knob: style1, style2, bg_color:red          #mix style sheets and local styles
     - knob: bg_color: black                               #if no state -> imiplicitly default
     - checked: bg_color: black                            #if no part -> imiplicitly main
```
```

> I think a first target towards resolving this issue would be to produce the required metadata for the existing implementation - in any machine-readable form.

IMO first we should find a pseudo code that conceptually looks good. 

> It can easily be transformed to a different language (and documentation) when needed - but it's a lot of work to look through all the code and figure out what attributes are used/useful and document the data types.

Lots of work but not complicated. Just takes time. It'd be great to write a script for it but first let's try to create the meta data ofa  widget manually to see how it should look like.

cjheath, 2021-09-21:
> LVGL saves the 50% so it can recalculate the size if the layout changes.

Ahh, excellent news. I thought I must be missing something.

> Let me rearrange my previous example [...]

I understand why LVGL has styles, but they're not really a natural or convenient concept to represent in a markup language. I think ADL can do it, but It will take some thought to make it seem fluent. Of course it's easy enough in an untyped markup language like YAML or JSON, but I would still like to have a type-safe alternative available. That's difficult when you have styles, which can assign an attribute that doesn't exist for an object it might be used with.

Here's an experiment. I haven't checked to see if this will compile, but it's close.

```
style1: style {
  bg_color = "blue"
}
slider1: slider {
  x = 100;
  y = 200;
  pressed event = "my_func";
  styles {
    pressed knob = [ style1, style2, : style { bg_color = "red" }];
    knob = [ : style { bg_color = "black" } ];
    checked = [ : style { bg_color = "black" } ]
  }
}
```

> Lots of work but not complicated.

There is a syntax example in ADL earlier in this thread. That is as nice a pseudo-code as you're going to find. And I can also make it machine-processable. (I have started work on a C implementation, but I think Python will also be necessary for this community. Neither is a terribly big job - but for C, I don't have a suitably small&tight regexp implementation yet).

kisvegabor, 2021-09-22:
> I understand why LVGL has styles, but they're not really a natural or convenient concept to represent in a markup language. I think ADL can do it, but It will take some thought to make it seem fluent.

What can be the alternative for styling?

> Of course it's easy enough in an untyped markup language like YAML or JSON, but I would still like to have a type-safe alternative available. Type safety is less effective when you have styles, which can assign an attribute that doesn't exist for an object it might be used with.

What are the advantages of being type safety in our case?

>  I think Python will also be necessary for this community.

Python would be perfect. Once the markup is processed is should be simple emit the UI code in any language (C, C++, Python, etc)

cjheath, 2021-09-22:
> > I think Python will also be necessary for this community.

> Python would be perfect.

Oh, sorry, yes, but runtime was not what I intended. I mean for the pre-processing, metadata and markup tools, generating code and documentation for example. Perhaps also as a target language for runtime, but definitely not in all cases - my targets on real-time embedded signal processing in particular cannot bear the vagaries of an interpreter or a garbage-collected language.

> What can be the alternative for styling?

In traditional UI toolkits, the widget classes define the attributes, and each attribute is directly assigned as necessary for each widget (or perhaps by a custom supertype). Apart from the WWW (with CSS) I'm not aware of another toolkit that defines sets of attribute assignments (styles) that are *later* assigned to widgets. That offers some advantages, but also another layer of complexity. And no possibility of static checking that an attribute defined on a style will actually apply to the widget that the style is assigned to. HTML and CSS are a complete dog's breakfast in this respect, barely preferable to the appalling X11 resource files they were inspired by.

The ability to master UI styling should not require an entire career choice, yet here we are.

> What are the advantages of being type safety

The ability to check at compile time that no attribute is used on an object that doesn't support it (or that has a locked assignment already - something not supported in LVGL I think - like *!important*?), or given a value that is incorrect or meaningless in some way. Also the ability of an interactive design tool to offer only valid possibilities.

kisvegabor, 2021-09-22:
>  I mean for the pre-processing, metadata and markup tools, generating code and documentation for example. 

I meant the same :slightly_smiling_face: 

> In traditional UI toolkits, the widget classes define the attributes, and each attribute is directly assigned as necessary for each widget (or perhaps by a custom supertype).

Local styles can be used too: "Set the bg_color of btn1 to red". Style sheets can be considered optional.

>The ability to master UI styling should not require an entire career choice, yet here we are.

Hehe, true :slightly_smiling_face: 

> The ability to check at compile time that no attribute is used on an object that doesn't support it (or that has a locked assignment already - something not supported in LVGL I think - like [...]

I agree that knowing the type of the attributes is required but only in "widget descriptors". Example from ADL's README:
```
Person: {
    Given Name: String[];
}
```

We could also write in JSON something like:
```
Person {
    GivenName: { type: "String", default: "None"}
}
``` 

By the way, how should we indicate the default values in ADL?

In UI markup:
```
- slider:
   - value = 50      //Ok, it's int
   - value = "50"   //Might be ok as can be converted to in 
   - value = "foo"  //Incorrect, not int at all 
```

So type safety means, we use the types defined int the widget descriptors to validate the correctness of values in UI markup, right?

If so what is the problem with the style? The description of `lv_obj` might look like this:
```
- obj
   - bg_color : Color
   - border_color : Color
   - border_width : Int
   - ...
   - styles : Style 
```

That is the values from styles are not applied to the object directly. The objects save only a reference to the style. 

Having states and parts related to styles and local style properties makes it less trivial though.

cjheath, 2021-09-23:
I need to make more than one response today. This one is about Styles. Some of it covers the underlying type theory, and I'm aware that some folk just aren't interested in this kind of theory, so ignore it, and benefit from the practical results anyway.

I had a major Aha! moment a bit after 3:30AM this morning. I realised why Styles seem painful and unnecessary (but aren't), and I unified that with a feature of ADL that I've been having trouble explaining (LVGL styles is an excellent demonstration of the need for this feature), while at the same time unifying both with a facet of type theory that's been troubling me for almost twenty years. So hang tight, and I'll try to explain the relevance here.

Firstly, my message yesterday was based on the idea that styles are required for the baseline state of objects. But they're not really. They're required to provide full flexibility for non-baseline states (button pressed, etc). The flexibility they provide for that is not really possible using the traditional toolkit approach. Those older toolkits simply can't do the things that LVGL can.

ADL has a feature called Contextual Extension. This allows the same object hierarchy to be modified and extended *from a specific point of view* (a context). The the underlying object appears unchanged from the normal perspective - the changes are only seen when viewed from that specific context. The example in the documentation shows how an object hierarchy containing English text can be modified to reflect a German context - the same objects are now seen as having German text. So this is rather like the main purpose of Styles in LVGL - to say how a button changes when it is pushed, etc.

In my domain modelling work (involving formal type theory), there are two kinds of subtype: static subtypes and dynamic (or role) subtypes. Example of a static subtype: "a Cat is a kind of Animal". Role subtype: "a Teenager is a kind of Person". So an object of type Person can change type - they can become a Teenager (activate the role), and they can cease being a Teenager (revert). This is what is happening when a non-baseline Style is being applied in LVGL.

So the Aha! moment is that these are all cases of the same thing: type migration between related role subtypes. When a Style is applied, the attribute assignments are applied which make the style's role subtype different from its base type. This is how type migration is actually being implemented by LVGL styles.

The point of all that is that it shows a logical path to using ADL to support Styles, while still providing type safety. I'll have to do some more work to figure out the best use of ADL's syntax, but I think my previous example would look more like this:

If we assume that a Slider is defined in the metadata files something like this:
```
function_name: String;
color_name: String;
lv_obj: {
  x: Integer;  // No defaults here
  y: Integer;
  bg_color: color_name = "black"; // default colors for all objects
  color: color_name = "white";
}
lv_part: lv_obj {
  // Special characteristics of a Part is omitted here
}
style: {
}
slider: lv_obj {
  border_color: color_name;
  knob: lv_part;
  pressed event: function_name;
  pressed: style;  // This style is activated while the slider is being operated, and reverted afterwards
  checked: style;  // This style is toggled if the slider knob is double-clicked
}
```

Then an application might use these definitions like this:

```
style1: style {
  bg_color = "blue"
}
slider1: slider {
  x = 100;
  y = 200;
  color = "blue";
  knob.color = "light gray";
  pressed event = "my_func";
  pressed {
    knob.bg_color = "red";
  }
  checked {
    border_color = "amber";
    knob.color = "green";
  }
}
```

You can see how user-defined objects can define as many styles as they wish, and can activate and revert styles programmatically, including from inside event handler code. And all the above attribute assignments can be type-checked at compile time - the attribute must exist, and the value must look plausible.

Furthermore, themes just become an extension of styles. They change the default value assignments, for example:

```
bright mode: style {
  lv_obj {  // Customise the lv_obj defaults
    bg_color = "white";
    color = "black";
  }
  ...
}
```

A theme can be applied just by activating that style. All derived attribute values will be re-computed to reflect the new theme.

cjheath, 2021-09-23:
Now my response to your last message.

If you write in JSON something like:

```
Person {
    GivenName: { type: "String", default: "None"}
}
```

that adds a type annotation to GivenName, but it doesn't enforce the values assigned. It would be perfectly possible to write:

```
Person {
    GivenName: { type: "String", default: 23}
}

CliffordHeath : Person {
  BirthName = 9876.54;
```

There's no enforcement to the existence of the BirthName attribute, nor to the data type of a name (declared as a String but assigned as a number). Assigning a number is *not* ok just because it is possible to represent a number as a string. Type safe systems will prevent that.

> If so what is the problem with the style?

Th problem is that a style may contain an assignment to an attribute that only applies (e.g.) to a Button, but the style gets applied to a TextArea. In LVGL the style will be ignored, but in a type-safe system that style could not be applied, it would cause an error. But my previous post begins to address that; a style is a `role subtype`, and can only modify attributes (of the object and its parts) that are defined.

kisvegabor, 2021-09-23:
Wow, it sounds pretty interesting! Glad to hear that you see the advantages of  LVGL's style system.

There is one thing that makes style slightly more complicated. Let's say there are `checked` and `pressed` states. But there is `checked + pressed` (and combination of the states) too. E.g. a when there is a "checked" checkbox, and you press it, it will be in `pressed+changed` state.  See the for example (In pressed checked state the indicator has dark blue color)
[image]

An other typical example is `disable + checked`. Or once we implement "hovered" state it can be combined with any state. E.g. there is red button by default, but make it darker if `hovered`. It might have a green outline when focused and set the outline to dark green when `focused+hovered`.

In C it's is easy to handle. E.g. for a very specific state:
```c
lv_obj_add_style(obj, style, LV_STATE_FOCUSED | LV_STATE_CHECKED | LV_STATE_PRESSED);
```

Do you see a solution for this?

---

In your slider example `style1` is declared but not used. Can show an example for it too?

cjheath, 2021-09-23:
It is not a problem to activate more than one style (role subtype, or aspect) simultaneously. There is a practical question to ask however: If two style set the same attribute differently, which one applies? I suspect the easiest would be based on the order of activation, but that is not explicit in the original ADL definition. A better but more difficult rule would be to use the order of declaration in the ADL. Last one wins, in both cases.

The implementation issue remains: there is no existing programming language that supports something resembling subtype migration. This is also the reason I've had difficulty promoting and explaining the concept. AOP (Aspect Oriented Programming) has an angle on this, but it's not dynamic - it resolves to static types at compile time. Ruby now has "Refinements", but they're really not the same thing either. I have long considered extending ADL to form an aspect-oriented programming language (different meaning than established AOP). But even without a language that has built-in support, at least we can arrange for the inconvenient parts to be generated from a markup language, and activate/revert aspects using a convenient API. All good food for thought.

What do you need help with to start building the metadata tables?

kisvegabor, 2021-09-24:
> If two style set the same attribute differently, which one applies? I suspect the easiest would be based on the order of activation

Yes, it's the order of "activation". E.g. in `styles = [style1, style2]` the properties in `style2` wins.

> A better but more difficult rule would be to use the order of declaration in the ADL. Last one wins, in both cases.

It'd be far less convenient. The declaration of the styles can be very far in the code form the actual usage. Keeping track with it seems cumbersome. 

> The implementation issue remains: there is no existing programming language that supports something resembling subtype migration.

To be honest, I didn't understand everything from it :sweat_smile:  
Not sure I need to know all your knowledge about designing a language though :slightly_smiling_face: 

> What do you need help with to start building the metadata tables?

I need some example to see how the result should look like.

The widgets are built from 2 APIs:
1. All API of `lv_obj`: It's large so I need an example only about the interesting/tricky parts 
2. The widget itself: I need a complete implementation of a widget (e.g. slider)

The other important part is styles:
1. How local styles should look like on parts on states?
2. How "style sheet" properties should look like?

Based on these I can add some other widgets, e.g. label, button, image, switch, checkbox, etc. With these and the styles we will have the basic building blocks and we can try to implement UIs with ADL.

What do you think?

cjheath, 2021-09-24:
> The declaration of the styles can be very far in the code form the actual usage

For the kind of usage I envisage in a markup language, there is no separate style object. It's just an aspect of the thing it affects. It's certainly not defined "very far from the actual usage".

Re "subtype migration" - this is where an object that is defined with optional role subtypes begins or ends playing one or more of those roles (this is akin to activating or reverting a style). So the type is changing; not fundamentally, but in specific ways. Object-oriented languages mostly have completely static types. Some (Smalltalk, Ruby) allow a type (class) to be changed on the fly, but that affects all instances of the type. Support for role subtypes is much more organised than that.

> I need some example to see how the result should look like.

My message of 5 August shows how it can be done using ADL syntax.

> All API of lv_obj: It's large so...

Does all the behaviour of lv_obj apply to all objects, or is some of it disabled for some kinds of objects?

> What do you think?

I'm still feeling my way here, but I think even your YAML syntax would be better than nothing. ADL can do as much, and probably more, but it will take some playing to be sure.

kisvegabor, 2021-09-24:
> For the kind of usage I envisage in a markup language, there is no separate style object. It's just an aspect of the thing it affects. It's certainly not defined "very far from the actual usage".

Could show an example? E.g. 
1. add `style1` to a `slider1` 
2. add `style1` and `style2` to `slider2`

> My message of 5 August shows how it can be done using ADL syntax.

Okay, I start to work on it next week.

> Does all the behaviour of lv_obj apply to all objects, or is some of it disabled for some kinds of objects?

Yes. No all makes sens though. E.g. scroll related settings on a switch is unlikely but possible.

> I'm still feeling my way here, but I think even your YAML syntax would be better than nothing. ADL can do as much, and probably more, but it will take some playing to be sure.

It wasn't YAML just bullet lists :slightly_smiling_face: 

I'll try adding some basic stuff in ADL.

cjheath, 2021-09-25:
> Could show an example?

My message from 2 days ago showed metadata and a use case for slider. The metadata shows that slider recognises two styles (*pressed* and *checked*) and the example includes attribute assignments for each one. Now the API needs to have something like `lv_activate_style(slider1, "pressed")` to cause the *pressed* style to apply. This would probably happen in a click handler function inside the implementation of slider. I used "pressed" as a string here, but it could be a magic identifier generated from a code-generator which processed the metadata ADL.

> Okay, I start to work on it next week.

If you don't have data types, you can define them by assigning a regular expression to Syntax, or leave me a comment with the details and I will construct a regular expression. For example when setting *width*, it's not just an integer, there are different extensions you can add (px, %, etc) and those need to be defined in the Syntax. So define a new data type (e.g. Dimension) and we can allow whatever value syntax makes sense . The main thing is that whenever a value type has some restriction or variation from the base type, define a new data type, don't just use "Integer" and "String" everywhere.

If you want to drop examples into a gist, I will ensure that they work correctly with the ADL parser. I think you can make a shared gist that I can edit?

cjheath, 2021-10-09:
I don't know if others have made more progress on this, but I don't think it is stale yet. But I can't remove the label.

cjheath, 2021-10-16:
No problem, and thanks for making a start. I have gotten stuck into making support libraries for an ADL implementation in C++. Not the parser itself, but a suitable string and regular expression library; the C++ stdlib is terribly clumsy compared to what I need for this. I realise that ADL in Python will be needed for the development tools, but as I'm not much of a pythonista I'm delaying starting that. I have had some interested expressed in making a version in Go however.

Your gist shows (not surprisingly) a number of misunderstandings of the language. Is it possible to have a publicly editable gist, or should I fork it and respond? What's the best way of managing this process?

But I can make a couple of comments here (these need to be in an ADL tutorial instead of here):

My version had "Help Text : String". This defines a new type "Help Text" which is derived from String. Later I use it by declaring that a "Flag" includes a Help Text, and in each flag, I assign a value to the Help Text (which uses the Syntax of a String, because Help Text did not modify the inherited syntax.

Your version defines "help : string", so your new type is called "help". Later when you say that "flag" contains a "help text" there is no such type (you defined "help", not "help text"), so this will fail. If you delete the word "text" here, the assignments will work, but I recommend that you keep the compound name, because it is not clear when kind of thing a "help" might be, where as "help text" is clear.

When you say "enum: integer" you are defining a new type called "enum" which is an integer. This won't work; enums are not integers, and ADL has no enum type. What it has is collections of direct subtypes of a super type. So when I say "Alignment:;" I'm saying "there is a new type called Alignment which derives from the bare Object type". The subsequent lines define subtypes of Alignment, which correspond to the different possibilities in the enum. Subtype checking of assignments mean that a reference to an Alignment may be assigned only to Alignment itself (which is a null enum value) or to one of the subtypes. At first, this seems like a strange way to think about enumerations, but it actually makes more sense than the accepted approach of mapping them to integers. You have quite a lot of enums that need to be reformatted this way.

The general pattern of ADL is "new-type : base-type { children... }". If new-type is missing you get an anonymous object. If base-type is missing it is the bare Object root type. Then assignments follow "var = val", where var is the name of a child in (or inherited into) the current context. "var" must have (or inherit) a defined Syntax regular expression, and "val" must match that regular expression. "->" and "=>" are shortcuts to define a variable of type Reference, and to assign a default (type-restriction) value to that variable.

So to allow "width = 80%", we cannot use the Integer syntax. We need a new Coord type with a Syntax that allows the % suffix (or px, or whatever). There's a commented section about that in my previous example.

Where you say "style => style_props", you mean "style: { => style_props }". Which says "there is a kind of Object called style. It contains a multi-value reference variable called style_props, and every value must be a style_props or a subtype of style-props". If you don't use any shortcuts, the long form of this is "style: Object { style_props : Reference[] = .style_props }". Note that there are two "style_props" here; the type, and the reference variable that refers to instances of that type. I would recommend instead that the type should be called style_prop, and the variable called style_props, like this: "style: { style_props => style_prop }". The variable is plural, but its type is singular.

Don't worry if you don't understand the details of the shortcuts and their expansion. It is rather subtle and you can use it without knowing how it works.

excitedbox, 2021-11-04:
Wow it seems I forgot to hit post on my last reply again. My tests are on another laptop with a broken CPU fan. I need to fix that before I can get to them.

Have you looked how styles are done in[ Elements gui](https://cycfi.github.io/elements/layout)? It is a C++ gui lib.

This is what I mean by making it easier to translate a markup language to gui code:
`margin({ left, top, right, bottom }, subject)
`

This allows storing a pointer to a function called "margin" and when a style that sets the margin of an element is encountered, you pull the function pointer from a table, and insert the values, and "subject" which is the element id.

Then for the DOM you only need to store a list of the relevant function pointers and values to use with those functions to draw that element. This is a highly efficient way of drawing and storing styles and also reduces the amount of code needed because you no longer need a "margin" function for each and every element.

The above is basically what I did for my tests to see how quickly it would allow pulling all the styles needed to draw all the widgets/elements. 2 nested loops to simulate the DOM and updates to the styles. And 31 Arrays, holding pointers to Arrays that acted as widget lists, and style lists. 

ie. Array 1 is the 30 widgets being used. Array 2-31 is the 30 styles used by each widget and the values to use with those styles. 
-Loop 1 iterates through Array 1 (the widgets to draw)
-loop 2 iterated the right array for the current widget (the list of styles for that widget).

It is an Array of Arrays storage system and because no Array holds more than 255 values you can identify any object in the style system with a 1 byte ID as long as you know what type of object you are looking for. 

My previous reply was much longer and detailed. I can see if I still have the browser window open. I keep getting sidetracked mid response and lose track in the 400+ tabs.

cjheath, 2021-11-04:
That's really interesting @excitedbox. I look forward to seeing some more detailed examples. Perhaps you can push some of your experiments up to a github repo so we can read, test and comment on them there?

I'm also working on C++ code to support a new ADL implementation (with a goal of being small and tight for embedded use - not relying on any huge frameworks), and would value your feedback (especially since this is the first significant C++ I've written in 15 years!), so if you send me a PM we can chat. My email address is on my github profile.

excitedbox, 2021-11-06:
I actually just ran into this explanation of [control tables](https://en.wikipedia.org/wiki/Control_table) on wiki and this is basically the best way to explain it. 

That is how the DOM would work as an efficient way of implementing metaprogramming. It is how multiway branch prediction is done and how jump tables in C++ work. When you look at the list of advantages they give, it pretty much lists everything we are looking for but also mentions the extra overhead I mentioned, although that is partially solved in our case.

The rest of the style system works like in the graphic I posted in the other thread, so you can not only retrieve/execute functions but also handle styles, inheritance and all the other features. Let me know If you want me to post it again here.

excitedbox, 2021-11-07:
[This article ](https://rmbconsulting.us/publications/jump-tables-via-function-pointer-arrays-in-cc/)goes into some more details and gives some code examples on how you would setup the pointers and call them.

Once lvgl's functions all work on a system like this, it is pretty "trivial" writing a style definition language on top of it. ADL, CSS, something custom, will all just need a parser that converts it to the function pointers, creates a list of styles, and loops through them to draw the screen output. 

It is quite a lot of work on the front end, so it will require lots of planning. It would reduce the future work load tremendously, though. Several magnitudes, actually.

@cjheath I will send you an email. You can contact me on discord though if you want. I have the same username there as I do here.

tvanfossen, 2021-11-10:
One thing I dont see mentioned much in this thread is animations and view hierarchies. This may be an over generalization to make but:

- Many guis are going to have more than 1 screen
- Many guis are going to have animations
- Many guis will have transitions between screens

What should an individual file represent in an LVGL markup? Should it represent a screen? Should it represent a collection of screens? Should it represent only static placement on the screen? Is it completely agnostic to the screen? Is the file used to generate raw code, or is the file used to build a view at runtime? Should the animations be represented in a markup language or should it be used ONLY for static placement? If only for static placement, does every animation then need to be hand coded by a developer?

Who is the end user of an LVGL markup language? A designer or a developer? How does the markup language relate to Edgeline (important connection to understand imo)?

I'd argue for JSON (with a personal bias) as the base format, given it is highly customizable, easily recognizable, can be used in both C and Python with ease and can be kept extremely small. Alternatively, yaml is another format that could work, and allows for commenting, while being easily convertible between json and yaml and arguably more readable.

I've started putting together a JSON view factory in LVGL in an embedded application on an ESP32. Files are dropped into a spiffs filesystem and each file represents a view which is built at runtime. JSON parsing is done with CJSON in my application, but there are plenty of other options. Each view is effectively a collection of widgets. Each widget is defined by an LVGL type. Each  widget contains its own animations relative to the view and any values for static placement.

Example JSON widget for the schema I've started. 

{
            "type": "arc",
            "values":  //representing generic values tied to lv object types 
                "ai": { //representing object data stored in user data
                    "xo": 0,
                    "yo": 0,
                    "al": 0
                },
                "size": {
                    "width": 0,
                    "height": 0
                }
            },
            "arc_values": { //each widget type has its own set of specific values tied to LVGL properties
                "rotation": 270,
                "start_angle": 0,
                "end_angle": 360,
                "bg_start_angle": 0,
                "bg_end_angle": 360,
                "arc_stroke_width": 20,
                "arc_rounded": 0,
                "arc_indicator_color": {
                    "r": 255,
                    "g": 255,
                    "b": 255
                },
                "arc_bg_color": {
                    "r": 102,
                    "g": 102,
                    "b": 102
                },
                "arc_clickable": 0,
                "arc_knob": 0
            },
            "aos": [ //Collection of animations to trigger when the view containing this widget is created
                {
                    "p": "arc_diameter",
                    "f": 0,
                    "to": 180,
                    "t": 400
                }
            ],
            "aoe": [ //collection of animations to trigger when the view containing this widget is teared down
                {
                    "p": "arc_diameter", //property
                    "f": 180, //from
                    "to": 0, //to
                    "t": 400 //time
                },
                {
                    "p": "arc_value",
                    "f": "current",
                    "to": 0,
                    "t": 400
                },
                {
                    "p": "arc_color",
                    "f": "current", //obtains the current color of the arc
                    "to": {
                        "color": {
                            "r": 255,
                            "g": 255,
                            "b": 255
                        }
                    },
                    "t": 400
                }
            ],
            "adc": [ // collection of animations to trigger when a data change occurs while the screen is active
                {
                    "p": "arc_value",
                    "f": "current", //Obtains the current value of the arc
                    "to": "state_of_charge", //unique property for my application to take latest data from battery management system
                    "t": 400
                },
                {
                    "p": "arc_color",
                    "f": "current", //obtains current color of the arc indicator
                    "to": {
                        "color": {
                            "r": 139,
                            "g": 195,
                            "b": 74
                        }
                    },
                    "t": 400
                }
            ]
}

This is definitely more specific to my application than a general use schema, but it has been working very well for me so far. I can build up entire views from scratch in JSON format without writing a line of code. 

One bit I dont have implemented yet is a view_controller.json file, to build up a view transition handler without touching code, where I currently I have hard coded view transitions between a set number of views.

cjheath, 2021-11-11:
@tvanfossen, I just wanted to say that there's a lot I like about your post, even though it doesn't address some things I think are important. Type safety is one that I've mentioned previously.

I think it's important that we focus on what needs to change or be added to LVGL to allow markup languages to be constructed. Not that we choose and implement one markup language in particular. I think the existence of detailed machine-readable metadata (including data types) is the most important one.

fvanroie, 2021-11-12:
Our project uses a parser that reads [JSON lines](https://haswitchplate.github.io/openHASP-docs/latest/design/pages/) formatted files describing objects *(widgets)* on its pages. We mostly using local styles, so no *global* style lists or animations yet.

I have indeed started to adopt lists of function-pointers to get/set attributes on-the-fly, quite similar to the control tables that @excitedbox mentioned.

A file might look like this:
```json
{"page":1,"id":1,"obj":"label","x":5,"y":5,"h":50,"w":50,"text":"Hello","text_color":"red"}
{"page":1,"id":2,"obj":"btn","x":5,"y":90,"h":90,"w":50,"text":"World","enabled":false,"hidden":false}
```

The attribute names *(keys)* are hashed first using the [`sdbm` algorithm](http://www.cse.yorku.ca/~oz/hash.html) and compared to function-lists grouped by prototype:

```cpp
static hasp_attribute_type_t specific_bool_attribute(lv_obj_t* obj, uint16_t attr_hash, int32_t& val, bool update)
{
    { // bool
        hasp_attr_update_bool_const_t list[] = {
            {LV_HASP_ARC, ATTR_ADJUSTABLE, my_arc_set_adjustable, my_arc_get_adjustable},
            {LV_HASP_BTNMATRIX, ATTR_ONE_CHECK, lv_btnmatrix_set_one_check, lv_btnmatrix_get_one_check},
            {LV_HASP_LINE, ATTR_Y_INVERT, lv_line_set_y_invert, lv_line_get_y_invert},
            {LV_HASP_LINE, ATTR_AUTO_SIZE, lv_line_set_auto_size, lv_line_get_auto_size},
            {LV_HASP_IMAGE, ATTR_AUTO_SIZE, lv_img_set_auto_size, lv_img_get_auto_size}};
        if(do_attribute(list, obj, attr_hash, val, update)) return HASP_ATTR_TYPE_BOOL;
    }

    { // bool but obj is not const
        hasp_attr_update_bool_t list[] = {
            {LV_HASP_DROPDOWN, ATTR_SHOW_SELECTED, lv_dropdown_set_show_selected, lv_dropdown_get_show_selected},
            {LV_HASP_IMAGE, ATTR_ANTIALIAS, lv_img_set_antialias, lv_img_get_antialias}};
        if(do_attribute(list, obj, attr_hash, val, update)) return HASP_ATTR_TYPE_BOOL;
    }

    return HASP_ATTR_TYPE_NOT_FOUND;
}

static hasp_attribute_type_t specific_coord_attribute(lv_obj_t* obj, uint16_t attr_hash, int32_t& val, bool update)
{
    { // lv_coord_t
        hasp_attr_update_lv_coord_t list[] = {
            {LV_HASP_IMAGE, ATTR_OFFSET_X, lv_img_set_offset_x, lv_img_get_offset_x},
            {LV_HASP_IMAGE, ATTR_OFFSET_Y, lv_img_set_offset_y, lv_img_get_offset_y},
            {LV_HASP_IMAGE, ATTR_PIVOT_X, my_img_set_pivot_x, my_img_get_pivot_x},
            {LV_HASP_IMAGE, ATTR_PIVOT_Y, my_img_set_pivot_y, my_img_get_pivot_y},
            {LV_HASP_DROPDOWN, ATTR_MAX_HEIGHT, lv_dropdown_set_max_height, my_dropdown_get_max_height}};
        if(do_attribute(list, obj, attr_hash, val, update)) return HASP_ATTR_TYPE_INT;
    }

    return HASP_ATTR_TYPE_NOT_FOUND;
}
```
Depending on the `update` boolean either the get or set function is executed on the `val` which is passed by reference. Whenever the prototypes don't match up, I write a small wrapper function with the `my_`-prefix instead of `lv_`.

`do_attribute` is a Template, as the logic is the same: It iterates over the items as the performance is OK for our application:
```cpp
template <typename T>
static inline bool do_attribute(T& list, lv_obj_t* obj, uint16_t attr_hash, int32_t& val, bool update)
{
    uint8_t obj_type = obj_get_type(obj);
    int count        = sizeof(list) / sizeof(list[0]);

    for(int i = 0; i < count; i++) {
        if(obj_type == list[i].obj_type && attr_hash == list[i].hash) {
            if(update)
                list[i].set(obj, val);
            else
                val = list[i].get(obj);
            return true;
        }
    }

    return false;
}
```

My initial suggestions for improvement would be:
- **Standardize protoypes for get/set functions by return type**
Some get functions use `(lv_obj_t * obj)` as a single parameter while others use `(const lv_obj_t * obj)`. I think it's a bit random at the moment. I would suggest using `(const lv_obj_t * obj)` for getter functions. Then *-in my example above-* I could use one list instead of two *(const/non-const)*. I think this change can even be transparent for most users who don't use the prototypes directly...
e.g.: `lv_coord_t lv_dropdown_get_max_height(const lv_obj_t * ddlist)` vs `lv_coord_t lv_img_get_offset_x(lv_obj_t * img)`

- **Provide *both* getter and setter function in the lvgl api**
This eliminates the need for helper functions that manipulate the `ext` properties of objects directly. `ext` might break between versions and require tweaking to match. If both are available in the api, it will make getting/setting values a lot easier and consistent.
e.g.: `lv_dropdown_get_symbol`, `lv_dropdown_get_max_height`

- **Provide getter/setter on *individual* parameters**
Now, sometimes, a function might change two or more parameters at the same time. This makes it harder to set these parameters iteratively. These can coexist with functions that set multiple parameters where it makes sense.
e.g.: `my_img_set_pivot_y`, `my_img_get_pivot_y` to set the individual coordinate.

I agree with @cjheath that the initial focus should be on standardizing the api calls/prototypes. The actual implementation of a markup language is a next step up, but the fundamentals need to be solidified first. Documenting the current schema will likely expose other areas where simplification/standardization is needed.

kisvegabor, 2021-11-12:
I agree that the current goal is the develop the environment (i.e. improve lvgl) to make it possible to create any kind of markup language (officially supported or custom). The first step is to understand inconsistencies in the current API and build a platform/language-independent description of the current API. In this description, we can map "names" to "functions". E.g. `x` should be resolved as `lv_obj_set_x`. (i.e. a jump table)

@tvanfossen 
So at this stage, I have no idea how to handle animations and screens because I think it should be a later topic.

@fvanroie 
Do you still use v7?

fvanroie, 2021-11-14:
> Do you still use v7?

Yes I do for the main project. I've been working on migrating to v8 though but that got deferred.

excitedbox, 2021-11-16:
To make any markup language possible lvgl needs to use the kind of api I described above with functions that draw the specific styles (ie. background_color(object_id, color) or some form of this). It is the only way to efficiently make a style system.

The official DOM spec is written in WEB IDL available with standardized mappings on how to implement them in C. If billion dollar corporations got together and decided this is the best way to do it, it seems pretty silly not to use it as well. Implementing the DOM and CSS according to the OMG CORBA mappings also makes LVGL compatible with C++, JAVA, PYTHON, UML, and industry specs for security, robotics, system modeling, etc.

[DOM SPEC](https://dom.spec.whatwg.org/) (I would read the below first and work off this when implementing)
[Corba 3.1 ISO Spec](https://www.omg.org/spec/CORBA/3.1.1/) (This explains how you would structure things)
[C mappings](https://www.omg.org/spec/C/1.0/) (These are the C mappings but they are mostly covered in the above)
[Corba Embedded](https://www.omg.org/spec/CORBAe)
[WEB IDL SPEC](https://webidl.spec.whatwg.org/) (This is the most modern spec used to describe DOM and CSS.

Looking at the [CSSOM](https://www.w3.org/TR/cssom-1/) and CSS 2.2 spec will be helpful for implementing the BASIC CSS. They also contain lexer indexes which means making a parser to work with the DOM will be much easier. We can strip out any styles we don't need for lvgl.

[All the other DOM, CSS, CSSOM, CSS API, etc specs can be found here.](https://www.w3.org/TR/)

JSON is the absolute worst option for a style system in my opinion. It is not meant to be used to define styles. Not trying to offend anyone, but I believe it is best if the underlying technical decisions are made by people with experience in those things and the wider implementation (ie function naming, api specification) is open to wider discussion. I wouldn't try to tell my doctor what tools to use to fix a broken leg either. We can't spend the next year teaching everyone all the different technology and standards, and this will also eliminate questions about features that are already available in the suggested options.

Json is meant for passing data and not for defining a layout, let alone a DOM. I think CSS is the best options so far. CSS is widely known, already has ALL the features we need and more. The upside to using ADL is type safety, but nobody has experience with it. It is actually obsolete/deprecated/abandoned from what I gather. It is also not a modern language, so my vote would still have to go to CSS. Second option is XML (it sucks, but is industry standard QT, .Net, GTK/glade etc. all use it).

@tvanfossen CSS has [animation](https://www.w3.org/TR/css-animations-1/), inheritance (hierarchy), multiple pages/screens is done by applying a new style to your screen element. 

To design a layout you need 2 files (1 is possible but messy). 1 defines styles, the other layout (think HTML). 

Your layout would simply be a list written as widget_name and the style_class it uses. A style class being a group of styles and this would only be needed for launching the program. The file is parsed and used to generate the DOM.

`Example:

screen:scnStyle;
mytabs:mytab_style;
tab1:background;
tab1_label:mytab_label;
tab1:content;
tab2;tab2_styles;
`

This way the order doesn't matter. naming conventions don't matter. It just needs to match your style sheet naming and be written in the right format (I assumed objectName:styleName; in this example). It is the same as using a <div class="styleName"> to build your entire layout.

I will make a list tomorrow of the important CSS spec documents. I think the CSS Layout Level1 API etc. is less known because they deal more with the code behind the browser/server and these specs are more hidden.

excitedbox, 2021-11-16:
@kisvegabor Right, but we need to plan ahead so that we can build on it without making massive changes. I think the functions should be lv_obj(action, item_id, property, value) otherwise the markup language needs 4x as many functions (get, add, edit, delete). I am not 100% sure though, which is why I posted the above specs.

If you read the [CORBA embedded spec](https://www.omg.org/spec/CORBAe/1.0/PDF) and then the DOM spec, I think you will get a much better understanding of how to structure lvgl so that the actual widget drawing will run much faster.

If you implement that it could all be done at once with much less work than splitting it into two updates. The improvements to lvgl would be enourmous as well from a performance standpoint.

PS. They are using the same structure as I previously suggested, but their description is much easier to understand than mine. Read pages 4-8 extra closely.

cjheath, 2021-11-16:
@excitedbox makes a lot of ill-founded or incorrect assertions here. I'm glad to see enthusiasm, but let's not allow it to run away with us.

> To make any markup language possible lvgl needs to use the kind of api I described above

No. That is just one of a number of possible ways. If it's not the only possible way, it doesn't *need* to use any one of them.

> The official DOM spec is written in WEB IDL available with standardized mappings on how to implement them in C. If billion dollar corporations got together and decided this is the best way to do it, it seems pretty silly not to use it as well. 

Billion dollar corporations join consortia like this for the *specific reason* of producing vast rambling specifications that are so difficult to implement that only a billion dollar consortium can do it. This is a well-documented pattern, and it is not a good recommendation. I lived through CORBA's active life (and have friends who were principal engineers in some of the most successful CORBA tools companies), and it is exactly this kind of technology, even if IDL is no worse than many other options for this job.

The WWW DOM is a rambling jungle of legacy implementations of unrelated widgets that seemed a good idea at the time, but no-one can quite let go of because it's deployed in every browser. CSS is a passable way to make sense of that mess, but it would be much better if the mess had not been created in the first instance. LVGL has not yet fallen down this hole, and it would be good if we can prevent that.

CSS is derived from the unusable mess that was X resource files https://en.wikipedia.org/wiki/X_resources - an idea that was far too powerful and low-level to see much effective use, and far too disconnected from the widget hierarchy of an application. In short, it was a good idea, but it needed a different ecosystem to work properly.

A good user interface toolkit needs a small set of building blocks that can be arrayed and composed into detailed features, and the styling system needs to work on the exterior of these custom features - hiding or re-mapping the internal elements that make them up. This is what is wrong with the WWW's DOM. I cannot make re-usable composite widgets that act like natural DOM objects. This is something we got right with OpenUI way back in 1990.

> CSS is widely known, already has ALL the features we need and more.

CSS selectors are the main drawback, in my view. They are quite difficult and expensive to implement efficiently because of the amount of wildcard matching involved. Almost all the algorithms are O(n^2) at least, and it's just not necessary or appropriate on an embedded target like an ESP32. The rest of CSS (the body of definitions) are quite bland and unoffensive, except that there is no way to know what keys or values will be accepted. It's possible to add "a: b;" for any "a" and any "b" and most of the tooling won't tell you anything is wrong - the declaration just won't do anything. In the 21st century, we have computers that know exactly what is wrong, but don't tell us what it is - unforgivable!

> The upside to using ADL is type safety, but nobody has experience with it. It is actually obsolete/deprecated/abandoned from what I gather. It is also not a modern language,

On the contrary, though it based on old ideas, it is a brand-new language still being actively implemented (I spent just yesterday on it in fact). As I said to you in our Discord conversation not ten days ago "ADL is something I invented and have reimplemented a number of times since 1983" - so I don't know why you have to go simply making up rubbish like this. Implementation requires Unicode support and a certain kind of regular expression library, so it needs the kind of first-principles tooling for a proper implementation. I'm working on that (for C++) every day at present: https://github.com/cjheath/strpp

A Python implementation (very suitable for dev tooling) would be a couple of day's work, but I'm currently focussed on C++ because I want an embedded implementation to make it simple to implement re-skinnable user interfaces. For my purposes the markup must be efficient to interpret on an embedded target.

X-Ryl669, 2021-11-19:
Seems like there are 2 discussions embedded in this issue.
The first one is mainly:
  1. Can we orthogonalize LVGL's objects method and properties so the API space (and likely binary space) is reduced, as it would ease documenting/making a interface of all of them ?
  2. If so, what would be the best form for this work (having a kind of generic `setPropertyFor(obj, prop, value)`) or instead a key/value&function pointer like solution
  3. Should the obj's method and properties be queryable/enumerable ?
  4. Is there a tool that could crawl the source code and make such work automagically ?

The second discussion is:
  1. Once we have a minimal DOM for all objects, how to specify a format to express the scene using them.
  2. Should event be specified in this format, and if yes, should it include code or only refer to compile time code ?
  3. Is it type safe ? Can it be compiled / type checked at build time and so the runtime does not have to do it (and be smaller) ?
  4. What would be the tooling provided? If none, what will be the effort required on the developer to use it ?

To answer the first discussion, I think a regex guru can probably write some expression that'll extract 80 to 90% of the properties in use of each object/widget from the code and the remaining stuff can be fixed by hand. Once this is done, it's probaby possible to sort the list and try to find common pattern and rework the code to re-use the common pattern more.
Maybe the method/properties can be documented in the code with some specific marker in the comments so the regex work will be easier and improve coverage (and also helps documentation).

Having a generic function with string parsing to method call is not a good idea for embedded system, unless using compile time string hash and switch/case based on this hash value (it's very simple in C++, less so in C). 
A key/ptr pair is typically used in dynamic virtual table (like glib), and is also possible to use, but the overhead is not negligible for embedded code (in term of binary space, not run time). It can be good if functions are removed to compensate.

To answer the second discussion, I think it's not a good idea to embed behavior in the styling concept. Behavior should be specified in code, not style. In CSS, there is only a limited number of "flag" for the behavior (like hover, visited). I think that's correct from a usage point of view, a styling concept should describe what is the style in case there is no action from the user (like hover, visited, visited+hover). If we need to specify event, give them an UID and let the code map them to action.

Ensuring type safety is good but the runtime overhead is too large IMHO. So maybe the style format file can be checked on the dev's computer and binary converted (if correct) to be read by the embedded runtime. Unlike what @cjheath asserted, CSS is typesafe with fallback, if a rule is not understood, a warning is emitted and the rule is ignored. I think this is very convenient.

Using ADL is, IMHO not adequate here because:
1. It's not known by developer and the cognitive effort to learn it is too high.
2. ADL mixes both discussion/concept here: the DOM and the style. There's a reason why both are 2 different files everywhere else: people working on some interface don't care about the final style (they can continue working on the behavior even if the buttons are ugly), and people working on style don't care about the behavior. 

So ADL would be useful if it were generated by a UI design tool (like Edgeline / GUIder) and compiled to both files to be digested in embedded runtime. It would be good if a tool existed that would parse the LVGL source code and build the interface automatically, but if it has to be maintain in addition to the source code, there'll be inconsistencies and it'll fail in the end.

CSS is not good either here since parsing CSS is not trivial, and there's no 1:1 mapping between LVGL's objects and CSS.
There used to be a C++ toolkit using CSS for styling (Rocket something) and it failed, because, well, at some point, the divergences between the standard CSS and the widget are just too big and can't be solved.

As a simple, temporary solution, the "setProperty" solution from first discussion could be used as a dumb "validator" of a YAML or plain key/value text file, if it's written in a way to reject any invalid style rule. All LVGL objects will have to be very very strict with what they accept and how they fails when invalid input is provided.

X-Ryl669, 2021-11-19:
In that case, the Simulator would be the validating too, when trying to load the file.

cjheath, 2021-11-20:
Thank-you X-Ryl669 for partitioning this discussion into the separate issues. Hopefully we can focus on what's important here without needing to create a new thread.

I must say I do find it amusing to see such over-confident assertions about what ADL is or isn't and can or cannot do. I cannot expect anyone to know this - it is a language unlike any that has existed before, despite its apparently familiar appearance. Most of what you said about it (except unfamiliarity) is wrong. Please stop assuming, and just drop this part of the discussion. I'll implement it when I can, and will enjoy using it even if no-one else does.

What is needed here is to find an effective way to standardise the API and provide metadata that a markup language can use.

I suggest that instead of trying to systematically extract metadata (DOM) from code (using regular expressions or any other method), we find a way to create the code skeletons from metadata that is expressed in any suitable language. We obviously need to build the metadata first, but to try to automate it is likely to be trouble-prone in the long term.

The metadata does not have to be included at runtime. It is possible to use it to implement static checking that removes the need for runtime type checking. So, contrary to it being too expensive at run-time, the opposite is true.

Anyhow, I hope I haven't misunderstood what you wrote. It's possible, and I'm open to further clarifications.

[175 more replies]

closed 2024-06-20 by kisvegabor