lvgl/lvgl #2430: lv_win border is cut

amirgon, 2021-08-03:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When setting a border and radius to a `win` object, the corners radius is cut 

### To Reproduce

```python
scr = lv.obj()
win = lv.win(scr, 20) # Replace to lv.obj to behave correctly
win.set_size(lv.pct(50),lv.pct(50))
win.set_style_border_width(10, lv.PART.MAIN)
win.set_style_border_color(lv.color_hex3(0), lv.PART.MAIN)
win.align(lv.ALIGN.CENTER, 0, 0)
win.set_style_radius(40, lv.PART.MAIN)
lv.scr_load(scr)
```

### Expected behavior

Avoid cutting corner radius.
If changing the `win` object to `obj`, it behaves correctly.

### Screenshots or video

win behaves like this:
[image]

but obj behaves like this:
[image]

kisvegabor, 2021-08-03:
You need to add this 2 lines too:
```py
win.set_style_border_post(True, lv.PART.MAIN)  #to draw border when all children are drawn
win.set_style_clip_corner(True, lv.PART.MAIN)    #to clip the overflowing children on the corners 
```

By default, the window's "main" object is transparent and has no style because the header and content object fully covers it.
So if you start to add styles yourself they need to be added carefully.

amirgon, 2021-08-03:
> You need to add this 2 lines too:

It works! I never would have guessed these two lines.
Why not set these as default for window?

kisvegabor, 2021-08-03:
`border_post` could be added but as there is no border by default I saw no point to do so.

Enabling `clip_corner` affects the performance because LVGL needs to apply a mask on every children of the window because they might be clipped on the corners.

embeddedt, 2021-08-03:
> Enabling clip_corner affects the performance because LVGL needs to apply a mask on every children of the window because they might be clipped on the corners.

True, but shouldn't there be no performance penalty if border_radius == 0? In that case I don't see an issue with enabling it by default, since the window has no border radius by default.

kisvegabor, 2021-08-03:
> True, but shouldn't there be no performance penalty if border_radius == 0? In that case I don't see an issue with enabling it by default, since the window has no border radius by default.

So no penalty by default, but if someone enables it will work as expected. Sounds good.

BTW, I've realized that the performance drop can be hugely mitigated if LVGL checked if the existing masks are actually affecting the area being drawn. E.g. when a letter is drawn in the middle of a corner clipped window no masks needs to be considered.  I'm planning to add it in a few days.

kisvegabor, 2021-08-18:
A few days ago I've optimized the clip corner and now enabled clip_corner and border_post by default for the window.

commit 493ace35 (2021-08-18): fix(win) enable  clip_corner and border_post by default
fixes #2430
closed 2021-08-18 by kisvegabor