lvgl/lvgl #2432: lv_obj

Miguel0101, 2021-08-04:
### Just a question

Is there a way to drag an object without scrolling?

kisvegabor, 2021-08-04:
See this example: https://docs.lvgl.io/master/widgets/obj.html#make-an-object-draggable

---

We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

Miguel0101, 2021-08-04:
> See this example: https://docs.lvgl.io/master/widgets/obj.html#make-an-object-draggable
>
> We use GitHub Issues for development-related discussions.
>
> Please ask your question in the [...]

When I try to drag the object, it is scrolled at the same time.

kisvegabor, 2021-08-06:
Adding this to the dragable object should do the job:
```
lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLL_CHAIN);
```

Miguel0101, 2021-08-06:
> Adicionar isso ao objeto arrastável deve fazer o trabalho:
> 
> ```
> lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLL_CHAIN);
> ```

Sounds good