lvgl/lvgl #2433: Add video player widget

donny681, 2021-08-05:
### Introduce the problem
Do you plan to add video player widget.I want to play video on lvgl,but I cannot find interface.

Miguel0101, 2021-08-06:
Must be a third-party library

kisvegabor, 2021-08-06:
> Must be a third-party library

I agree, there are too many ways to get the frames of a video. From LVGL's point of view lv_canvas can be used to display the frames.

Anyway, an implementation with a 3rd party library (as demonstration) would be great!

Rajssss, 2021-08-19:
> > Must be a third-party library
> 
> I agree, there are too many ways to get the frames of a video. From LVGL's point of view lv_canvas can be used to display the frames.
> 
> Anyway, an implementation with a 3rd party library (as demonstration) would be great!

An example will be great too, with basic OSD :)

closed 2021-09-10 by lvgl-bot