lvgl/lvgl #2435: Lvgl color display error, this is not the color I want

superCong, 2021-08-05:
Lvgl color display error, this is not the color I want

code：
    lv_obj_t * label1 = lv_label_create(lv_scr_act());
    lv_label_set_long_mode(label1, LV_LABEL_LONG_WRAP);     /*Break the long lines*/
    lv_label_set_recolor(label1, true);                      /*Enable re-coloring by commands in the text*/
    lv_label_set_text(label1, "#0000ff Re-color# #ff00ff words# #ff0000 of a# label, align the lines to the center "
                              "and wrap long text automatically.");
    lv_obj_set_width(label1, 150);  /*Set smaller width to make the lines wrap*/
    lv_obj_set_style_text_align(label1, LV_TEXT_ALIGN_CENTER, 0);
    lv_obj_align(label1, LV_ALIGN_CENTER, 0, -40);

    lv_obj_t * label2 = lv_label_create(lv_scr_act());
    lv_label_set_long_mode(label2, LV_LABEL_LONG_SCROLL_CIRCULAR);     /*Circular scroll*/
    lv_obj_set_width(label2, 150);
    lv_label_set_text(label2, "It is a circularly scrolling text. ");
    lv_obj_align(label2, LV_ALIGN_CENTER, 0, 40);

[image]

superCong, 2021-08-05:
The color seems to be the opposite?

embeddedt, 2021-08-05:
Try enabling `LV_COLOR_16_SWAP` if you're using 16bpp.

superCong, 2021-08-05:
Sorry, I ignored the width and height of st7789 driver, resulting in incorrect display color

closed 2021-08-05 by superCong