lvgl/lvgl #2438: Error dragging object with align center

Miguel0101, 2021-08-06:
https://user-images.githubusercontent.com/56565368/128512602-eec4b43e-78bf-4d00-8072-caa0ed12d786.mov

embeddedt, 2021-08-07:
Hmm... I think this is invalid. You cannot simultaneously change an object's position with `set_pos` and also align it, as only one of those can be respected at a time.

Miguel0101, 2021-08-09:
> Hmm... I think this is invalid. You cannot simultaneously change an object's position with `set_pos` and also align it, as only one of those can be respected at a time.

Like this?

kisvegabor, 2021-08-09:
Actually it's valid. `lv_obj_set_pos` setc X and Y by respecting the set `align`. But there is an inconsistency in LVGL and I still don't know how to resolve it (ideas are welcome :slightly_smiling_face:  )

The problem is that `lv_obj_get_x/y` doesn't respect `align` and always returns the distance of the top left corner. E.g.if you align the object to the center and `lv_obj_get_x/y` will return e.g. 100, 200 instead of 0,0. If you apply 100, 200 it will move the object away from the enter with 100, 200.

Maybe we need functions like `lv_obj_get_x/y_aligned`.

Miguel0101, 2021-08-09:
Could it be like this `x/y -= lv_obj_get_style_align(parent, LV_PART_MAIN)`?

Miguel0101, 2021-08-10:
> Maybe we need functions like `lv_obj_get_x/y_aligned`.

I agree.

kisvegabor, 2021-08-11:
I've added them.

closed 2021-08-11 by kisvegabor

Miguel0101, 2021-08-16:
> I've added them.

Can we rename lv_obj_get_x/y_aligned to lv_obj_get_x/y?

embeddedt, 2021-08-16:
I'm pretty sure that would be backwards-incompatible.

Miguel0101, 2021-08-16:
> I'm pretty sure that would be backwards-incompatible.

How like?

embeddedt, 2021-08-16:
The [current implementation of `get_x`/`get_y`](https://github.com/lvgl/lvgl/blob/51a82a17ffe938d07d94660f49fd18962060943a/src/core/lv_obj_pos.c#L481-L497) takes other factors into account like scrolling and padding, whereas [`get_x_aligned`/`get_y_aligned`](https://github.com/lvgl/lvgl/blob/98bc1fe09e12a64333e91b4c25327c283a700af5/src/core/lv_obj_style.h#L225-L233) are just returning the X/Y style property, which is what you [previously set with `set_x`/`set_y`](https://github.com/lvgl/lvgl/blob/98bc1fe09e12a64333e91b4c25327c283a700af5/src/core/lv_obj_pos.c#L63).

Miguel0101, 2021-08-16:
> The [current implementation of `get_x`/`get_y`](https://github.com/lvgl/lvgl/blob/51a82a17ffe938d07d94660f49fd18962060943a/src/core/lv_obj_pos.c#L481-L497) takes other factors into account like [...]

Ok, thanks for your reply.

commit 98bc1fe0 (2021-08-11): feat(obj) add lv_obj_get_x/y_aligned

fixes #2438