lvgl/lvgl #2439: [OSX] Build failure if custom mem management is used

snazarkin, 2021-08-06:
### Perform all steps below and tick them with [x]
- [ X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ X] Update lvgl to the latest version
- [ X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I use OSX Catalina to make example for SDL backend (simulator). If LV_MEM_CUSTOM=1,  I get following error  on linking stage (I shrink the log since it contain all references to the roots listed at lv_gc.h)
```c
  "__lv_timer_ll", referenced from:
      __lv_timer_core_init in liblvgl.a(lv_timer.c.o)
      _lv_timer_handler in liblvgl.a(lv_timer.c.o)
      _lv_timer_create in liblvgl.a(lv_timer.c.o)
      _lv_timer_del in liblvgl.a(lv_timer.c.o)
      _lv_timer_get_next in liblvgl.a(lv_timer.c.o)
  "_lv_mem_buf", referenced from:
      _lv_mem_buf_get in liblvgl.a(lv_mem.c.o)
      _lv_mem_buf_release in liblvgl.a(lv_mem.c.o)
      _lv_mem_buf_free_all in liblvgl.a(lv_mem.c.o)
     (maybe you meant: _lv_mem_buf_free_all, _lv_mem_buf_get , _lv_mem_buf_release )
```
Hovewer I observe no error if set  LV_MEM_CUSTOM to 0. And there is no error if compilation is done on linux host. 

I checked lv_gc.o, it contains all referenced symbols in both cases. 

### To Reproduce
Compile any example with  LV_MEM_CUSTOM=1 on OSX host

### Expected behavior
Binary link should complete without error.

### Screenshots or video

kisvegabor, 2021-08-06:
Please modify `lv_gc.c` to see if those variables are really declared:
```
#if(!defined(LV_ENABLE_GC)) || LV_ENABLE_GC == 0
#error "DECLARING GC ROOTS"
    LV_ROOTS
#endif /*LV_ENABLE_GC*/
```

snazarkin, 2021-08-06:
That was one of the things I did for debugging. Compilation of `lv_gc.c` is surely failed
`lvgl/src/misc/lv_gc.c:30:2: error: "DECLARING GC ROOTS"
#error "DECLARING GC ROOTS"
 ^
1 error generated.`

kisvegabor, 2021-08-06:
@amirgon have you already seen an issue like this?

What about this? It tests if the variable is actually declared:
```c

#if(!defined(LV_ENABLE_GC)) || LV_ENABLE_GC == 0
    LV_ROOTS

    volatile lv_mem_buf_t * __lv_test_gc_error = &lv_mem_buf[0];
#endif /*LV_ENABLE_GC*/
```

snazarkin, 2021-08-06:
That doesn't help as well. But if `LV_ROOTS` is moved to other module (like `lv_mem.c`), the link completes successfully.

embeddedt, 2021-08-06:
In that case, it sounds like `lv_gc.o` is not being linked in your build system for some reason.

kisvegabor, 2021-08-06:
@snazarkin 
> That doesn't help as well.

It means there was or wasn't a build error? :slightly_smiling_face:

snazarkin, 2021-08-06:
> @snazarkin
> 
> > That doesn't help as well.
> 
> It means there was or wasn't a build error? 🙂

Sorry, forget to mention that  the file is compiled but linker failed with same output.   I'm going to try with GCC, will see if it helps

snazarkin, 2021-08-06:
Forget to mention I uses `cmake` as build system.  In conjunction with clang it gives such artifacts. Hovewer there is workaround:
```
$git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index aa54a409..e38d7f51 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -59,8 +59,8 @@ zephyr_library_sources(${SOURCES})
 endif() # CONFIG_LVGL

 else()
-
+add_compile_options(-Wall)
 file(GLOB_RECURSE SOURCES src/*.c)
-add_library(lvgl STATIC ${SOURCES})
+add_library(lvgl OBJECT ${SOURCES})

 endif()
```

snazarkin, 2021-08-06:
BTW,  build with GCC reveal no issue.

amirgon, 2021-08-07:
> @amirgon have you already seen an issue like this?

I'm not familiar with this. But it looks like the problem is already resolved, right?

snazarkin, 2021-08-09:
I'd say it is workarounded.  Not sure if it's worth to add to docs.

kisvegabor, 2021-08-09:
If you also agree, I'd close this issue. If someone will have the same problem he can find this issue even is google.
And we can reopen it if comes up for others too.

closed 2021-08-24 by snazarkin