lvgl/lvgl #2441: /*clang-format off*/ not recognized

higaski, 2021-08-07:
Various headers (e.g. [lv_conf_template.h](https://github.com/lvgl/lvgl/blob/50ea6fb3fefb3a6edc958154c575dcdcacbfdb3a/lv_conf_template.h)) use 

`/*clang-format off*/`

to supposedly disable clang-format. This however does not work, at least not with clang-format 12.0.1.
According to the clang-format [documentation](https://clang.llvm.org/docs/ClangFormatStyleOptions.html) there are two whitespaces missing and this should be changed to 

`/* clang-format off */`

embeddedt, 2021-08-07:
Thanks! These are leftovers from the brief period (v6) where we used clang-format. I have corrected the directive in `lv_conf_template.h` (since this file is customized by downstream projects) and removed it from all other human-made headers.

closed 2021-08-07 by embeddedt

higaski, 2021-08-08:
May I further suggest to move the clang-format off comment up a little?
At the position it currently stands at it still allows clang-format to indent the first preprocessor directives.

kisvegabor, 2021-08-09:
> May I further suggest to move the clang-format off comment up a little?
At the position it currently stands at it still allows clang-format to indent the first preprocessor directives.

Done here 3c86d777

commit 2122583e (2021-08-07): style fix usage of clang-format directives

We do not use clang-format anymore, so the directive has been removed from most headers.

lv_conf_template.h is untouched as a convenience for downstream projects which still use the tool.

Fixes #2441