lvgl/lvgl #2443: Refreshed area too big when spinner animation restarts

jakob-gen, 2021-08-09:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When using a spinner or arc, I noticed that the area lvgl refreshes at the start of the animation is bigger than it should be.

I've added a pixel counter inside the `flush_cb` callback of my display port and I can see that lvgl refreshes the entire rectangle around the spinner at the start of the animation. For example, if I create a spinner with a size of 60x60 pixel and an arc of 60 degrees (see example code), lvgl will refresh an area of 4096 pixels at the start of the animation which roughly equals the rectangle around this spinner. But the refreshed area should only include the rectangle around the arc.

I have attached the lvgl trace output. Just search for "disp_flush: refreshed 4096 px".
[lvgl_refresh_px.log](https://github.com/lvgl/lvgl/files/6954123/lvgl_refresh_px.log)

### To Reproduce
The code to create the spinner:
```c
static lv_style_t style_spinner;
lv_style_init(&style_spinner);

lv_style_set_arc_width(&style_spinner, 5);
lv_style_set_arc_rounded(&style_spinner, true);

lv_obj_t *spinner = lv_spinner_create(lv_scr_act(), 1000, 60);
lv_obj_set_size(spinner, 60, 60);
lv_obj_add_style(spinner, &style_spinner, LV_PART_INDICATOR);
lv_arc_set_rotation(spinner, 270);
lv_arc_set_mode(spinner, LV_ARC_MODE_REVERSE);
lv_obj_center(spinner);
```

I have added the following to output the pixel count in the flush callback:
```
unsigned int pixel_count = 0;

for (y = area->y1; y <= area->y2; y++) {
    for (x = area->x1; x <= area->x2; x++) {
        pixel_count++;
        ...
        color++;
    }
}

LV_LOG_TRACE("refreshed %d px", pixel_count);
```

### Expected behavior
Fewer pixels should have changed (in the range of 120 to 250).

kisvegabor, 2021-08-09:
I've pushed a fix. Now it work well for me. 

Just  out of curiosity, how did you spot this isseu?

closed 2021-08-09 by kisvegabor

jakob-gen, 2021-08-09:
Wow! You are incredibly fast :D thank you! I will check it tomorrow.
I was seeing some "stutters" when I was flushing the buffer's contents to the display. As you might have seen in the logs it took well over 100 ms to flush it. That's when I got suspicious :)

higaski, 2021-08-16:
I've run into the same issue just now and I fear that https://github.com/lvgl/lvgl/commit/0f14f49465ca701c98f76ac95bda4a537c0fadfa doesn't fix everything.

The parameters start_angle and end_angle might actually be bigger than 360 deg.  An easy way to repro this is by taking lv_example_meter_2.c and changing the rotation argument to 135 deg instead of 90.

https://github.com/lvgl/lvgl/blob/51a82a17ffe938d07d94660f49fd18962060943a/examples/widgets/meter/lv_example_meter_2.c#L28

This produces strange stutter in the animation right at the turnaround at 360 deg. In my opinion start_quarter and end_quarter must use a 360 modulo like the lv_trigo_sin function internally does.

```c
uint8_t start_quarter = (start_angle % 360) / 90;
uint8_t end_quarter = (end_angle % 360) / 90;
```

This would also make the two special cases which check *_quarter == 4 obsolete.

kisvegabor, 2021-08-17:
Thanks, for the report. 

Instead of the modulo division I added `if(angle > 360) angle -= 360` lines because `angle==360` must be kept as a  large number. `%` would make it `angle = 0`. In other words `angle == 360` should be in the `quarter = 3` and `angle == 0` in `quarter == 0`.

commit 0f14f494 (2021-08-09): fix(arc, merter) fix invaidation in special cases

fixes #2443