lvgl/lvgl #2448: Removing style of bar is delete text of button of list widget and bar

ercanersoy, 2021-08-10:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Removing style is delete text of button of list widget and bar.

### To Reproduce

```c

void menu_item_slider_add()
{
    // Create button of menu item slider
    menu_button = lv_list_add_btn(menu_list, LV_SYMBOL_DIRECTORY, "Button");

    // Remove all callbacks to menu item
    lv_obj_remove_event_cb(menu_button, LV_EVENT_ALL);

    // Create bar of menu item slider
    lv_obj_t *menu_bar = lv_bar_create(menu_button);

    // Set size of bar of menu item slider
    lv_obj_set_size(menu_bar, 100, 20);

    // Menu bar main style variable
    static lv_style_t menu_bar_main_style;

    // Initialization of menu bar main style
    lv_style_init(&menu_bar_main_style);

    // Set menu bar main style color
    lv_style_set_bg_color(&menu_bar_main_style, lv_color_make(0, 128, 64));

    // Set menu bar main style opacity
    lv_style_set_bg_img_opa(&menu_bar_main_style, LV_OPA_50);

    // Apply menu bar main style to menu bar
    lv_obj_add_style(menu_bar, &menu_bar_main_style, LV_PART_MAIN);

    // Menu bar indicator style variable
    static lv_style_t menu_bar_indicator_style;

    // Initialization of menu bar indicator style
    lv_style_init(&menu_bar_indicator_style);

    // Set menu bar indicator style color
    lv_style_set_bg_color(&menu_bar_indicator_style, lv_color_make(0, 128, 64));

    // Apply menu bar indicator style to menu bar
    lv_obj_add_style(menu_bar, &menu_bar_indicator_style, LV_PART_INDICATOR);

    // Set value of bar of menu item select
    lv_bar_set_value(0, LV_ANIM_OFF);

    // Add menu item to menu group
    lv_group_add_obj(menu_group, menu_button);
}

void menu_item_slider_change_style()
{
    // Get menu item from menu group
    lv_obj_t *menu_item_focused = lv_group_get_focused(menu_group);

    // If a menu item focused
    if(menu_item_focused != NULL)
    {
        // Get menu bar
        lv_obj_t *menu_bar = lv_obj_get_child(menu_item_focused, -1);

        // Remove all styles of menu bar
        //lv_obj_remove_style_all(menu_bar);

        // Menu bar main style variable
        static lv_style_t menu_bar_main_style;

        // Initialization of menu bar main style
        lv_style_init(&menu_bar_main_style);

        // Set menu bar main style color
        lv_style_set_bg_color(&menu_bar_main_style, lv_color_make(0, 128, 64));

        // Set menu bar main style opacity
        lv_style_set_bg_img_opa(&menu_bar_main_style, LV_OPA_50);

        // Apply menu bar main style to menu bar
        v_obj_add_style(menu_bar, &menu_bar_main_style, LV_PART_MAIN);

        // Menu bar indicator style variable
        static lv_style_t menu_bar_indicator_style;

        // Initialization of menu bar indicator style
        lv_style_init(&menu_bar_indicator_style);

        // Set menu bar indicator style color
        lv_style_set_bg_color(&menu_bar_indicator_style, lv_color_make(0, 128, 64));

        // Apply menu bar indicator style to menu bar
        lv_obj_add_style(menu_bar, &menu_bar_indicator_style, LV_PART_INDICATOR);
    }
}

void menu_item_slider()
{
    menu_item_slider_add();
    menu_item_slider_change_style();
}
```

### Expected behavior

Removing style of bar is not delete text of button of list widget and bar.

Removing style of bar and adding style of bar must change style of bar.

### Screenshots or video

Screenshots or video is not available.

kisvegabor, 2021-08-10:
Hi,

Please send a minimal code snippet to demonstrate only the issue.

This line seems incorrect: `lv_bar_set_value(0, LV_ANIM_OFF);`
It should be lv_bar_set_value(menu_bar, 0, LV_ANIM_OFF);`

There should be at least a warning about it but be sure there are no warnings.

ercanersoy, 2021-08-10:
I have updated my demonstration code.

I cannot share my originial code.

Thanks.

kisvegabor, 2021-08-10:
No need for the original code but only a few lines as example to show what the problem is. 

In your above example some variables are not declared, the list is not created, etc. We need something that we can simply copy paste to the simulator to see the issue.

ercanersoy, 2021-08-10:
I have updated my demonstration code.

ercanersoy, 2021-08-10:
I have updated my demonstration code again.

kisvegabor, 2021-08-10:
Thanks. The problem is that if you remove the styles `bg_opa` will be `0`. You need to set it too with 
`lv_style_set_bg_opa(&menu_bar_main_style, LV_OPA_COVER);`

Also note that removing the styles [will remove the set sizes](https://docs.lvgl.io/master/overview/coords.html#removing-styles) too.

ercanersoy, 2021-08-11:
Thanks. But, it does not work properly. The text of button of list widget does not remove now and the bar does remove now.

ercanersoy, 2021-08-11:
I have updated my demonstration code.

You could call menu() to seeing this issue.

ercanersoy, 2021-08-11:
> Thanks. But, it does not work properly. The text of button of list widget does not remove now and the bar does remove now.

My new demonstrate code was given below and test it in a simulator.

```c
lv_obj_t *menu_list;
lv_obj_t *menu_bar;

void menu_create()
{
    menu_list = lv_list_create(lv_scr_act());

    lv_obj_set_size(menu_list, 300, 220);

    lv_obj_align(menu_list, LV_ALIGN_CENTER, 0, 0);

    lv_list_add_text(menu_list, "Deneme");
}

void menu_item_slider_add()
{
    // Create button of menu item slider
    lv_obj_t *menu_button = lv_list_add_btn(menu_list, LV_SYMBOL_DIRECTORY, "Button");

    // Create bar of menu item slider
    menu_bar = lv_bar_create(menu_button);

    // Set size of bar of menu item slider
    lv_obj_set_size(menu_bar, 100, 20);

    // Menu bar main style variable
    static lv_style_t menu_bar_main_style;

    // Initialization of menu bar main style
    lv_style_init(&menu_bar_main_style);

    // Set menu bar main style color
    lv_style_set_bg_color(&menu_bar_main_style, lv_color_make(0, 128, 64));

    // Set menu bar main style opacity
    lv_style_set_bg_img_opa(&menu_bar_main_style, LV_OPA_COVER);

    // Apply menu bar main style to menu bar
    lv_obj_add_style(menu_bar, &menu_bar_main_style, LV_PART_MAIN);
}

void menu_item_slider_update()
{
    // Remove all styles of menu bar
    lv_obj_remove_style_all(menu_bar);

    // Set size of bar of menu item slider
    lv_obj_set_size(menu_bar, 100, 20);

    // Menu bar main style variable
    static lv_style_t menu_bar_main_style;

    // Initialization of menu bar main style
    lv_style_init(&menu_bar_main_style);

    // Set menu bar main style color
    lv_style_set_bg_color(&menu_bar_main_style, lv_color_make(0, 128, 64));

    // Set menu bar main style opacity
    lv_style_set_bg_img_opa(&menu_bar_main_style, LV_OPA_COVER);

    // Apply menu bar main style to menu bar
    lv_obj_add_style(menu_bar, &menu_bar_main_style, LV_PART_MAIN);
}

void menu()
{
    menu_create();
    menu_item_slider_add();
    menu_item_slider_update();
}
```

ercanersoy, 2021-08-12:
Given screenshot as before style removing
[image]

Given screenshot as after style removing
[image]

kisvegabor, 2021-08-14:
In your example tehre is `lv_style_set_bg_img_opa` instead of `lv_style_set_bg_opa`.

closed 2021-09-05 by lvgl-bot