lvgl/lvgl #2456: What do you think about having one more repo for LVGL?

kisvegabor, 2021-08-17:
### Introduce the problem

Now it's quite difficult (and not really solved) to handle the versioning, branching, and testing of repos other than the main lvgl repo. For example lv_demos, lv_lib_png, lv_lib_split_jpeg, lv_lib_qrcode, etc.

It'd be important to
- make it clear which version of the other repos should be used with a given version of lvgl
- have the same branching and release strategy for all repos
- be sure a change in lvgl doesn't break the other repos

### Suggested solution

On solution can be to merge the demos, drivers, and l library repos into the lvgl repo (and create a larger monorepo). This way everything would in one place, in sync, have the same branching, releases, etc. Besides the contributors would be concentrated into one common repo which is IMO good because get closer the the library itself. 

Finally, only the `lv_port_...` repos would remain as external repos.

Please, share you ideas, concerns, experiences with this monorepo concept.

embeddedt, 2021-08-17:
@kisvegabor FYI: I changed the title of the issue, as the original one sounded like we were creating an extra repository, not reducing the number.

kisvegabor, 2021-08-17:
Ah typo. I wanted to write "one mono repo" :slightly_frowning_face: 
Thanks!

embeddedt, 2021-08-17:
> Finally, only the lv_port_... repos would remain as external repos.

Can we make an exception for the Emscripten port, which has a very tiny amount of platform-specific code in it, as it is used directly by the docs? If we did, [this extra hack](https://github.com/lvgl/lvgl/blob/8ae894ebd4c6fda92dcd4e11d30483525bd9230b/scripts/build_html_examples.sh#L4-L13) would not be needed.

Miguel0101, 2021-08-17:
I agree to merge all branches into one.

kisvegabor, 2021-08-17:
@embeddedt 

> Can we make an exception for the Emscripten port, which has a very tiny amount of platform-specific code in it, as it is used directly by the docs? If we did, this extra hack would not be needed.

Would it mean adding only a script to the lvgl repo?

@Miguel0101
> I agree to merge all branches into one.

It's about repositories, not branches.

Miguel0101, 2021-08-17:
Oh, i'm sorry.

cmumford, 2021-08-17:
I am very much in favor of a mono repo. One concern of mine is to provide a better workflow (certainly no worse) for the various ways in which LVGL is consumed. Specifically:

* Desktop (Ubuntu, Widows, macOS, etc.)
* ESP-IDF
* STM (W4STM32 (Ac6) IDE)
* Micropython
* Emscripten

Also, to be able to easily use LVGL in:

* ESP-IDF
* CMake (on desktop)
* VSCode
* Eclipse
* Visual Studio

Also in various distro methods like
* Git submodule
* Installation to system (cmake install)
* PlatformIO
* Arduino??

I think it would be a good idea to see if it is possible to include all ports into the mono repo in a clean way. What is the argument for keeping `lv_port....` repos external to LVGL?

What are everyone's thoughts about creating a new mono-repo (which I think is being suggested in this issue) vs. merging the other repos into lvgl/lvgl. I think this project will have mildly dissatisfied users either way. I don't have a strong opinion, but am in favor of using lvgl/lvgl and not creating a new one.

kisvegabor, 2021-08-17:
> I think it would be a good idea to see if it is possible to include all ports into the mono repo in a clean way. 

Into the same repo or another mono repo?

> What is the argument for keeping lv_port.... repos external to LVGL?

(For the previous question I assume you meant the same mono repo)

A question is what is the purpose of these ports?
1. Projects that can be used to start the development and add other libraries as needed?
2. Just some examples to quickly try out LVGL and not meant for real development?

In the former case, the user needs to modify `lvgl` with some project-related code/settings and therefore he couldn't easily update `lvgl` later. 

In the latter case, it might work but still it'd result in an (IMO) ugly project hierarchy in which a library contains the project(s).  For SDKs and frameworks it's normal to have ports right in the main repo but for a simple library, it could be strange. 

Some other arguments:
- I've checked that the drivers in [lv_port_stm32f769_disco](https://github.com/lvgl/lv_port_stm32f769_disco) (HAL_Drivers, CMSIS, Utilities) has ~30MB size. With 10-20 projects the repo size could be easily bloated.
- It'd be much harder to use managed builds especially for novice users or in frameworks like Arduino or PLatformIO. Extra steps are needed (not sure if even possible) to exclude the source files of the ports from the build. 

> What are everyone's thoughts about creating a new mono-repo (which I think is being suggested in this issue) vs. merging the other repos into lvgl/lvgl. I think this project will have mildly [...]

I'm also in favor of merging here. 

> I think this project will have mildly dissatisfied users either way.

Do you mean the "mono-repo project" or lvgl in general?

embeddedt, 2021-08-17:
>> Can we make an exception for the Emscripten port?

> Would it mean adding only a script to the lvgl repo?

We'd need the `main.c` and `CMakeLists.txt` files as well. The only reason I want to break the rules with this port is because the tests/CI builds use it, and it's very rarely used for a real project (based on the low number of opened issues in the 2.5 years it's been available).

The submodules wouldn't be needed since the drivers and examples would be available in the same repository.

kisvegabor, 2021-08-17:
> We'd need the main.c and CMakeLists.txt files as well. The only reason I want to break the rules with this port is because the tests/CI builds use it, and it's very rarely used for a real project (based on the low number of opened issues in the 2.5 years it's been available).

I see no problem with it if the content of `main.c` and `CMakeLists.txt` is enabled only in Emscripten environment or by other defines set when building tests.

cmumford, 2021-08-17:
> > I think it would be a good idea to see if it is possible to include all ports into the mono repo in a clean way.
> 
> Into the same repo or another mono repo?

The same repo. If not, then we'll still have the same sync issues - just less so. That being said, we don't want to create a mess of starter projects either. If they can be cleanly added to the mono repo then I think that would be best.

> > What is the argument for keeping lv_port.... repos external to LVGL?
> 
> (For the previous question I assume you meant the same mono repo)

Yes, keeping external to the mono repo.

> A question is what is the purpose of these ports?
> 
> 1. Projects that can be used to start the development and add other libraries as needed?
> 2. Just some examples to quickly try out LVGL and not meant for real development?

I see the reason for having the ports - only questioning where they should be.

> In the former case, the user needs to modify `lvgl` with some project-related code/settings and therefore he couldn't easily update `lvgl` later.

I'm thinking that it should be possible to create these ports where they will compile/run in place with LVGL being at ../src (or something like that), but also where the user could copy just the porting project directory, tweak a few CMake settings, and the build against a LVGL in a git submodule.

> In the latter case, it might work but still it'd result in an (IMO) ugly project hierarchy in which a library contains the project(s). For SDKs and frameworks it's normal to have ports right in the main repo but for a simple library, it could be strange.

Yes, certainly pro's and cons to having them in the same repo as LVGL - just wanted to have this discussion.

> Some other arguments:
> 
> * I've checked that the drivers in [lv_port_stm32f769_disco](https://github.com/lvgl/lv_port_stm32f769_disco) (HAL_Drivers, CMSIS, Utilities) has ~30MB size. With 10-20 projects the repo size could be easily bloated.

The `.git` folder is only 2.95 MB for lv_port_stm32f769_disco (but the extracted files are 29.5 MB). When it comes to size, what do you think are the concerns:

1. Download time.
2. Size on disk.
3. Number of files/dirs, and the ensuing confusion for some users.

Download time is only a function of the `.git` directory size, I'm not sure how much disk space matters, so my guess is that # of files (i.e. simplicity) is probably the biggest concern.

> * It'd be much harder to use managed builds especially for novice users or in frameworks like Arduino or PLatformIO. Extra steps are needed (not sure if even possible) to exclude the source files of the ports from the build.

I think there are several things that might work for exclude files:

1. Git [sparse-checkout](https://git-scm.com/docs/git-sparse-checkout)
2. Creating a release tarball/zip that excludes the ports/examples.

Just a few thoughts. I don't want to bog down the discussion too much, and am OK with whatever direction you guys want to take. I think that any consolidation will be good.

> I'm also in favor of merging here.
>
>
> Do you mean the "mono-repo project" or lvgl in general?

I think that if lvgl/lvgl becomes the mono repo (my vote) then users will have duplicate symbols, and need to remove some git submodules, etc. If we instead go with lvgl/lvgl-mono (name for discussion only), then users will be confused with why lvgl/lvgl stopped receiving updates, and will need to swap one git submodule and still delete some. Either way it's going to be mildly confusing for downstream users.

cmumford, 2021-08-17:
One more thought. When the repos are combined we should decide if history should be preserved (I vote for this) or not. So, let's say that the decision is to merge lv_lib_png, lv_lib_split_jpeg (etc.) into the existing lvgl repo. Also, let's say we wanted the layout to be like this:

```
docs/
examples/
src/ 
lib/
   lv_lib_png/
   lv_lib_split_jpeg/
```

**Note**: Not a recommended directory structure - just for discussion.

So we could either just copy the HEAD of lv_lib_png into `lib/lv_lib_png`, and commit as a single commit, or we can do it in such as way that the full history of lv_lib_png is preserved. I think it would be best to preserve history.

kisvegabor, 2021-08-18:
> Download time is only a function of the .git directory size, I'm not sure how much disk space matters, so my guess is that # of files (i.e. simplicity) is probably the biggest concern.

I also see simplicity as the biggest concern. 

> I think there are several things that might work for exclude files:
1. Git sparse-checkout
2. Creating a release tarball/zip that excludes the ports/examples.

We have almost no control on these if lvgl is downloaded from a packet manager (e.g. in Arduino)

> When the repos are combined we should decide if history should be preserved (I vote for this) or not. 

I also vote for keeping the history. It wouldn't be fair to erase the attributions for contributors. 

--- 

Regarding the directory structure: I'm thinking about whether we need to move all sources to `src` including examples, drivers, and libraries.

---

For the same of simplicity it'd be great to keep lvgl free from submodules. It can work for the majority of external repos but e.g. [lv_lib_rlottie](https://github.com/lvgl/lv_lib_rlottie) has a very complex submodule from Samsung.

---

@amirgon How it would affect the MicroPython related things?

embeddedt, 2021-08-18:
> How it would affect the MicroPython related things?

`lv_micropython` itself definitely needs to stay in its own repository, as we need to be able to cleanly merge upstream MicroPython into it. We may be able to move the bindings here.

kisvegabor, 2021-08-18:
> lv_micropython itself definitely needs to stay in its own repository, as we need to be able to cleanly merge upstream MicroPython into it.

I agree, as lvgl is only submodule in the upstream micropython project. 

> We may be able to move the bindings here.

That would be great if possible.

cmumford, 2021-08-18:
I wrote a test script ([mono_repo_test.py](https://github.com/cmumford/lv_utils/blob/mono-repo/mono_repo_test.py)) to see how big a mono-repo would be, and what the history would look like. I think it's possible to preserve the commit ID's of all repos - I haven't checked if this approach does that. It's probably leaving off non-master branches too. Please ignore the list of merged repos - I just grabbed what thought were *not* ports or things mentioned as excluded in this issue.

embeddedt, 2021-08-18:
Hmm... it looks like the renaming might be hiding the rest of the commit history, as running `git log libs/lv_binding_micropython` only returns the "Move files to libs/ directory" commit.

cmumford, 2021-08-18:
I'm not sure if you can follow a directory over renames. As you noted `git log libs/lv_binding_micropython` has one commit, as does `git log libs/lv_binding_micropython/lv_conf.h`. But, `git log --follow libs/lv_binding_micropython/lv_conf.h` gives the full history. I'm not sure if there is a repo merge strategy that can do this.

embeddedt, 2021-08-18:
The inability to follow a directory is not the biggest issue; I generally look at file history only. I wasn't aware that the `--follow` option exists so that's nice. :slightly_smiling_face:

amirgon, 2021-08-18:
I'm not a big fan of monorepos.  
I don't particularly like the fact that when I only need LVGL library, I would get along with it a bloated repo with lots of stuff I'm not interested in.

Alternatively, I prefer solving these problems with git-submodules and automation.  
When a library depends on LVGL it can include LVGL as a git submodule instead of being included in LVGL directly.
I'm in favor of automation (such as GitHub actions) to check that nothing is broken, this can work with either monorepo or multiple repos with submodules.

> @amirgon How it would affect the MicroPython related things?

Specifically regarding Micropython:
- lv_micropython is a fork of Micropython so obviously must remain independent.
- lv_binding_micropython could be merged into an LVGL monorepo **but I would prefer not to merge it**.

I can think of multiple examples in the past where it was useful for me to have lv_binding_micropython independent from LVGL.
For example - 
- When LVGL introduced a bug into the lv_micropython ESP32 build (which wasn't caught be CI) I was still able to use latest lv_binding_micropython and commit changes to it by setting the LVGL submodule to point to the last good LVGL version until the bug was fixed.
- I have control over which LVGL version is used on each lv_binding_micropython branch.
- It is very easy to switch LVGL versions used in a specific lv_binding_micropython version to test different things.

All these would be much more difficult with monorepo. I just don't see the great benefit in moving lv_binding_micropython into LVGL.
I can see why you would want to merge docs, examples and demos to LVGL, they are tightly related and integrated with LVGL. If LVGL changes the example should change accordingly.  
But for repos that are not that tightly coupled with LVGL, I don't think the best solution is to blindly merge everything to a monorepo.

kisvegabor, 2021-08-19:
@cmumford 
> I wrote a test script (mono_repo_test.py) to see how big a mono-repo would be, and what the history would look like.

Thanks Chris. I've run your script and uploaded it result to the [monorepo](https://github.com/lvgl/lvgl/tree/monorepo) branch to make visible for everybody. 

I've noticed that the submodule from `lv_lib_rlottie` is missing. 
- Original: https://github.com/lvgl/lv_lib_rlottie/tree/master/rlottie_lib 
- Merged: https://github.com/lvgl/lvgl/tree/monorepo/libs/lv_lib_rlottie/rlottie_lib

@amirgon 
> I'm not a big fan of monorepos.

I see your point and your arguments about `lv_binding_microptyhon` makes sens. 

What do you think about merging the `lv_lib_..` repos? The biggest advantage I see in monorepo is that these small "side projects" will follow the branching and versioning of the main lvgl repo. Therefore you automatically get compatible `lvgl`-`lv_lib` pairs. Besides CI becomes much simpler.

embeddedt, 2021-08-19:
> I've noticed that the submodule from lv_lib_rlottie is missing.

I think we should strive to keep submodules out of the main repository if possible. Submodules complicate merges and checkouts quite a bit (since they often need manual updates). For example, if @amirgon and I both bump `lv_bindings` on our separate `lv_micropython` branches, merging of the latter will always fail even if the former had merged fine. Git's submodule resolution logic is quite limited in my experience.

Right now, with LVGL, you really only need to add one submodule to your own project (LVGL itself), and it doesn't have more submodules underneath it which you have to deal with as well.

kisvegabor, 2021-08-19:
@embeddedt 
We can decide to merge only those libs that doesn't contain submodules. If the library is small (1-2 files) we can hard copy them as probably some tailoring will be required anyway. 

If the library is large we can keep it as an external repo (e.g. rlottie). Adding that many "foreign" files to lvgl as they are would make complicated to build lvgl in some automatic build systems anyway.

amirgon, 2021-08-19:
> What do you think about merging the `lv_lib_..` repos? The biggest advantage I see in monorepo is that these small "side projects" will follow the branching and versioning of the main lvgl repo. Therefore you automatically get compatible `lvgl`-`lv_lib` pairs. Besides CI becomes much simpler.

If these `lv_lib_..` repos are small side projects, it might not be an issue.  
But my concern is that a monorepo would eventually become bloated, if not right now then in the future.  
I think it's valuable to have a "core-lvgl" repo that contains just that, core LVGL code. Monorepo goes against that idea - there is no easy way to just get LVGL core without all the satellite projects around it which are really needed only for specific cases.

@embeddedt mentioned some disadvantages of git-submodules specifically regarding merging.  
But I see many advantages to git submodules and I think they can be used efficiently exactly for that purpose - splitting large projects into "libraries" with clear dependencies between them and a simple way to manage these dependencies.

So I would like to suggest another way to achieve your goals of clear versioning, branching strategy and testing of everything together, but a way that keeps lvgl-core an independent repo.

What if lvgl repo was "core-lvgl" without any `lv_lib_..` libraries, but we also had another repo "lvgl-ecosystem" (or something like that) which included "lvgl-core" and all `lv_lib_..` libraries as submodules?
Simple `lv_lib_..` libraries could be added to "lvgl-ecosystem" directly, more complex or independent libraries as submodules.  
Then, you can do all your tagging branching and testing on that "lvgl-ecosystem", not on "lvgl-core".
CI becomes simple because it is done centrally on a single location on the "lvgl-ecosystem".
We can also have an Action that automatically bumps the "lvgl-core" submodule on "lvgl-ecosystem" branch after every commit to  "lvgl-core" branch, and then triggers tests on the entire ecosystem to make sure everything plays together.  

This way, If I want, I can always get the slim non-bloated "lvgl-core". But "lvgl-core" is always tested with the entire ecosystem.  
Looking at an "lvgl-ecosystem" commit it's easy to tell exactly which versions of each library work together, and the whole "lvgl-ecosystem" can be tagged, branched etc.

kisvegabor, 2021-08-19:
> What if lvgl repo was "core-lvgl" without any lv_lib_.. libraries, but we also had another repo "lvgl-ecosystem" (or something like that) which included "lvgl-core" and all lv_lib_.. libraries as submodules?

I see the good points in this idea but see some problems too. You have collected the advantages so I'll mention only the drawbacks I see. :slightly_smiling_face: 
 Lets take some use cases:
1. `lv_micropython`: Now `lvgl` and `lv_lib_png` is included. Would you include `lvgl-ecosystem` for the PNG lib (and maybe for others) or only `lvgl-core`?
2. NXP GUIGuider: I don't know which repo would make more sense for them (AFAIK now only `lvgl` is used).   
3. EdgeLine: It relies on the MicroPython binding. But we need to make the user decide whether he uses the core or the ecosystem repo to enable/disable certain features in the editor. Might be confusing for novice users. 
4. STM32F429-Disco port: Code size doesn't matter as the compiler/linker will optimize unused function and data. Migrating from one repo to the other by the user is complicated (submodule deinit, remove, add, init, update) and harder to update from the upstream once the submodule has changed. 
5. Arbitrary UI development: There are 2 (or more) `lv_conf*.h` files that isn't handy. Needs to look for features, issues, etc at 2 places. 
6. Development of LVGL in general: Versioning and branching becomes clearer but still more cumbersome than with 1 repo. It also separates the contributors to 2 places. 
7. Package managers (e.g. Arduni, PlaformIO): They might not support libraries with submodules so `lvgl-ecosystem` might not work in some cases.

> But my concern is that a monorepo would eventually become bloated, if not right now then in the future.

@cmumford added a [good comment](https://github.com/lvgl/lvgl/issues/2456#issuecomment-900677780) about it:
> The .git folder is only 2.95 MB for lv_port_stm32f769_disco (but the extracted files are 29.5 MB). When it comes to size, what do you think are the concerns:
> 1. Download time.
> 2. Size on disk.
> 3. [...]

So the question what's the biggest problem with a bloated repo? Do you also agree that it's the number of files (what correlates with complexity to some extent)?

Why is it better to have a bloated ecosystem repo than a bloated monorepo? It should be possible to disable the unused libs via config files in both cases. What can be the case when `lvgl-core` is better `lvgl-mono` with all the libs disabled?

> Then, you can do all your tagging branching and testing on that "lvgl-ecosystem", not on "lvgl-core".

`lvgl-core` still needs to be versioned, branched, tagged for the users who uses only this repo. So the versioning issue still persists. E.g. if we release `lvgl-core v.8.5.9` shall we release `lvgl-ecosystem v8.5.9` too?

amirgon, 2021-08-19:
A basic difference between a monorepo and "lvgl-ecosystem" is that in monorepo you always get everything, while on "lvgl-ecosystem" you can select what you want.  

As an example, upstream Micropython includes lots and lots of git-submodules, but depending on the port you are building, you usually only need a few of them. Calling `make -C ports/unix submodules` would init and update only the few required git-submodules for the unix port.

In case of "lvgl-ecosystem" you also got this choice - if you need LVGL on ESP32 you would init and update "lvgl-core" and "lv_esp32" submodules, but you won't update `lv_lib_rlottie` for example. It won't be dowloaded from network and won't consume disk space it simply won't be there if you don't need it.

> 1. `lv_micropython`: Now `lvgl` and `lv_lib_png` is included. Would you include `lvgl-ecosystem` for the PNG lib (and maybe for others) or only `lvgl-core`?

Actually `lv_micropython` uses lodepng directly, not `lv_lib_png`.  
But a good question is whether `lv_micropython` would use `lvgl-ecosystem` or `lvgl-core`. I would say it makes sense to use `lvgl-ecosystem` as lv_micropython submodule, but initialize only the relevant git-submodules in it, such as  `lv_binding_micropython` but possibly others as well.

> 3\. EdgeLine: It relies on the MicroPython binding. But we need to make the user decide whether he uses the core or the ecosystem repo to enable/disable certain features in the editor. Might be confusing for novice users.

Again - it can use lv_micropython with `lvgl-ecosystem` and init the relevant submodules, or all of them if you don't care bloating it.

> 5\. Arbitrary UI development: There are 2 (or more) `lv_conf*.h` files that isn't handy. Needs to look for features, issues, etc at 2 places.

I'm not sure I understand why there would be 2 (or more) `lv_conf*.h` files.  
Probably one is enough on "lvgl-ecosystem", I don't think we need one on each repo. Even on "lvgl-core" you can have none, just provide a simple way to generate one.

> 6\. Development of LVGL in general: Versioning and branching becomes clearer but still more cumbersome than with 1 repo. It also separates the contributors to 2 places.

Correct. Contributors would be spread across repos when the repos are submodules of "lvgl-ecosystem". I'm not sure if this is good or bad. You can have better access control with multiple repos for example. Today I can push my changes directly to `lv_micropython` but I have to open a PR to make a change in LVGL.

> 7\. Package managers (e.g. Arduino, PlaformIO): They might not support libraries with submodules so `lvgl-ecosystem` might not work in some cases.

This is interesting. I very much doubt this is true, because git-submodules are so widely used.  
If these Package managers support Micropython, for example, they must support submodules because they are widely used there.

> So the question what's the biggest problem with a bloated repo? Do you also agree that it's the number of files (what correlates with complexity to some extent)?

All of them. I once tried to use Zephyr and gave up after it consumed Gigabytes of disk space just to be able to build a tiny project.  
Over time, when the number of satellite projects increase and when these projects become more complex, each of these - download time, size on disk and number of files - would increase to the point it starts bothering people.

> Why is it better to have a bloated ecosystem repo than a bloated monorepo? It should be possible to disable the unused libs via config files in both cases. What can be the case when `lvgl-core` is better `lvgl-mono` with all the libs disabled?

As mentioned - because you only initialize the submodules you really need on "ecosystem repo".
Uninitialized submodules aren't downloaded and don't consume disk space.

> `lvgl-core` still needs to be versioned, branched, tagged for the users who uses only this repo. So the versioning issue still persists. E.g. if we release `lvgl-core v.8.5.9` shall we release `lvgl-ecosystem v8.5.9` too?

This can be solved with automation.  
The same way you'd have an Action to bump "lvgl-core" on the ecosystem repo for every commit on "lvgl-core", you can also have an action to tag/branch "lvgl-core" on every tag/branch of the ecosystem. And you never tag/branch "lvgl-core" directly.
Yes, it's a bit more cumbersome, but might be worth it if you want to avoid a bloated monorepo.

kisvegabor, 2021-08-19:
I got the point. The concept of leaving the submodules uninitialized sounds reasonable. 
I'm going to a holiday and will be back on Tuesday (24.08) so can join again in a few days. In the meanwhile I'll think about it. :slightly_smiling_face:

amirgon, 2021-08-19:
> I'm going to a holiday and will be back on Tuesday (24.08) so can join again in a few days.

Enjoy your holiday!! :smile:

kisvegabor, 2021-08-20:
> Enjoy your holiday!! smile

Thanks ^^

Something just came to my mind. If the main concern is bloating with a monorepo adding tests for the core (>100 files) or 15 new widgets won't cause the same problem? Adding 5-10 libs or even the drivers means far less files. The examples (that are already in `lvgl`) consist of 112 C files and bloating was not an issue neither from user nor from developer's side.

amirgon, 2021-08-20:
> If the main concern is bloating with a monorepo adding tests for the core (>100 files) or 15 new widgets won't cause the same problem? Adding 5-10 libs or even the drivers means far less files. The [...]

In principle, in my suggestion "lvgl-core" does not include examples or tests, only the core. We can even consider splitting out widgets.

Your point is valid - today no one (yet) complained about LVGL repo size, including all examples and tests.  
So today, a monorepo that includes all satellite projects might still be small enough and easier to manage than other solutions.

But did you consider future scaling? What is the growth rate you expect in the number and size of LVGL "satellite" projects?
What would happen if in a year or two your monorepo eventually becomes bloated?  

The "lvgl-ecosystem" suggestion aims to solve the *scalability* issue. Monorepos are not scalable - may be good enough for today but not necessarily for tomorrow.

higaski, 2021-08-24:
I can see the upside from a maintainers point of view, but as a user I wouldn't be particularly thrilled. I'm using CMake to fetch LVGL and some of it's sideprojects at configuration time. It was easy for me to write my own Find*.cmake files so far since most of the repos are small. Writing a CMake file for a mono repo sounds like a proper challenge, given that it might need to define dozens of targets...

kisvegabor, 2021-08-24:
> What is the growth rate you expect in the number and size of LVGL "satellite" projects?

Good question. Let me estimate it here. 

We already have libs for the major image format, freetype, qr code, rlottie (not sure it really needs to be part of lvgl). I can imagine some more such as SVG or other vector render, some kind of 3D lib and video player and maybe 5 more (?). Usually these consist of 2-4 files so probably they wont bloat lvgl much.

The drivers are an other topic, we can't clearly see the new structure yet, so still don't know about it. Probably 1 source and 1 header file for each display and touch controller. Let's say 50 source + 50 header files.

So let say in roughly 100 source files in total not counting the tests and examples. 

> I can see the upside from a maintainers point of view, but as a user I wouldn't be particularly thrilled. I'm using CMake to fetch LVGL and some of it's sideprojects at configuration time. It was [...]

I imagined only 1 CMakeFile for the whole monorepo so there would be only 1 target. Do you think we should use different CMakeFile-s for each satellite project?

@embeddedt @cmumford do you have any comments for @amirgon's and @higaski's thoughts?

fvanroie, 2021-08-24:
I'm worried about how this monorepo will work in PlatformIO and possibly Arduino IDE, where libraries have a defined structure.
PIO does the library management per project and build target. Atm I can just point to which libraries and version to include:

```
lib_deps =
    lvgl/lvgl @ ^8.0.1
    lvgl/lv_examples @ ^8.1.0-dev
```
or I can pinpoint specific branches or commits for each and have some tweaked forks:
```
lib_deps =
    git+https://github.com/lvgl/lvgl.git#release/v7
    ;git+https://github.com/lvgl/lvgl.git#797212a3f346e9fdf71aef6f193889357eeae3f0  ; bug-fix
    ;lvgl/lvgl@^7.11.0                ; from PIO library
    git+https://github.com/lvgl/lv_lib_png.git#release/v7
    git+https://github.com/fvanroie/lv_drivers
```
By default, PIO uses `src/` only to find headers and compile sources. For instance, the lvgl/examples are not *found* when specifying: `lvgl/lvgl @ ^8.0.1` library since they live  outside lvgl/src. I have to either copy them into my project or use some `src_filter` magic. I would have to jump through several hoops to make the monorepo branch work in its current structure.

I really like the **simplicity** the structured libraries offer to the (novice) user or the "lvgl-ecosystem" as @amirgon puts it. Repo size and compile time are also a concern, but not nearly as much as the usability/flexibility (i.e. for people _not_ is using cmake).

In order to cut down on the number of repo's, maybe it's useful to group some of them together like `lv_img_file_if` for bmp, jpg, png, gif, svg, etc...  The same way `lv_fs_if` does for fat, posix, pc.

higaski, 2021-08-24:
> I imagined only 1 CMakeFile for the whole monorepo so there would be only 1 target. Do you think we should use different CMakeFile-s for each satellite project?

CMakes notion of a "target" is either
- an executable
- a library (static/shared/header only)
- or even a bunch of object files

Usually bigger monorepos split their CMake stuff into multiple CMakeLists.txt files where each subdirectory/project adds 1 or more targets. At my employer I've consciously moved away from a monorepo structure as it feels counterintuitive to how CMake fundamentally works.

cmumford, 2021-08-24:
> The "lvgl-ecosystem" suggestion aims to solve the scalability issue. Monorepos are not scalable - may be good enough for today but not necessarily for tomorrow.

I would argue that monorepos **are** scalable, and are in use today on some of the largest software companies in the world specifically to solve the problem of scalability. The [scalability issues](https://en.wikipedia.org/wiki/Monorepo#Scalability_challenges) are generally only hit on the very massive monorepos that hit the limits of Git, filesystems, review systems, etc. LVGL is really a tiny project by comparison. It has (I'm guessing) about 20K lines of code in the entire LVGL ecosystem.

I agree with @higaski that when considering the monorepo we need to consider both the developer and the LVGL user. I find dealing with Git submodules difficult - especially when they are nested recursively in a directory tree. I'm likely projecting, but I'm guessing that the average LVGL user would find it simpler to get one project, whether via Git, PlatformIO, or Arduino library manager, instead of several related libraries which must also be kept in sync.

Being able to grep through all project source, do atomic commits, run tests quicker, will contribute to developer velocity.

@fvanroie We can't break PlatformIO or make it more difficult to use for LVGL development. Are you concerned about the end user experience, or for LVGL developers?

At present I can build LVGL and it's test programs in under 20 seconds on a laptop. I don't think that building all `lv_lib_*` files will dramatically impact the build times. I do think that simplifying the LVGL developer/user experience will have a measurable impact.

kisvegabor, 2021-08-24:
> By default, PIO uses src/ only to find headers and compile sources. For instance, the lvgl/examples are not found when specifying: lvgl/lvgl @ ^8.0.1 library since they live outside lvgl/src

AFAIK it's the same for Arduino but we can move `examples`, `lv_lib_...` in `src`. I don't think it'd matter too much in the end.

> I really like the simplicity the structured libraries offer to the (novice) user or the "lvgl-ecosystem" as @amirgon puts it. Repo size and compile time are also a concern, but not nearly as much as the usability/flexibility (i.e. for people not is using cmake)

I agree that PIO and Arduino support is must. But I think a monorepo would make both of them simpler. Just add `lvgl` and you get PNG, JPEG, QR code etc libraries, you only need to enable them in lv_conf.h.

> At my employer I've consciously moved away from a monorepo structure as it feels counterintuitive to how CMake fundamentally works.

I see no problem with using "local" CMakeFiles but having a top level lvgl CMakeFile too which collects the local ones.

cmumford, 2021-08-24:
I'm not sure there's a need to move `examples` or `tests` into `src`. I think it is only files that must be in `src` are those that make up the library source(s) that will be linked with the client application. I think we can leave other code next to `src` and it will just be vestigial - unused by the higher level application.

I think it would be nice to support a "build everything" workflow for LVGL contributors using PlatformIO. I think that's maybe what @fvanroie was eluding to above.

kisvegabor, 2021-09-07:
In the pas weeks I was thinking a lot about the monorepo idea and believe the advantages are larger then the drawbacks. As possible issues the followings were mentioned:
- increased repo size: IMO it's a marginal issue
- increased complexity: with proper library structure it shouldn't be a problem 
- possible bloat up in the future: I myself can't foresee a significant bloat expect adding drivers and tests 
- problems with library managers: with a monorepo all should work smoother as the libs will look like a widget.
- harder to scope permissions of contributors: It really can be an issue but I'm optimistic in this regard. If we notice harmful behavior we can  change the permissions of that user. 

The main advantage I see is the reduced fragmentation:
- all libs and can use the "tools" of `lvgl`: Kconfig, lv_conf.h, test environment, docs, build, library manager integration, integration to ports
- versioning and branching will be consistent automatically 
- contributors are in one place: IMO once contributors are here it's more likely the will contribute to other parts too.

For the discussion it was clear that there is no perfect choice but I believe this structure will be easir to maintain in the long term. So if there is no blocking objection I prose these:
- move examples to: `src/examples` to make them available on Arduino too
- add the following libs to `src/extra/libs`: png, jpeg, bmp, gif, qrcode
- start to design and add drivers to `src/drivers`

Let me know if you are ok with this direction.

This is related to https://github.com/lvgl/lvgl_esp32_drivers/issues/105 so I cc  @C47D and @tore-espressif.

C47D, 2021-09-07:
Hi all,

Talking about the drivers I think it would be better to have them in a separated repo, we're still trying to figure out how to:
1. Remove MCU dependencies.
2. Have all of the drivers supported by ESP32 in the LVGL drivers repo.
3. Provide default implementations for common MCUs and interfaces.
4. Being able to configure the displays without using Kconfig or provide it as alternative as we did in the main LVGL repo.

So that repo still needs a lot of work, maybe you could integrate the examples repo into this one (as you said) and incorporate the drivers once we finish the current work and if the mono repo works as expected with the other modules. I haven't read the whole discussion tho.

What do you think?

kisvegabor, 2021-09-07:
@C47D We could work on the driver on a different branch of lvgl. This way we can use e.g. `lv_conf_template/internal.h` that solves the Kconfig/`#define` based config issue. In addition I believe the moving the development of the drivers to the `lvgl` repo will bring more attention, more ideas, more contribution etc.

puzrin, 2021-09-17:
My 2c:

- Monorepo is not about distribution, it's about keeping things in sync. You still can release multiple packages.
- Github tracker is poor. It may be inconvenient to use single tracker for completely orthogonal packages.

For example - it's good idea to keep extra widgets & drivers in monorepo. But i'm not sure about lv_i18n (it doesn't need sync and has independent issues).

kisvegabor, 2021-09-17:
> For example - it's good idea to keep extra widgets & drivers in monorepo. But i'm not sure about lv_i18n (it doesn't need sync and has independent issues).

I agree. Only the code directly compiled with LVGL should be in a monorepo. At least for first it looks like a safe step. 

---

If there is no blocking objection I'll move the `lv_lib` projects (png, jpeg, bmp, gif, qrcode) and `lv_fs_if` into `lvgl`.

Please give a thumbs up or drop a line to let me know you read this comment and agree.

kisvegabor, 2021-09-19:
I've started to work on it: https://github.com/lvgl/lvgl/pull/2569

Some polishing is still required but I meant something like this. Please let me know what you think about it in PR. 

As the monorepo was kicked off I close this issue. 

Thank you for contributing and sharing your ideas!

closed 2021-09-19 by kisvegabor