lvgl/lvgl #2457: Problems caused by lv_obj_set_style_clip_corner

qinshijing, 2021-08-17:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When clip conner is set to true, if the opacity of the parent and children background is not 100%, the color mixing area will be abnormal.

### To Reproduce
static void test(void)
{
    lv_obj_set_style_bg_color(lv_scr_act(), lv_color_black(), 0);
    lv_obj_t* obj = lv_obj_create(lv_scr_act());
    lv_obj_set_style_bg_opa(obj, LV_OPA_20, 0);
    lv_obj_set_style_pad_all(obj, 0, 0);
    lv_obj_set_style_border_width(obj, 0, 0);
    lv_obj_set_style_clip_corner(obj, true, 0);
    lv_obj_set_style_radius(obj, 50, 0);
    lv_obj_set_size(obj, 400, 300);
    lv_obj_center(obj);
    lv_obj_t* obj1 = lv_obj_create(obj);
    lv_obj_set_style_bg_opa(obj1, LV_OPA_20, 0);
    lv_obj_set_style_border_width(obj1, 0, 0);
    lv_obj_set_style_radius(obj1, 0, 0);
    lv_obj_set_size(obj1, 400, 100);
    lv_obj_align(obj1, LV_ALIGN_BOTTOM_MID, 0, 0);
}

### Expected behavior
[image]

### Screenshots or video
[image]

kisvegabor, 2021-08-17:
Thanks, I've just pushed a fix.

closed 2021-08-17 by kisvegabor

qinshijing, 2021-08-18:
Thanks, it's working normally.

commit 01237da4 (2021-08-17): fix(draw) fix rectangle drawing with clip_corner enabled

fixes #2457