lvgl/lvgl #2459: Removing of an evenrt callback using the _lv_event_dsc_t address does not work

glueckm, 2021-08-17:
### Describe the bug
When you add an event callback to an object the address for the event_dsc will be returend.
This can be used to remove the event callback at a later time.

But, if you add a second callback, the internal array ```obj->spec_attr->event_dsc``` get's ```remalloced``` which can lead to a new location of the WHOLE array.
Therefore when you try to remove the event callback by using the address returned by ```lv_obj_add_event_cb``` is no longer valid and the removal of the object fails.

kisvegabor, 2021-08-17:
That's true. We can either store only pointers to `event_dsc`s (has extra memory cost) or remove the option of removing event callback this way.

glueckm, 2021-08-17:
For my use case a need the option with the event descriptors because use the same callback multible times and therfore cannot use the callback variant. 

Or we add the option to add the user data for removing. The callback/userdata combination is unique. 

Martin

kisvegabor, 2021-08-18:
> Or we add the option to add the user data for removing. The callback/userdata combination is unique.

I like this idea. @embeddedt what do you think?

embeddedt, 2021-08-18:
I agree as well.

glueckm, 2021-08-18:
I have already implemented the feature in project.
Should I provide PR?
if yes, how should I name the new function?

kisvegabor, 2021-08-18:
I've just checked how to implement it and stucked at naming too. :smile: 

So now we have
```c
//Remove an event with matching callback
bool lv_obj_remove_event_cb(struct _lv_obj_t * obj, lv_event_cb_t event_cb);

//Remove an event with matching dsc
bool lv_obj_remove_event_dsc(struct _lv_obj_t * obj, struct _lv_event_dsc_t * event_dsc);
```

And now we need
```
//Remove an event with matching callback and user_data
bool lv_obj_remove_event_cb_user_data(struct _lv_obj_t * obj, lv_event_cb_t event_cb, const void * user_data);
```

But this name is ugly. What about simply `lv_obj_remove_event`?

glueckm, 2021-08-18:
> But this name is ugly. What about simply lv_obj_remove_event?

I have no problem with this name... But it feels like it does not fit well with the other names.
```lv_obj_remove_event_cb_user_data``` does fit better compare to the other names

embeddedt, 2021-08-18:
`lv_obj_remove_event_with_user_data`?

glueckm, 2021-08-18:
> ```lv_obj_remove_event_with_user_data```
Sounds better

kisvegabor, 2021-08-19:
> lv_obj_remove_event_with_user_data

With this logic `lv_obj_remove_event_cb` should be `lv_obj_remove_event_with_cb`. Anyway, `lv_obj_remove_event_with_user_data` seems ok. I think there is no perfect name to describe this function.

kisvegabor, 2021-09-03:
I've just implemented it.

commit 4eddeb35 (2021-09-03): feat(event) add lv_obj_remove_event_cb_with_user_data
fixes #2459
closed 2021-09-03 by kisvegabor