lvgl/lvgl #2462: lv_obj_swap does not change the focus order

glueckm, 2021-08-18:
### Describe the bug
When ```lv_obj_swap``` is used on a ```lv_list``` to swap two items of the list, the order of the focus is not changed.

### To Reproduce
Here is an exaple showing the problem. When I use mouse wheel, the focus jumps form Item 1 (which is the last in the list) to Item 2 (which is the seconf in the list).
And from Item 4 (the second to last) it jumps to Item 5 (which is the frist after the swap).

```c
#include <Windows.h>

#if _MSC_VER >= 1200
// Disable compilation warnings.
#pragma warning(push)
// nonstandard extension used : bit field types other than int
#pragma warning(disable:4214)
// 'conversion' conversion from 'type1' to 'type2', possible loss of data
#pragma warning(disable:4244)
#endif

#include "lvgl.h"
#include "win32drv/win32drv.h"
#if _MSC_VER >= 1200
// Restore compilation warnings.
#pragma warning(pop)
#endif

int main(int argc, char** argv) {
    lv_init();

    if (!lv_win32_init(
        GetModuleHandleW(NULL),
        SW_SHOW,
        320,
        240,
        nullptr))
    {
        return -1;
    }
    lv_win32_add_all_input_devices_to_group(NULL);
    lv_obj_t* list = lv_list_create(lv_scr_act());
    lv_list_add_btn(list, NULL, "Item 1");
    lv_list_add_btn(list, NULL, "Item 2");
    lv_list_add_btn(list, NULL, "Item 3");
    lv_list_add_btn(list, NULL, "Item 4");
    lv_list_add_btn(list, NULL, "Item 5");
    lv_obj_swap(lv_obj_get_child(list, 0), lv_obj_get_child(list, 4));
    while (!lv_win32_quit_signal) {
        lv_task_handler();
    }
}
```

glueckm, 2021-08-18:
https://user-images.githubusercontent.com/1733529/129915526-6b878135-5d5a-4a97-9f63-9e72611e0cd9.mp4

kisvegabor, 2021-08-18:
Good catch. The position in the group needs to be swapped too. 

I'll add a fix tomorrow.

kisvegabor, 2021-08-19:
I've pushed a fix.

glueckm, 2021-08-19:
First of all thanks for the quick fix. Unfortunately, i have to reopen the issue, becasue there seems to be a problem if you swap the item which currently has the focus:
Internal, the system thinjs the new item is the focused, but the old item is drawn as focued.
Once the focus chanin moves to the "old" first, the screen looks correct.
It's easier to see I guess:

https://user-images.githubusercontent.com/1733529/130103207-d8afc422-cec1-4de8-99d7-667f6b4fa40c.mp4

kisvegabor, 2021-08-19:
Thanks for the report, I haven't think of that. I leave this issue open until your confirm it's working well. :slightly_smiling_face:

glueckm, 2021-08-19:
Perfect. Now it works as expected.
Thanks again for the quick fix!

closed 2021-08-19 by glueckm

kisvegabor, 2021-08-19:
Great, thank you for the feedback!

commit 52c7558a (2021-08-19): fix(obj) swap objects in the group too in  lv_obj_swap()

fix #2462
commit a9971471 (2021-08-19): fix(group) keep the focused object in lv_group_swap_obj

fixes #2462