lvgl/lvgl #2464: issue for calendar

ZhaoQiang-b45475, 2021-08-19:
As the capture below, I use lv_calendar_header_arrow_create to create a header for calender and set the 
bg_color, but seems mismtach, there are gaps between the bg_color and the button.
BTW, is there any way to set bg_color of the button?
[image]

In addition, according to the description of LV_PART_ITEMS, I have no idea how to set the style for day names/previous-next month days/todays and highlight days.
[image]

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2021-08-19:
By default, there is a shadow below the buttons. 

I've pushed a fix to make behave better on non-white backgrounds.

kisvegabor, 2021-08-19:
> In addition, according to the description of LV_PART_ITEMS, I have no idea how to set the style for day names/previous-next month days/todays and highlight days.

I've forgot to answer these. As there are to many config options here the calendar uses a "custom drawer" to modify the appearance of the text and buttons. You can add an other draw event to do something similar. 
See https://github.com/lvgl/lvgl/blob/master/src/extra/widgets/calendar/lv_calendar.c#L261

closed 2021-08-19 by kisvegabor

ZhaoQiang-b45475, 2021-08-20:
Hi kisvegabor,
do you mean, I could define a cb function like draw_part_begin_event_cb, then add it to my calendar instance?
such as:

static void draw_custom_event_cb(lv_event_t * e)
{ ... }

lv_objt_t *calendar = lv_calendar_create(lv_src_act());
lv_obj_add_event_cb(obj, draw_custom_event_cb, LV_EVENT_DRAW_PART_BEGIN, NULL);

kisvegabor, 2021-08-24:
Yes, exactly.

commit c5342e93 (2021-08-19): fix(theme) use opacity on button's shadow in the default theme

fixes #2464