lvgl/lvgl #2466: Tests not run or reported in GitHub workflow (CI).

cmumford, 2021-08-19:
I wanted to see a test failure, so forked LVGL, and pushed an [intentional failure](https://github.com/cmumford/lvgl/commit/7d0d117312fdfe1935b924445d0235b7615c5554) to my repo. The tests, when run locally, do fail expected:

```
Run
-----------------------
./src/test_cases/test_dropdown.c:35:test_dropdown_create_delete:FAIL: Expected 5 Was 4
...
```

If I am reading the [workflow output](https://github.com/cmumford/lvgl/runs/3373694657) correctly, the test is being run, but there is no failure.

### Workflow CI
```
Build
-----------------------
./src/test_cases/test_dropdown.c:7:test_dropdown_create_delete:PASS
./src/test_cases/test_dropdown.c:8:test_dropdown_set_text_and_symbol:PASS
./src/test_cases/test_dropdown.c:9:test_dropdown_set_options:PASS
[Warn]	(0.000, +0)	 lv_dropdown_get_selected_str: lv_dropdown_get_selected_str: the buffer was too small 	(in lv_dropdown.c line #377)
./src/test_cases/test_dropdown.c:10:test_dropdown_select:PASS
./src/test_cases/test_dropdown.c:11:test_dropdown_click:PASS
./src/test_cases/test_dropdown.c:12:test_dropdown_keypad:PASS
./src/test_cases/test_dropdown.c:13:test_dropdown_encoder:PASS
./src/test_cases/test_dropdown.c:14:test_dropdown_render_1:PASS
./src/test_cases/test_dropdown.c:15:test_dropdown_render_2:PASS
```

My guess is that either the the forked master branch is being compiled and tested, or lvgl/master. I think the desired behavior would be to test the repo/branch containing the commit.

embeddedt, 2021-08-19:
CI is [only run on `master` or a release branch](https://github.com/cmumford/lvgl/blob/52c7558ab46a7024e05499edb483f115b13086f0/.github/workflows/ccpp.yml#L4-L7) at the moment. I'm not sure what the reason for this is. @kisvegabor?

cmumford, 2021-08-19:
Yeah, I saw that in the docs, but the build is triggered upon a push to *another* branch - which doesn't test the code in that branch.

kisvegabor, 2021-08-19:
It's really strange. I have no idea why it was triggered on your `test-test` branch, 

@embeddedt knows much more about GitHub actions than me but we use the "standard" [checkout@v2](https://github.com/lvgl/lvgl/blob/master/.github/workflows/ccpp.yml#L15) which I guess checks out the branch that triggered the action.

> I'm not sure why we don't run CI on all branches. @kisvegabor?

Let's say there is a `fix/something` branch for `v7` or there is a direct commit on the `release/v7` branch. We can't run v8 tests on them. So I decided to run the test only on branches that are certainly related to v8.

We haven't discussed it so probably is more sophisticated solution for this.