lvgl/lvgl #2469: Arc gradient

qinshijing, 2021-08-20:
### Introduce the problem
Hello, I need to implement a circular arc gradient, but I can't seem to find the relevant API. How can I achieve similar functions?

### Examples and cases
[image]

### Suggested solution

kisvegabor, 2021-08-24:
You can use an background image for the arc and the arc will be masked out from that image. 
See  https://docs.lvgl.io/master/overview/style-props.html#arc-img-src

qinshijing, 2021-08-25:
Yes. But this seems to only solve the background of arc, not the indicator of arc.

kisvegabor, 2021-08-25:
You can set it for the indicator too

```c
lv_obj_set_style_arc_img_src(arc, &img1, LV_PART_INDICATOR).
```

qinshijing, 2021-08-25:
Maybe the arc gradient I want is a gradient similar to a rectangular background, given a bg_color and grad_color, grad_dir (clockwise, counterclockwise) and then arc automatically follows an arc gradient. Now LVGL doesn't seem to support this feature. If it does, I think this is pretty cool.

closed 2021-08-25 by qinshijing