lvgl/lvgl #2471: If the radius is not zero, obj of the same size cannot cover another.

qinshijing, 2021-08-20:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
If the radius is not zero, obj of the same size cannot cover another.

### To Reproduce
void test(void)
{
    lv_obj_set_style_bg_color(lv_scr_act(), lv_color_black(), 0);
    lv_obj_t* test1 = lv_obj_create(lv_scr_act());
    lv_obj_set_size(test1, 200, 200);
    lv_obj_set_style_bg_color(test1, lv_color_hex(0xff0000), 0);
    lv_obj_set_style_bg_opa(test1, LV_OPA_100, 0);
    lv_obj_set_style_border_width(test1, 0, 0);
    lv_obj_set_style_radius(test1, LV_RADIUS_CIRCLE, 0);
    lv_obj_set_style_clip_corner(test1, true, 0);
    lv_obj_center(test1);
    lv_obj_t* test2 = lv_obj_create(lv_scr_act());
    lv_obj_set_size(test2, 200, 200);
    lv_obj_set_style_bg_color(test2, lv_color_hex(0x000000), 0);
    lv_obj_set_style_bg_opa(test2, LV_OPA_90, 0);
    lv_obj_set_style_radius(test2, LV_RADIUS_CIRCLE, 0);
    lv_obj_set_style_clip_corner(test2, true, 0);
    lv_obj_set_style_border_width(test2, 0, 0);
    lv_obj_center(test2);
}

### Expected behavior
Completed cover another.

### Screenshots or video
[image]

kisvegabor, 2021-08-24:
Just to be sure you mean t color beading on the circumference of the circle, right?

If so I think there is no simple fix for it on LVGL's rendering engine level. On resource constrained system we need to make compromises. :slightly_frowning_face:

qinshijing, 2021-08-24:
Maybe my expression is not clear enough. I mean, as described in Reproduce, the sizes of test1 and test2 are the same, and test2 is above test1, why can't test2 completely occlude test1?

kisvegabor, 2021-08-24:
I thought it's intentional: `test2` has only 90% opacity:
```c

lv_obj_set_style_bg_opa(test2, LV_OPA_90, 0);
```

qinshijing, 2021-08-24:
Yes. But if test2 is 100% opacity, you can still see a circle with 1 pixel around it, I'm not sure if this is an abnormality. I don't think this should happen.
[image]

kisvegabor, 2021-08-24:
Yes, that's what I meant by "color beading on the circumference of the circle". I'm afraid there is no simple fix for it if want to keep the fast rendering with low memory usage.

qinshijing, 2021-08-24:
Okay. I wanted to use an obj of the same size as a mask to change the overall opacity. It seems that I have to use other methods.

kisvegabor, 2021-08-24:
Depending on the other parts of the UI you can make `test2` 1-1 px greater in every direction:

```c
  lv_obj_set_size(test2, 202, 202);
```

[image]

qinshijing, 2021-08-24:
Thanks! I used the following methods to change the overall transparency, there shouldn't be a problem, right?

lv_color_t color_filter_cb(const struct _lv_color_filter_dsc_t* dsc, lv_color_t color, lv_opa_t opa)
{
    return lv_color_mix(lv_color_black(), color, opa);
}

void test(void)
{
    lv_obj_set_style_bg_color(lv_scr_act(), lv_color_black(), 0);
    lv_obj_t* test1 = lv_obj_create(lv_scr_act());
    lv_obj_set_size(test1, 200, 200);
    lv_obj_set_style_bg_color(test1, lv_color_hex(0xff0000), 0);
    lv_obj_set_style_bg_opa(test1, LV_OPA_100, 0);
    lv_obj_set_style_border_width(test1, 0, 0);
    lv_obj_set_style_radius(test1, LV_RADIUS_CIRCLE, 0);
    lv_obj_set_style_clip_corner(test1, true, 0);
    lv_obj_center(test1);

    static lv_color_filter_dsc_t dsc = { 0 };
    dsc.filter_cb = color_filter_cb;
    lv_obj_set_style_color_filter_dsc(test1, &dsc, 0);
    lv_obj_set_style_color_filter_opa(test1, LV_OPA_70, 0);
}

kisvegabor, 2021-08-24:
It also works but won't change the color of the children, except the labels because they can inherit their color from the parent.

qinshijing, 2021-08-24:
Thanks, I understand how to do it.

kisvegabor, 2021-08-24:
Great, I close this issue than.

closed 2021-08-24 by kisvegabor