lvgl/lvgl #2474: Converting `tests/Makefile` to CMake

cmumford, 2021-08-21:
### Introduce the problem
1. LVGL tests are too closely coupled to the OS (AKA don't run on Windows).
2. LVGL tests do build and run in parallel. (we do `make -j`, but only one make at a time).

### Examples and cases
1. The dominant OS for LVGL development is probably Linux, having the tests be platform agnostic would be beneficial, as more developers could run them before uploading to GitHub.
2. The tests would run approx. 4x faster on GitHub (maybe locally too) if run in parallel. 

### Suggested solution
One solution is to convert `tests/Makefile` to a CMake project file. The LVGL project files are already CMake, so this would eliminate one project file format from the mix. It would collect all build artifacts into a common subdirectory, differently named for each test configuration. This would allow parallel build/execution, as well as simplifying the cleaning task.

kisvegabor, 2021-08-25:
Sounds reasonable. 
@embeddedt attempted to migrate the test to CMake but has some issues.

cmumford, 2021-08-25:
OK. I have a version that is close to being ready for a PR. I'll tag @embeddedt when I send it out.