lvgl/lvgl #2480: How to get  an obj width

surmo, 2021-08-23:
### Introduce the problem

### Examples and cases

### Suggested solution

embeddedt, 2021-08-23:
That's correct. There is an explanation about it in the [documentation](https://docs.lvgl.io/master/overview/coords.html#postponed-coordinate-calculation).

----------------------
We use GitHub issues for development-related discussions.
Please use the [forum](https://forum.lvgl.io/) to ask questions.

closed 2021-08-23 by embeddedt

FloHomi, 2024-09-18:
I am currently in the same spot, for some reason it does not work for me. (current main branch)

`    lv_obj_t* icon_button = lv_btn_create(parent);
    lv_obj_set_size(icon_button, LV_SIZE_CONTENT, LV_SIZE_CONTENT);

    if (text_top != NULL)
    {
        //create a label for the bottom text
        lv_obj_t* label_top = lv_label_create(icon_button);
        lv_label_set_text(label_top, text_top);
        lv_obj_align(label_top, LV_ALIGN_CENTER, 0, -40);
        lv_obj_set_user_data(label_top, BWI_INDEX_TXT_TOP);
    }
    if (icon != NULL)
    {
        //create a icon label, instead of an image
        lv_obj_t* label_icon = lv_label_create(icon_button);
        lv_obj_add_style(label_icon, icon_style, LV_PART_MAIN);
        lv_label_set_text_fmt(label_icon, "%c", icon);
        lv_obj_center(label_icon);
        lv_obj_set_user_data(label_icon, BWI_INDEX_ICON);
    }
    if (text_bottom != NULL)
    {
        //create a label for the middle text
        lv_obj_t* label_bottom = lv_label_create(icon_button);
        lv_label_set_text(label_bottom, text_bottom);
        lv_obj_align(label_bottom, LV_ALIGN_CENTER, 0, 40);
        lv_obj_set_user_data(label_bottom, BWI_INDEX_TXT_BOTTOM);
    }

    lv_obj_update_layout(icon_button);
    int icon_width = lv_obj_get_width(icon_button);
    int icon_height = lv_obj_get_height(icon_button);
`

I Always get the same width. It does not matter how long the text in the label is.

also I noticed that lv_obj_update_layout is updateding the whole layout of the entire screen. is that correct? That is a lot of looping through a lot of widgets. is there a more performant way?

kisvegabor, 2024-09-19:
I've simplified it a little an it's working well for me:
```c
  lv_obj_t* icon_button = lv_btn_create(lv_screen_active());
  lv_obj_set_size(icon_button, LV_SIZE_CONTENT, LV_SIZE_CONTENT);

  lv_obj_t* label_icon = lv_label_create(icon_button);
  lv_label_set_text_fmt(label_icon, "%s", "Hello world longer version!");
  lv_obj_center(label_icon);

  lv_obj_update_layout(icon_button);
  int icon_width = lv_obj_get_width(icon_button);
  int icon_height = lv_obj_get_height(icon_button);
  printf("%d, %d\n", icon_width, icon_height);
```

FloHomi, 2024-09-19:
yeah,
it appears to be a problem with the y_offset. as soon this is not 0, it does not work anymore. x_offset appears to be fine.

kisvegabor, 2024-09-19:
You can CENTER in a container with CONTENT size. t'd be circular dependency as size depends on the position, but position (centered) depends on the size. You can use
```c
lv_obj_align(label_top, LV_ALIGN_TOP_MID, 0, 30);
```
instead