lvgl/lvgl #2482: add UTF16 support.

incity, 2021-08-23:
Will lvgl support utf16 in the future?

embeddedt, 2021-08-23:
I think it's unlikely, since we already support UTF-8, and as far as I know, UTF-8 can represent any Unicode codepoint. Also, I'm pretty sure supporting UTF-16 requires us to either implement our own wide character handling functions (`wstrlen`, etc.), or rely on the underlying platform to provide them.

kisvegabor, 2021-08-24:
I also believe UTF8 is a more versatile option, especially because it's backward compatible with ASCII. 

Do you prefer UTF16 because it needs less bytes for CJK characters?

Miguel0101, 2021-08-25:
UTF-8 is compatible with ASCII while UTF-16 is incompatible with ASCII.

incity, 2021-08-25:
BMP (basic multilingual plane) characters, including Latin, Cyrillic, most Chinese (the PRC made support for some codepoints outside BMP mandatory), most Japanese can be represented with 2 bytes. This speeds up indexing and calculating codepoint count in case the text does not contain supplementary characters.

incity, 2021-08-25:
utf-16 was chosen as a balance between size in bytes (utf-8) and performance (utf-32). It is faster than utf-8, but not as big as utf-32. also Windows and some Linux, use utf-16 by default

Miguel0101, 2021-08-25:
> I also believe UTF8 is a more versatile option, especially because it's backward compatible with ASCII.

I agree.

kisvegabor, 2021-08-25:
UTF-8 is very deeply wired in LVGL so unless it's absolutely required I wouldn't change it.

closed 2021-09-16 by lvgl-bot