lvgl/lvgl #2489: Unwanted Artifacts/Dots in Glyphs

elTeriblePacman, 2021-08-24:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2021-08-24:
Do you also see the issue if the fonts are converted to C array?

elTeriblePacman, 2021-08-25:
Hello,
No I wasn't able to replicate the issue when using C array as the output format. Please see below image for reference (highlighted in yellow):

[image]

The repo I linked above was updated.
I used the lines below to generate the fonts in C array format:

With compression, C array
env DEBUG=* lv_font_conv --font BAUHS93.TTF -r 0x20-0x7F --size 16 --format lvgl --bpp 4 -o bauhaus_16_compressed_array.c

Without compression,  C array
env DEBUG=* lv_font_conv --no-compress --font BAUHS93.TTF -r 0x20-0x7F --size 16 --format lvgl --bpp 4 -o bauhaus_16_notcompressed_array.c

kisvegabor, 2021-08-25:
Thanks. I think I found the problem. Please try adding this line to `lv_font_conv.c`:
```diff
--- a/src/font/lv_font_loader.c
+++ b/src/font/lv_font_loader.c
@@ -454,6 +454,8 @@ static int32_t load_glyph(lv_fs_file_t * fp, lv_font_fmt_txt_dsc_t * font_dsc,
                 }
             }
             glyph_bmp[cur_bmp_size + bmp_size - 1] = read_bits(&bit_it, 8 - nbits % 8, &res);
+            /*The last fragment should be on the MSB but read_bits() will place it to the LSB*/
+            glyph_bmp[cur_bmp_size + bmp_size - 1] = glyph_bmp[cur_bmp_size + bmp_size - 1] << (nbits % 8);
             if(res != LV_FS_RES_OK) {
                 return -1;
             }
```