lvgl/lvgl #2490: Arc widget does not draw properly after a full revolution

weixiongmei, 2021-08-24:
### Describe the bug
The Arc widget some times does not draw properly after a full revolution. Please just look at the Arch example from the link https://docs.lvgl.io/master/examples.html , it happens once every few revolutions in the simulator, but one the actual device, it happens every time after a full revolution.

### To Reproduce
```c
#include "../../lv_examples.h"

#if LV_USE_ARC && LV_BUILD_EXAMPLES

static void set_angle(void * obj, int32_t v)
{
    lv_arc_set_value(obj, v);
}

/**
 * Create an arc which acts as a loader.
 */
void lv_example_arc_2(void)
{
  /*Create an Arc*/
  lv_obj_t * arc = lv_arc_create(lv_scr_act());
  lv_arc_set_rotation(arc, 270);
  lv_arc_set_bg_angles(arc, 0, 360);
  lv_obj_remove_style(arc, NULL, LV_PART_KNOB);   /*Be sure the knob is not displayed*/
  lv_obj_clear_flag(arc, LV_OBJ_FLAG_CLICKABLE);  /*To not allow adjusting by click*/
  lv_obj_center(arc);

  lv_anim_t a;
  lv_anim_init(&a);
  lv_anim_set_var(&a, arc);
  lv_anim_set_exec_cb(&a, set_angle);
  lv_anim_set_time(&a, 1000);
  lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);    /*Just for the demo*/
  lv_anim_set_repeat_delay(&a, 500);
  lv_anim_set_values(&a, 0, 100);
  lv_anim_start(&a);
}
#endif
```

### Expected behavior
Good
[image]

### Screenshots or video
No Good
[image]

kisvegabor, 2021-08-25:
Thanks for the report. I've just fixed it.

kisvegabor, 2021-09-02:
Do you still see the issue? If so can you reproduce it in C as well? (not Micropython)

weixiongmei, 2021-09-02:
Yes, i do still having the same issue event updated to the most recent commits.  And the Micropython simulator is having the exact same issue that i'm having. I deleted the sources, and clone it from git again, doesn't helps. And i tried the exact C code that you posted, it's the same, so really don't know what's going on....
[image]
[image]

kisvegabor, 2021-09-03:
I'm looking into it. 
As work around to hide the issue you can call `lv_obj_invalidate(arc)` after setting a new value.

weixiongmei, 2021-09-04:
Thank you, then work around works.... Hope the issue can be fixed soon...

kisvegabor, 2021-09-09:
I took me some time to understand what is the real problem but I believe now it work well. If still don't at least know I better know how to approach the issue :slightly_smiling_face:  

Please test it and let me know how it works.

weixiongmei, 2021-09-09:
Hi @kisvegabor, the issue is fixed now, thank you so much for your works...

kisvegabor, 2021-09-10:
Great, thank you for the feedback!

commit 98b9ce59 (2021-08-25): fix(arc) fix full arc invalidation

fixes #2490
closed 2021-09-10 by kisvegabor