lvgl/lvgl #2493: Problems encountered when using textarea

Johnxjj, 2021-08-25:
There is a problem that has bothered me for a long time. The function I want to do is: click on the textarea, an input method interface will pop up below, click on the blank space, the input method interface will be hidden. But I don’t know how the textarea distinguishes whether the click position is in the textarea area or outside the area.

Johnxjj, 2021-08-25:
There is a problem that has bothered me for a long time. 
 The function I want to do is: click on the textarea, an input method interface will pop up below, click on the blank space, the input method interface will be hidden. But I don’t know how the textarea distinguishes whether the click position is in the textarea area or outside the area.

kisvegabor, 2021-08-25:
You can use the `LV_EVENT_DEFOCUSED` on the text area to hide it.

---

We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account. keyboard.

closed 2021-08-25 by kisvegabor